luc-github/ESP3D #140: Firmware Updater. How to use?

khseal, 2016-12-03:
I compile the firmware in the Arduino Ide.
I tried to flash. I got a firmware failure.

khseal, 2016-12-03:
I have a flash memory of 1MB. Under spiff allocated 512kb. Possible to update the firmware is not enough memory, and it was necessary to allocate a spiff 128kb?

luc-github, 2016-12-03:
Did you check the FAQ? https://github.com/luc-github/ESP3D/issues?utf8=%E2%9C%93&q=label%3AFAQ+

khseal, 2016-12-03:
I checked. Nothing about the update is not found. But I realized that himself has created the problem.

closed 2016-12-03 by khseal

luc-github, 2016-12-03:
Yes I am confused you talk about SPIFFS size necessary for having webupdate working, so this is in FAQ
and now you talk about update not found ? it is a file upload in system page

khseal, 2016-12-03:
My English is very bad. Therefore, I can not right the first time to find information in English. =)