luc-github/ESP3D #141: After printing begins, SD cards, freeze ESP.

khseal, 2016-12-03:
After starting from SD card printing. ESP stops responding and hangs. No ping ESP. 
Firmware printer Marlin.

luc-github, 2016-12-03:
I cannot duplicate at least on Repetier
please share what version of Marlin and how you launch the SD print
I all case the nothing should freeze ESP so I suspect a power stability issue -  does it freeze when you start to heat extruder ?

khseal, 2016-12-03:
ESP has a separate power supply. When I print from a computer very well.
I use this firmware http://www.thingiverse.com/thing:1859628
I run printing with web interface ESP3D.
I think the problem occurs when using the web interface.
3d printer does not freeze.

luc-github, 2016-12-03:
does problem happen when launch print from printer?

khseal, 2016-12-03:
Later I check. Now the printer is busy.

khseal, 2016-12-03:
I'm running with the printer printing from sd card problem continues. I think if you start printing on the wi-fi and open the printer's interface, it in the same freezes.
Here's a video https://www.youtube.com/watch?v=YG_-JJlo7ps&feature=youtu.be

luc-github, 2016-12-03:
can you share your  test gcode ? I cannot replicate with mine

khseal, 2016-12-03:
I think the problem is not gcode.

[test2.gco.tar.gz](https://github.com/luc-github/ESP3D/files/629131/test2.gco.tar.gz)

luc-github, 2016-12-03:
I just try to be same condition as you as I cannot reproduce

luc-github, 2016-12-03:
OK can reproduce now - will investigate

luc-github, 2016-12-03:
You are correct it affect only web interface - no issue if use only data port for M23 <filename> then M24

It seems it is side effect of a recent update for catching command answer 
when heating, printer send a lot of temp status which overload / make dead loop of ESP which is waiting for a "quiet" period to send the command and catch the answer

I will see how to work around

luc-github, 2016-12-04:
I did several fixes to avoid to enter to long loop and fire WD
there is a detection of multiple answer like `T:27.84 / 0 B:53.50 / 0 B@:0 @:0 T0:27.84 / 0 @0:0 T1:1.89 / 0 @1:0` if over 5 then the answer check is skipped 

I think need to add a specific handle for silent command which do not need to wait for answer, so this would also save the multiple temperature detection time out.

But currently issue is mitigated -  I will close when silent command is implemented, need UI and FW update

khseal, 2016-12-04:
Ok, I'll check hotfix.

khseal, 2016-12-04:
This bug has been fixed. But I found another bug.
I pressed stop printing after printing with the launch of the SD card and the ESP again freeze after message: pause user. Though I pressed the stop print.

luc-github, 2016-12-04:
the printer not responding to stop button when heating and printing from SD is not related to ESP but printer FW - even sending M112 is not working because printer FW is in internal  loop checking temp and not taking any new command. Stop / pause are not immediate but depend on current command buffer size 

I also  just see some freeze  on my long test run but not sure where it come from yet, so need to dig in

luc-github, 2016-12-04:
repetier has a flag: `busy:<action>` that may overflow the answer which only check temp when heating 
like `busy:heating` and `busy:processing` 
also I need to verify the freeze come from the web command part so need to do several long test run

Should have some update tomorrow

khseal, 2016-12-04:
Ok, I'm waiting. These flags can be attributed to the log status.

luc-github, 2016-12-04:
I just finished the long test run with latest fixes https://github.com/luc-github/ESP3D/commit/1269e9b09a779b4e0140581c9fa3c6713eca0f7e on repetier and no more freeze
I can navigate on web pages and come back to printer monitoring page with data port used, I can stop the print, M205 still give the content of EEPROM properly 
 
Still several tests to do but it looks good,

tomorrow I will do more test: no use of tcp monitoring, and long run tests on smoothie and marlin firmwares

khseal, 2016-12-05:
I continued to test and broke all =) I put the printer latest firmware marlin. Version RCBugFix.
http://www.thingiverse.com/thing:1935266
A computer in the direct USB work without any problems.
ESP does not see the 3D printer. This is strange.

luc-github, 2016-12-05:
Ho this is due to latest commit - on my test only smoothie has answer starting by ok, not marlin, but I do not have latest marlin, seems now marlin act like smoothie

I just remove marlin from the ok filtering : https://github.com/luc-github/ESP3D/commit/29e835d101c66899414b092770a766d2e55f6004
so it will filter ok answer only on repetier, this should solve temperature display, for status I am not sure

khseal, 2016-12-05:
I checked the changes. Status and temperature working.

luc-github, 2016-12-05:
good, silent command is now implemented, I will do long run on smoothie and marlin this afternoon

khseal, 2016-12-05:
I started printing the card. The temperature and the status of the work to the table heating. After heating the table, status and temperature of the missing.

luc-github, 2016-12-05:
Not sure I get your meaning - is it google translation ?

khseal, 2016-12-05:
This is such bad English =)  Temperature and status do not work in web interface after heating table.

luc-github, 2016-12-05:
Do you mean after heating bed :  the temperature are not displayed ?

Do you have screenshot of web UI and also the data port output ?

khseal, 2016-12-05:
Sometimes errors occur
No Line Number with checksum, Last Line: 7051
Line Number is not Last Line Number + 1
These errors occur if you open the web interface when printing from a computer.

A little later, I will show a screenshot and logs port.

khseal, 2016-12-05:
` T:34.8 /0.0 B:69.2 /70.0 @:0 B@:127 W:5
 T:34.5 /0.0 B:69.4 /70.0 @:0 B@:127 W:4
 T:34.1 /0.0 B:69.6 /70.0 @:0 B@:127 W:3
 T:34.3 /0.0 B:69.6 /70.0 @:0 B@:127 W:2
 T:34.0 /0.0 B:69.8 /70.0 @:0 B@:127 W:1
 T:34.1 /0.0 B:69.8 /70.0 @:0 B@:127 W:0
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:1
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:2
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:4
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:6
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:8
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:11
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:14
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:18
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:21
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:26
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:30
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:35
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:40
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:46
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:52
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:58
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:64
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:71
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:79
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:86
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:96
ok
echo:Unknown command: "MM114"
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X: 0 Y:0 Z:111
ok
 T:38.0 /215.0 B:70.9 /70.0 @:127 B@:0 W:?
 T:39.7 /215.0 B:70.8 /70.0 @:127 B@:0 W:?
 T:40.8 /215.0 B:70.7 /70.0 @:127 B@:0 W:?
 T:41.9 /215.0 B:70.8 /70.0 @:127 B@:0 W:?
 T:44.3 /215.0 B:70.6 /70.0 @:127 B@:0 W:?
 T:46.1 /215.0 B:70.5 /70.0 @:127 B@:0 W:?
 T:48.0 /215.0 B:70.5 /70.0 @:127 B@:0 W:?
 T:50.3 /215.0 B:70.4 /70.0 @:127 B@:0 W:?
 T:52.8 /215.0 B:70.2 /70.0 @:127 B@:0 W:?
 T:55.2 /215.0 B:70.2 /70.0 @:127 B@:0 W:?
 T:57.5 /215.0 B:70.1 /70.0 @:127 B@:0 W:?
 T:59.9 /215.0 B:69.9 /70.0 @:127 B@:0 W:?
 T:62.0 /215.0 B:70.0 /70.0 @:127 B@:0 W:?
 T:64.6 /215.0 B:69.9 /70.0 @:127 B@:127 W:?
 T:67.0 /215.0 B:69.8 /70.0 @:127 B@:127 W:?
 T:69.5 /215.0 B:69.9 /70.0 @:127 B@:127 W:?
 T:71.8 /215.0 B:70.1 /70.0 @:127 B@:127 W:?
Strange "echo:Unknown command: "MM114"" ?
Screenshot web interface printer after update. Before update freeze indication.
[image]

luc-github, 2016-12-05:
M114 is send every 3 sec to get axis position 
if you get MM114 it means serial communication issue
do your baud rate is identical on printer and esp ?
if yes you may try to decrease a little to see if it improve 

I do not have a same output from my marlin kalimbra and it is just a board without heater - I need to setup a system to reproduce your issue, I see you do not have anymore the ok in front of Temperatures did you changed something ?

khseal, 2016-12-05:
I have not changed anything. Ok in front, appears when the printer is idle.
Baud rate of use 250000. I'll try to reduce Baud rate.

luc-github, 2016-12-05:
Also how did you connect your esp to mega/ramps 1.4 ? 
I will try to build same configuration

khseal, 2016-12-05:
[image]

khseal, 2016-12-06:
Сontrol feed filament in the web interface does not work.

luc-github, 2016-12-06:
is temperature enough hot?
what is the output on data port?

khseal, 2016-12-06:
Temperature extruder 200C.
Data port message:
ok T:200.0 /200.0 B:61.3 /0.0 @:59 B@:0
ok
ok
ok
X:0.00 Y:0.00 Z:15.05 E:0.00 Count X: 0 Y:0 Z:12040
In Repetier-Host all ok.

luc-github, 2016-12-06:
sending the commands manually G91,G1 E50 F120, G90 works ?

luc-github, 2016-12-06:
I see It is an old bug, wrong GCODE is in printer. tpl orz...E0 and E1 instead of E...
will fix later

khseal, 2016-12-06:
Ok. Oh, new verison firmware https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC8 =)
I need tested =)

luc-github, 2016-12-06:
the extruder issue should be fixed by https://github.com/luc-github/ESP3D/commit/8e7e2ba7619b54253259da2cb1a6b1677985604d, I fixed it long time ago but seems I overwrited the changes - so it was never implemented ....

luc-github, 2016-12-06:
Seems there is a command to stop heating when printing from SD M108 - you shoud add it to stop button command

khseal, 2016-12-07:
I checked the command M108.
It does not change anything.
No output data after command.
This is because it is necessary to enable in the firmware
// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
//#define EMERGENCY_PARSER
Tomorrow, I check this option.

khseal, 2016-12-07:
M108 command does not stop printing. It only stops heating.

luc-github, 2016-12-07:
yes I know, but because heating stop command is not working, so before sending stop gcode, send M108 twice, one to stop heating extruder, one to stop heating bed, then stop command can be catched and work properly

khseal, 2016-12-09:
Yes, this set of commands will work.

luc-github, 2016-12-21:
any update on this ? 
I have tried and not able to freeze

khseal, 2016-12-21:
Yes, the problem seems to be solved.

luc-github, 2016-12-22:
Ok I close issue then - thanks a lot for your help

khseal, 2017-02-27:
Not all freezing esp resolved. 
ESP freezes when open web management interface and run heating in Repetier-Host.
If the web interface is closed then everything is fine.
Algorithm freezing ESP difficult to define.

luc-github, 2017-02-27:
you mean using esp connected to serial and host also connected to serial and repetier host is heating from manual command ?

khseal, 2017-02-27:
repetier host connected to the ip port of ESP. Yes, heating manual command.

luc-github, 2017-02-27:
which version of FW are you using ?

khseal, 2017-02-28:
Version one of the last. Commits on Jan 30, 2017.

luc-github, 2017-02-28:
Ok will check that when back home in 2 weeks - I cannot reproduce now due to lack of device

skish, 2017-03-03:
Check the esp power source. I got freezes due bad power, after fix that freezes gone.

skish, 2017-03-03:
Also ramps tx should be connected to esp rx thru voltage level shifter.

khseal, 2017-03-03:
No. It is software problem.

luc-github, 2017-03-05:
Yes it is my mistake it same as : https://github.com/luc-github/ESP3D/issues/141#issuecomment-264713305
I missed some test case so ESP try to purge serial when printer is heating and send flow of temperature because command expect to catch answer and send it back
I need to add some timeout in all purge serial command / handle printer is busy in same loop

luc-github, 2017-03-05:
Ok seems different issue - I do not have printer now so I use a python script to send a flow of `T:25.0 /0.0 B:25.0 /90.0 @:0 B@:127 W:?` like marlin do when heating and all timeout are working well, esp does not freeze 
Seems I cannot reproduce issue without printer so will dig in later

khseal, 2017-03-05:
I will try to clarify the details of the problem.

luc-github, 2017-04-22:
Any update on this ?

khseal, 2017-04-22:
Maybe the problem is on my side. I can not find a pattern of a problem.

[2 more replies]

closed 2017-05-19 by luc-github