luc-github/ESP3D #142: Async WebSockets Client

Sparkxxx, 2016-12-09:
Hello Luc and congratulations for greatest creation i've seen so far in the esp config world. I know it's kinda off topic but I'm asking for your advice although others could benefit from your answer.
I have made some tests with both esp3d versions (including async webserver) trying to integrate websockets client library Links2004/arduinoWebSockets. With normal websockets it works but the transmit intervals are not even and it's throttling so I thought to try the async client and in both esp3d versions I get the ESP to crash with the exact same exception:

Exception (28):
epc1=0x4021214e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff2150 end: 3fff2450 offset: 01a0

>>>stack>>>
0x4020d5ec: WebSocketsClient::clientIsConnected(WSclient_t*) at ?? line ?
0x4020e45d: WebSocketsClient::handleHeader(WSclient_t*, String*) at ?? line ?
0x402135ec: std::_Function_handler    (WebSocketsClient*, WSclient_t*, String*)> >::_M_invoke(std::_Any_data const&, bool, void*) at ?? line ?
0x4021250f: std::function ::operator()(bool, void*) const at ?? line ?
0x40212544: AsyncTCPbuffer::stop() at ?? line ?
0x4020e85e: WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int) at ?? line ?
0x4020d57d: WebSocketsClient::clientDisconnect(WSclient_t*) at ?? line ?
0x40202d74: yield at ?? line ?
0x4020e8a4: WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int) at ?? line ?
0x4020204c: String::~String() at ?? line ?
0x4020156a: HardwareSerial::available() at ?? line ?
0x40214119: BRIDGE::processFromSerial2TCP() at ?? line ?
0x4020d9ec: WebSocketsClient::disconnect() at ?? line ?
0x40213d65: loop at ?? line ?
0x40202d48: loop_wrapper() at core_esp8266_main.cpp line ?
0x40204ea0: cont_norm at cont.o line ?
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v02f0c112
~ld

I think that the problem is an incompatibility of some sort between the libraries but I don't have the knowledge to debug it. Hope you can help,
Thanks

luc-github, 2016-12-09:
well can be a lot but if you check here : https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md

exception 28 is a memory access violation - I would suggest to add debug flag and see what command generate the exception, but looking at the stack decoding, I see `0x40202d74: yield at ?? line ?` and async method must not use yield per readme (https://github.com/luc-github/ESPAsyncWebServer#important-things-to-remember) :
__You can not use yield or delay or any function that uses them inside the callbacks__

I hope it help

luc-github, 2016-12-21:
if no update I close issue

closed 2016-12-21 by luc-github