luc-github/ESP3D #143: Add progression during upload/update on printer

luc-github, 2017-01-01:
Currently there is only ESP upload or Update Firmware or SD upload message displayed, It would be nice IMHO that progression % show up on printer screen instead of fixed message - to confort user process is running on ESP side.
This is due to the fact that post arguments are not available during file upload processing 
2 solutions come to me:
1 - add a check query before sending the file(s) to send the file(s) size and eventually check space availability
so if no space or no connection the sending process is not launched and the files sizes is available before send
2 - embedded the file size in filename argument and extract file name and file size before processing

I think both methods can be done as if no split char is found for method 2, then rely on method 1 for size, and if size is not defined just display fixed message as before

denmorru, 2017-03-28:
Currently , when I upload file with web UI to printer (M28\M29), it does not show any real progress. It goes to 100% in 1st second, and stuck until file loaded.  
And also there is no refresh of files list after SD upload finished.

luc-github, 2017-03-29:
the progression bar speed depend on which OS/browser you use
the refresh delay may be due to this too - serial upload is said as done then UI send M20 the files list is created then the query to guet the list is done - may be add more time would sold your issue
also this ticket  is an enhancement on printer lcd not on web UI so nothing related to your issue

denmorru, 2017-03-29:
So, if this issue is about M117 or similar, I'm not sure how to show progress in LCD, because there is nothing extra could come between M28 and M29. So, this is more about ESP8266-memory uploads.

luc-github, 2017-03-29:
I do not follow what you said ...
are you talking about this ticket issue (progression on printer lcd) or about web UI (progression on browser interface) if about browser - please open another issue as it will become hard to properly track and fix issue

luc-github, 2018-12-09:
The file size is now accessible during post upload - so will be part of 3.X

closed 2018-12-09 by luc-github