luc-github/ESP3D #144: File upload over wifi to the printer slow

bubnikv, 2017-01-13:
I just installed the ESP3D on a NodeMCU board. It seems to be working. I tried uploading a file into the printer (no printer attached, just a blind test, the printer serial interface set to 250kbits per second) and the upload seems to be very slow, in the ballpark of 230kB / min. What file upload speeds can we expect with the ESP3D?

luc-github, 2017-01-13:
Hi - did you ever try file transfert using the M28/M29 command ?
this transfert using serial is slow like hell - and if you use repetier host / repetier fw it is even slower as every line is interpreted and converted to binary before being saved

unfortunatly the wifi is not the limitation the transfert protocol (M28/M29) is the limitation, the only solution are:
- to use another protocol but there is none supported by Marlin/repetier
- to connected SD card board to ESP but this is hardware rework in addition of firmware modification

File transfert on SD is not good unfortunatly

bubnikv, 2017-01-15:
Yeah, I am painfully aware about the g-code upload speed issues. It could
be quite improved though by modifying the firmware to support protocols
confirming multiple lines or blocks of binary data. Just by sending 5 lines
at once should speed up the communication 5 times.

Did you try to upload to a SD card from the esp8266 directly? What is the
upload speed then? I could imagine putting some kind of switch between the
ESP and Arduino to share the SD card.

luc-github, 2017-01-15:
there is no really way to improve speed by serial - as each command is checked one by one by main board and we need to get ack before going sending next one.

bubnikv, 2017-01-15:
But you can change Marlin or repetier to act differently. And at least on
the Arduino side you can switch the serial to 1mbit per second.

luc-github, 2017-01-15:
yes of course if marlin/smoothie/repetier had better upload process would be great - but I do not intend to change them it is out of my influence ^_^- I just do my best to handle what they do

bubnikv, 2017-01-15:
We did some changes to the Marlin firmware at Prusa Research to bump up
uploads for our print farm, so it is doable.

I just found out that the duet board uses the ESP8266 board as well and
that someone reused the ESP8266 firmware for a standalone controller,
similar to your project:
https://github.com/toxuin/ESPRINTER

Anyway, thinking about the duet and the ESP8266 and your project, I think
the best way would be to put the ESP8266 between the SD card and the
Arduino, maybe by simply relying the SD card commands by the ESP8266 and by
developing a synchronization protocol between let's say Marlin and the
ESP8266 firmware. That way the upload speed would not be limited.

luc-github, 2017-01-15:
@bubnikv yes of course is doable ^_^ - as said just out of my firmware scope as I want to be as portable as possible and I am not willing to push printer firmware (they are too many) to have better upload process and not sure everyone will implement it in same way 

About duet I checked their code already and they use another approach - ESP is not connected by serial but directly by SPI -fimrware and board are designed for it which make things easier of course.

Yes agree with you - I worked on it (https://github.com/luc-github/ESP3D/issues/93#issuecomment-237893338) and speed is very good, as good as my wifi sd card transfer

bubnikv, 2017-01-15:
Thanks for the link. Is the web upload / download to a SD card connected to the ESP8266 supported by the project?

I see the SDCARD_FEATURE define, but I am not quite sure what it does. My impression is, it only allows the html/css/javascript files to be stored on a SD card, but it does not support uploading / storing on the SD card. Am I correct?

What does DIRECT_SDCARD_FEATURE mean?

Thanks for your time. It is confusing to me, I tinkered only a bit with the ESP8266 Arduino before, so this is mostly new to me.

luc-github, 2017-01-15:
yes what current github code does not support direct SD card access, what you see is for an evolution of current code

bubnikv, 2017-01-15:
One way to make the SD access by the ESP8266 fast is to do it the octopi way: To ignore the SD card in the printer and to print from the ESP8266 online. This makes only sense though if the ESP8266 is stable enough. I know many who trust the octoprint to print multiple days prints from it.

I am going to evaluate the SDWebServer first. Let's see how convincing it will be.

luc-github, 2017-01-21:
the issue using esp as host is when streaming SD content it cannot handle well the wifi - esp is not multitasking and easily overloaded, a good example is do SD upload and try to connect to web page also doing some refresh - you will get several package lost and sd upload will failed

Also because it is not multitasking send print commands continuously and handle webUI will be a problem, so even it can be ok for cnc, for laser and 3d printer, to loose the printer control command flow  during the time it answer/ handle wifi/web request is not good at all.
for example during few ms of handleing command the flow will continue and the result will be overextrusion - same if using a laser the laser may overheat waiting for next command.

need also to handle pause and communication error which is feasible but I would consider esp32 in this case instead of esp8266

Just my 2 cents

luc-github, 2017-01-29:
if no update I close issue

closed 2017-01-29 by luc-github