luc-github/ESP3D #145: Web interface redesign

VitaliyAndreevich, 2017-01-21:
Web interface based on templates takes alot of time for CPU process and data sending. I have make some web modifications but for old relese 0.6.0. I think it is posible port it to current develop branch.

common ideas:
move css, js to separate files and add caching for it like:
 WebServer.serveStatic("/css/", SPIFFS, "/css/", "max-age=8640");
 WebServer.serveStatic("/js/", SPIFFS, "/js/", "max-age=8640");

also replace tpl files by html and also add it to chach. And all data which you need to replace like '$SOME_PARAM$' gets by json requests and replace by using javascripts inside web browser

luc-github, 2017-01-21:
if you use git version of espcore is very fast actually and based on my tests there is not a lot of difference but need latest git version of espcore

about separe files - esp web server do not like so much to handle more than 5 connections at once so split the page in several files decrease the responsiveness of webserver - that is why I packed all but again this is  just a choice, I do not rely so much on catching.

Using current git version (not old 0.6) all pages already have json format by adding `output=JSON` in query or in post form to get answer in json format instead of default HTML (https://github.com/luc-github/ESP3D/issues/114) so not only you can get ride of the template, but you also can use the compressed html pages which speed up the transfert

 The draw back is you need to load first html compressed or not  page, then when page is loaded, query the data to display like you describe, which it is slower than use template but give more flexibility in UI

Another issue I meet using json and pure html was the handling of authentication in a clean way but I am sure there is several solutions - just did not had time yet to review the best one.

Anyway using pure html (compressed or not) and json is already available from firmware point of view - 'just' need to rewrite all html pages supporting the new way - you more than welcome to push PR

VitaliyAndreevich, 2017-01-21:
Thank you for quick responce!

About sepate files (css and js):
You are right, first page loading will be slower becouse of increasing connections count, but next loading will faster becouse of PC browser caching and one more connection for css or js files loading will not be created. this idea can be implemented with current tpl realisation:

header.inc:
-<style>
-$INCLUDE[css.inc]$
-</style>
+<link rel="stylesheet" type="text/css" href="/css/style.css">
and move css.inc to /css/style.css

It will helpfull for cases with slow speed connections (thought router for example).

luc-github, 2017-01-29:
@VitaliyAndreevich Here the thread when I was testing asyncwebserver which is faster than current one : https://github.com/luc-github/ESP3D/issues/110#issuecomment-254247779
I was testing with 2 js and one css and got issues - got worst with sync server before
 
Esp8266 core developper and async webserver owner do not recommend to split files but to combine them and also to compress them, and even to put them in PROGMEM 

as based on your station test results - I can see only printer and home may get gain, still need to see the impact on server responsiveness as mentioned - the others pages do not get gain and even are worst 

I suggest you to have a look at the fork of web smoothieware interface I did to be used with other FW the footprint is small because use cdn links for js and css, the file is minified and compressed 

I will update the repo next week with a lot of fixes for marlin and repetier and even smoothieware ,but you can have look now, it should meet your expectations as it does not use template and use caches :wink:

leseaw, 2017-05-13:
Hi Luc,
and when will you release the new version? :)

luc-github, 2017-05-13:
Before end of this month - I finished it it is in test now, and I need to write the doc 
Of course I always fine some issue but current version is pretty stable so ready to be released

This FW is not compatible with current one so UI / FW are couple in this release not standalone

Also UI will have a separate UI I think

leseaw, 2017-05-13:
Hvae you an preview? or do you want test for you alone? :)

luc-github, 2017-05-13:
I may release a preview next week, but authentication support is not implemented on UI

luc-github, 2017-05-19:
Please check here : https://github.com/luc-github/ESP3D/tree/devt for Fw update 
and here for new UI files https://github.com/luc-github/ESP3D-WEBUI

luc-github, 2017-05-19:
as now UI moved to  https://github.com/luc-github/ESP3D-WEBUI I close topic

closed 2017-05-19 by luc-github