luc-github/ESP3D #147: About Nodemcu V3 board connect?

pelitt, 2017-01-29:
Hi i bought this board but not connect.
İ open 192.16... But Printer is busy. İ dont understand.

pelitt, 2017-01-29:
Not busy show offline

luc-github, 2017-01-29:
it means esp do not get communication with printer
what printer board and firmware do you use ?
how do you connect nodemcu to printer?
what baud rate on printer and ESP?

pelitt, 2017-01-30:
thanks for the answer.
1 My printer board MKS Base  1.5 and frimware is Marlin 1.1
2 i connect modemcu direct aux1 port  Rx-Tx and Tx-Rx 
this item https://www.amazon.com/OSOYOO-MKS-3D-printer-kit/dp/B01GJNBPPK
3 baud rate is same.
does it work? 
Let's not bother
Marilyn pins (for Tx Rx) D0 D1 I think. But it's different on the card scheme. It could stem from it
thank

pelitt, 2017-01-30:
Aux1 port is MKS TFT SCREEN port.
There is in the picture

luc-github, 2017-01-30:
I do not have this board as mine is Smoothieware  compatible not for Marlin so I cannot confirm connection but Marlin works

luc-github, 2017-01-30:
If you are able to share connecting pins, it would be nice - I could add it to wiki

pelitt, 2017-01-30:
ok but 
If you help me. I will try again.
We first need to find the Mks v1.5 circuit diagram.
We need to check if aux1 is available.

pelitt, 2017-01-30:
English is not enough you can research for the Mks v1.5 card. Producer etc.

luc-github, 2017-01-30:
MKS has a github with all data try here : https://github.com/makerbase-mks/Datasheet/blob/master/MKS%20Base2%20DataSheet.pdf

where are you from ?

luc-github, 2017-01-30:
Seems wiki was broken for MKS images I fixed it

pelitt, 2017-01-30:
i am from turkey. i show picture and is 
My card is a top version. But the same card. 

You need to ask the manufacturer. Aux1 (Mks TFT) port Rx Tx supports? 

Then i will try it again

luc-github, 2017-01-30:
did you tried the Aux1 D1 D2 pins ?

pelitt, 2017-01-30:
Yes i tried D1 and D2 but They are defined as Marlin D0 and D1

pelitt, 2017-01-30:
i will fix marlin

luc-github, 2017-01-30:
so what is the problem ? Rx D0 and Tx D1 is ok to use they are the same as usb connector

pelitt, 2017-01-30:
Should i use resistance with Nodemcu? Because ramp is 5v. But is Nodemcu  support 5v?

luc-github, 2017-01-30:
yes you should try first with tension divider

pelitt, 2017-01-30:
+5v>10k>3,3V>20K>Gnd
This will work?

luc-github, 2017-01-30:
little bit strong no ? but you can try - I use : https://github.com/luc-github/ESP3D/wiki/ESP8266-12E-F

pelitt, 2017-01-30:
Will I connect all the resistances?

pelitt, 2017-01-30:
i used for Rx  pin

luc-github, 2017-01-30:
just RX ones - for 5v and gnd NodeMCU has 5v input (https://github.com/luc-github/ESP3D/wiki/NodeMCU) so no need

pelitt, 2017-01-30:
Okey thanks i will try night and say you

pelitt, 2017-01-31:
İ try it and it is run thanks. but esp3d Server (192.168.0.1 ) online but not send data, it receiving data i dont understand why do this. So i try 3dfox Android app (with 192.168.0.1 ip adress and port number) and very stabil and good run with this app.
Connection is True but not use resistance because its support 5v. (dirrect connect ) thanks

luc-github, 2017-01-31:
Ok great - I close issue then as same as nonemcu and MKS-SBASE connection

closed 2017-01-31 by luc-github