luc-github/ESP3D #149: Add wireless discoverability and printing for Cura

probonopd, 2017-02-20:
Cura has wireless discoverability and printing for the Ultimaker 3 built in. Since the software is open source, it may be possible to use the same mechanism with ESP3D.

That would allow one to print over WLAN using Cura with 3rd-party printers using software such as https://github.com/luc-github/ESP3D on the printer side.

Looks like the printer needs to announce itself on Zeroconf with `ultimakersystem-xxxxxxxxxxxx._ultimaker._tcp.local.` and export a socket on __port 49674__.

I asked whether there is a description of the protocol on https://github.com/Ultimaker/UM3NetworkPrintingPlugin/issues/7.

luc-github, 2017-02-20:
it is available in latest cura ?

Looks like it is very expecting a JSON file with info ESP may not be aware of :
https://github.com/Ultimaker/Cura/blob/master/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevice.py#L55-L87
Looks like specific to ultimaker printers

probonopd, 2017-02-20:
Found the port when searching the web for `._ultimaker._tcp.local.`. Unfortunately I don't have more information on how this works, but I think this would be a _great_ addition to this project.

luc-github, 2017-02-20:
mdns (Zeroconf) is already implemented - 
hostname used in mdns is already configurable, so set ultimakersystem-xxxxxxxxxxxx._ultimaker._tcp.local. is just a setting if mandatory
data port is also configurable, so set as 49674 is just a setting

So need to dig in what information cura really expect and may be just copy a fake json file on SPIFFS first to see if it is working

luc-github, 2017-02-21:
looking at the code all is http using normal port
```
        self._api_version = "1"
        self._api_prefix = "/api/v" + self._api_version + "/"
```
So query like :
```
url = QUrl("http://" + self._address + self._api_prefix + "printer")
```
is actually http://192.168.1.15/api/v1/printer

so need to list all necessary handler and generate responses in json format accordingly

probonopd, 2017-02-21:
See the answer from @nallath on https://github.com/Ultimaker/UM3NetworkPrintingPlugin/issues/7#issuecomment-281273413. He suggests to write our own Cura plugin, similar to [OctoPrintPlugin](https://github.com/fieldOfView/OctoPrintPlugin).

Not sure why much more than the existing raw socket on port 8888 and the existing Zeroconf functionality would be needed.

But I still think it would be cool to have this work out of the box in Cura, without having to manually install a Cura plugin that is not shipped with Cura.

luc-github, 2017-02-21:
Well looking at UM3NetworkPrintingPlugin it do not do much more than what ESP3D do already

the only part that seems printer firmware dependant is the Active print part

[image]

I did not printed with Marlin yet as I only have a board system but smoothiware and repetier M27  command only display % of progress for smoothieware and number of byte processed vs total for repetier

printing time / job name and estimated time left are not available on printer FW

probonopd, 2017-02-21:
Well I'd pe happy already if I could print over WLAN directly from Cura. Should be cached on the SD card so that the PC doesn't have to run all the time. Everything else is secondary to me.

luc-github, 2017-02-21:
what I can see is that the pluggin only monitor the SD Card printing - or I missed something ?

Print over WLAN is like usb printing, it is not really as good  SD card printing IMHO, it  add more chance to fail ( computer fail, communication issue, etc...)

probonopd, 2017-02-21:
Hence I would like to send the file to the printer via WLAN, which should store it on SD card.

luc-github, 2017-02-21:
this feature is the upload command on ESP3D using M28  - Start SD write (M28 filename.g)/ M29  - Stop SD write but this is very slow, slow like hell actually....
the printer which has embedded wifi capability can  use proprietary protocole to speed up transfert
Another solution that I use today is a wifi SD card to tranfert file

probonopd, 2017-02-21:
Yes https://github.com/MarlinFirmware/Marlin/wiki/M28 is what I imagined.

luc-github, 2017-02-21:
every command send by this protocol is checked by board so transfert is not like a file but gcode command
a least for repetier and each one need a ack to send the next one
so slow like hell means very very very slow

probonopd, 2017-02-21:
OK, what about storing gcode to spiffs and stream it out from there?

luc-github, 2017-02-21:
SPIFFS is currently limited to 3M which means very small print - bigger flash is no really available and so far I did not saw a code ready to use it yet

luc-github, 2017-02-21:
the best solution would be : https://github.com/luc-github/ESP3D/issues/93
but it is not ready yet

probonopd, 2017-02-21:
Was about to propose that :-) :+1:

luc-github, 2017-02-21:
be patient ;)

nallath, 2017-02-22:
As I said before, you guys shouldn't "misuse" the um3 plugin. It kind of does what you need, but not exactly. The UM3 network print plugin is directly linked to working with an UM3, so it won't work that well with others (unless you change things).

THis is why I suggested to have a look at the octoprint plugin. It kind of does what you guys need.

probonopd, 2017-03-05:
[image]

If you are interested in sending G-Code from Cura to your 3D printer using an ESP8266, have a look at https://github.com/probonopd/WirelessPrinting/

closed 2017-03-05 by probonopd