luc-github/ESP3D #150: No information on 3D Printer

probonopd, 2017-02-21:
I am using a Renkforce RF100 with the latest released ESP3D version, 0.6.2.
The printer runs Marlin on a variant of the RAMPS board called HICTOP MPX.3.

I had to change the baud rate to 115,200 and now the communication is working insofar that the printer shows the IP address of the ESP on the display of the printer.

But when I go to the _3D Printer_ web page, it looks unpopulated, like this:

[image]

I can see that ESP printer sends periodically:
`M114\r\nM105\r\n`

And the printer responds with (this can also be seen on telnet port 8888):

```
ok
ok T:18.7 /0.0 T0:18.7 /0.0 @:0 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
ok
ok T:18.7 /0.0 T0:18.7 /0.0 @:0 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
ok
ok T:18.7 /0.0 T0:18.7 /0.0 @:0 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
```

Apart from that, there seems to be no communication and the status page is not populated.

Additional information:

```
CPU Frequency: 160 MHz
Free Memory: 26688 bytes
SDK Version: 1.5.4(baaeaebb)
Hostname: ESP_057D64

SSDP Protocol: Enabled

Network: 11g
Sleep mode: Modem
Boot version: 31
Baud rate: 115200
Web port:80
Data port:8888
```

probonopd, 2017-02-21:
Also clicking any icons seems to have no effect.

luc-github, 2017-02-21:
this is old version which had issue with marlin - I should do a realease soon but I need to wait esp core do a release too 
I suggest you to use esp core git version and my git version also which is now : 0.9.77

luc-github, 2017-02-23:
Hi may I know how you did connect your ESP to the HICTOP ?  -  this could be added to the wiki

probonopd, 2017-02-23:
> Hi may I know how you did connect your ESP to the HICTOP ? - this could be added to the wiki

Nothing special, I used the upper row of pins from the EXT-1 header, which appears to have the same pinout as on a RAMPS board. (Verify this information with your own board before connecting anything, I am not responsible if your board is different.)

luc-github, 2017-02-23:
Ok looks like HICTOP is a MKS Base V1.3  so it is already in wiki

probonopd, 2017-02-23:
Components don't look 100% the same but possibly the pinouts are similar.

luc-github, 2017-03-08:
If no update I close issue

closed 2017-03-08 by luc-github