luc-github/ESP3D #151: Send GCODE from SD card attached to ESP

probonopd, 2017-02-22:
I would like to attach a SD card to the ESP8266, be able to upload GCODE code to it, and have the ESP8266 send that GCODE over the serial connection to the 3D printer. I think that this should be pretty straightforward to implement, but the sending part must listen to "ok" and other such messages coming from the printer, and only send further code if they arrive.

luc-github, 2017-02-22:
straightforward to implement, well not so sure, ESP as host a working solution not really IMHO

ESP8266 is not a multitask MCU - some part can be emulate but not sure it will be Ok, when it could be Ok for CNC,  it won't be good for laser and FFF  printer.

GCODE need to be send to printer in regular flow to avoid any delay, why ? because if the flow is too slow - laser will stay too long in same place and burn too much, or in case of 3D printer melted filament due to inertia and gravity will continue to go out of extruder and it will result an over extrusion.
Flow will be slow down because ESP send query about position, temperature, handle answer to web page etc...   to sum up, unlike octoprint working on multitask OS, ESP is monotask,  the printing result will be terrible

luc-github, 2017-02-22:
I think ESP32 would be a better candidate for acting as host but SDK is still in early stage

probonopd, 2017-02-22:
Have you actually tried it? I think the 3D printer has some internal buffer for GCODE, so it is not as time-critical as you may think. The 3D printer asks for more GCODE well before it actually prints it.

luc-github, 2017-02-22:
No I did not because of what I explained above and based on my experience of ESP usage

Now you are free to try

probonopd, 2017-02-22:
I guess that's what I must do :-)
Do you know how to send GCODE?

luc-github, 2017-02-22:
have many way - simple gcode without answer check, with answer check, with line numbering + GCODE + check sum, text mode, have also binary mode like repetier, etc...

probonopd, 2017-02-22:
Do you know a good place where to learn this?

luc-github, 2017-02-22:
I would suggest you to start by a firmware flavour - seems you use marlin - start by marlin then when you know well the GCODE and the printer communication process of this FW, have a check of marlinkimbra to see if they work identicaly (they don't 😉 ) , then study another one like repetier or smoothiware or reprap or ultimaker to see the difference and the common part, this how I did for ESP3D originally was only for repetier then adapted for smoothy then marlin then marlinkimbra, I did not worked on reprap because duet3d has an embeded ESP so was not necessary 

I suggest you also to study how repetier host works or pronterface or cura, each one all or part of their source code available  so can be a good reference of communication process

probonopd, 2017-02-22:
Got it. So essentially port https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/USBPrinterOutputDevice.py to C, correct?

Have you done it somewhere else as part of ESP3D? or know any Arduino sketch that does it?

luc-github, 2017-02-22:
well ESP send gcode but simple ones, you must also handle the fact that the esp tx buffer is limited to 128 byte so you must not overload it - you may check what it is done for SD upload which is like acting has host but without printing

probonopd, 2017-02-23:
So it looks like essentially as a bare minimum we have to port this over to Arduino C, and possibly include more logic from the Cura code linked above.

[image]

Source: http://fightpc.blogspot.de/2016/08/g-code-over-wifi.html

probonopd, 2017-03-03:
I have successfully printed an object from SD card attached to the ESP8266. Here is my sketch:
https://gist.github.com/probonopd/16a148e227c286b69c296c0bf705d8bf

The only issue right now is that uploading the file to the SD card results in a reboot of the ESP8266 if the G-Code file is larger than a few lines. So I cheated and copied my G-Code (from Cura) to the SD card on the computer, to a file `cache.gco`. Then I attached the ESP8266 to the 3D printer and pressed "Start" in the HTTP "GUI".

Any clues as to why uploading to SD card does not work?

luc-github, 2017-03-04:
I do not use the hacks you use, so coud be a side effect of being "too" fast - try add `delay(0);` at the end of upload function before line 106 or try remove them to see if they are the root cause

luc-github, 2017-03-04:
And I do not use esp native SD library but sdfat one because I want lo support long file names

. and I regularly upload 10Mb files without issue and it is pretty fast , same speed as my wifi SD card  I think

probonopd, 2017-03-04:
Right now I am trying to get to a minimum viable implementation that has no dependencies outside of what Arduino 1.6.11 with esp 2.3.0 provide. I will use this to fiddle with uploading directly from Cura. Once I have that working, maybe you may consider to integrate it properly into ESP3D.

luc-github, 2017-03-04:
The project I mentioned is not yet public yet as it is linked to special hardware which is not owned by me
But code is same as upload to spiffs , also command ESP700 is doing same as you for macro
As no need robust sending but for long print  it is mandatory to check what printer ack for each command sent

probonopd, 2017-03-04:
Can you share just the part where you upload a file to SD card from your soon-to-be-released project, e.g., on https://gist.github.com/? Would help me a tremendous lot, thanks. I am really lost on this, trying for 2 days now.

luc-github, 2017-03-04:
1 - Did you do the changes I suggested ? it should be the solution looking at your code
2 - did you read the part I wrote it is same as SPIFFS ?
instead of open file to SPIFFS - open file to SD :
https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L2651-L2721
3 - check the minimal host from SPIFFS https://github.com/luc-github/ESP3D/blob/master/esp3d/command.cpp#L414-L444
Which is what you did but I read from SPIFFS instead of SD card , functions are 99% identicaln if you use native SD library
4 - to upload to SD card, if you do not believe me, check the example in espcore and use it to see  : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino

probonopd, 2017-03-04:
I started from `SDWebServer.ino` and it also does not work for me.

luc-github, 2017-03-04:
well it worked for me, and still work - was my base also for all upload (SD, Serial, SPIFFS)
Check also SPI speed if too slow or too fast you may have issue - it should match your SDCard reader
I usually use half speed

bubnikv, 2017-03-04:
The SD webserver crashes with a SD card containing a single long file or
folder name. It does not even boot. It took me a while to find out.

luc-github, 2017-03-04:
yes embeded library is limited - that is why I use SDFat library

probonopd, 2017-03-04:
Reformatting my 3 SD cards now using the [SD Card Formatter](https://www.sdcard.org/downloads/formatter_4/) from the SD Association.

probonopd, 2017-03-04:
Ah, now getting 

```
Initializing SD card...Wiring is correct and a card is present.

Card type: SD2
Could not find FAT16/FAT32 partition.
Make sure you've formatted the card
```

probonopd, 2017-03-04:
On to `sudo mkfs.vfat /dev/sdX -I`... so no partition...

probonopd, 2017-03-04:
Getting

```
Initializing SD card...initialization failed. Things to check:
* is a card inserted?
* is your wiring correct?
* did you change the chipSelect pin to match your shield or module?
```

now when trying to use the CardInfo sketch.

probonopd, 2017-03-05:
Got it working:
https://github.com/probonopd/WirelessPrinting/

closed 2017-03-05 by probonopd

ETE-Design, 2019-09-18:
@luc-github - Could you implement this, so it would be possible to send the file direct from Cura to the SD-Card on the Printer?

luc-github, 2019-09-18:
No need the project already exists as wrote above your post

ETE-Design, 2019-09-18:
Is it working with your ESP3D?

luc-github, 2019-09-18:
It is different project, fw is different, purpose are different

ETE-Design, 2019-09-18:
Okey, would just be nice if it was possible... But now I have ordered the ESP32 Cam... BTW How long are you in the development with the interface for that, so the Cam could work direct in the interface? And another good thing is that it has SD-Card integrated, so should be possible to use that as memory card to transfer files fast over WiFi :-)

luc-github, 2019-09-18:
2.1 allows wifi print from repetier host, pronterface and any host that allow ip connection. For Cura itself or even slic3r there is few chance if need third party support as I do not use them and no time to investigate on them. Also I have never been in favor of streaming, SD print is far safer IMHO.
Last but not least streaming support (ESP3D as Host) is not yet done but will be in idea of ESP3D touch panel solution for 3.0. Still also need to work on UI definition.
I am in vacations now until october so I work only on small snippet and small feature (I just finished littleFS support for example, 3.0 support Fat, SPIFFS on ESP32 and LittleFS, SPIFFS for ESP8266). I have frozen 2.1 already now just need time.

ETE-Design, 2019-09-19:
Okey, just reach out if you need help testing the SDCard and Cam part in the ESP32 Cam... Delivery time from China to me is about 6-8 Weeks so no hurry :-)

probonopd, 2019-09-21:
This is what we are doing in https://github.com/probonopd/WirelessPrinting:

Upload from PC to ESP 32 SD-Card (NOT 3D orinter SD card) -> Stream Gcode from ESP 32 SD-Card to printer

ETE-Design, 2019-09-21:
Looks great, but would like to have it all working on the ESP3D GUI, to have All features there... Is there a GUI in WirelessPrinting? Else it give No meaning to integrate the Cam I think...

probonopd, 2019-09-21:
>  Is there a GUI in WirelessPrinting?

Sure.

> would like to have it all working on the ESP3D GUI

Then the ESP3D project would need to integrate this type of functionality into ESP3D. I started WirelessPrinting precisely because this was not happening.

ETE-Design, 2019-09-21:
Where to find some info / screenshot about the GUI then?

probonopd, 2019-09-21:
I need help coding (= porting to the ESP32-Cam board), not just testing.