luc-github/ESP3D #156: Compilation error.

Aleks-MO, 2017-03-15:
Compilation error, I can not understand what's wrong....?
[image]

luc-github, 2017-03-15:
as you use master git version you must use latest git version of esp core to get all update from core

Aleks-MO, 2017-03-15:
It is possible with the link to the latest version, my version is 2.3.0, I did not find the other one.

luc-github, 2017-03-15:
read 
https://github.com/esp8266/Arduino/blob/master/README.md#using-git-version

unfortunatly AFIK the night build is not working yet so no way to use the board manager yet

luc-github, 2017-03-15:
the latest naight build seems ok so you may want to give a try  http://arduino.esp8266.com/nightly/package_esp8266com_index.json
but this is not same as latest git one - as it is from 22nd february

luc-github, 2017-03-15:
but I cannnot confirm will still be the case in future update

Aleks-MO, 2017-03-15:
Thanks, the compilation went fine! I'm very impressed.
[image]
I will conduct the test of 3D print ...

luc-github, 2017-03-15:
good , ^_^
I close issue then

closed 2017-03-15 by luc-github

Aleks-MO, 2017-03-15:
the question is closed.

denmorru, 2017-03-25:
Rename this issue to smart searchable error text:
esp3d:99: error: 'class HardwareSerial' has no member named 'setRxBufferSize'

luc-github, 2017-03-25:
hmm issue is closed and FAQ is clear IMHO

denmorru, 2017-03-25:
Clear? Maybe. But badly searchable.
"Error when compiling" is absolutely not smart topic (header) for issue. There are lots of different errors on compilation with different solutions.
Link to  http://arduino.esp8266.com/nightly/package_esp8266com_index.json is not found by text search in Wiki, in readme... Troubleshooting needs higher searchabilities. 
Look at this issue https://github.com/luc-github/ESP3D/issues/137 with proper error description "'class HardwareSerial' has no member named 'setRxBufferSize'" put to topic. I offered rename this issue like that.

luc-github, 2017-03-25:
well if your read what I wrote : night build are not working  yet
so that is why you hardly find some reference to : http://arduino.esp8266.com/nightly/package_esp8266com_index.json
even on esp core github