luc-github/ESP3D #157: "command_silent" query from UI seems not to do anything with "plain"

denmorru, 2017-03-28:
There is issue with file delete M30 command (sent by click on BIN icon button in UI): it does not seem to work at all in web interface of current version. 
UI makes confirmation popup dialog, and nothing else (the same problem I have about "Print" button).
Tested Marlin 1.0.2/RAMPS1.4. 
In case I send same "plain" from "Command" form, proper GCODE, it does deletion well.

debugger showed
`/command_silent?plain=M30%20COM.GCO `
and
` /command_silent?plain=M20`
both returned "OK"

esp3d FW: V0.9.77 default GIT in ESP-12F device
Compiled with ESP8266 package 2.4.0., but UI shows SDK Version: 2.0.0(656edbf),
Arduino IDE 1.6.8

luc-github, 2017-03-29:
I cannot reproduce issue on mega/arm + marlin 1.1 rc08 + GLCD
I even use same name file COM.CGO
and I can upload - print and delete 
It is may be browser problem that I am not aware of or a baud rate not matching on ESP and board

luc-github, 2017-03-29:
it may related to https://github.com/luc-github/ESP3D/issues/158
if have error browser may hold all requests but I cannot reproduce

denmorru, 2017-03-29:
Updated command.cpp from FW_VERSION "0.9.78"  , no change. So, that issue is not this error case.
Browser  = Firefox 51.0.1, Baud 250000,  IE browser cannot even load this page /PRINTER content...
there is no problem in upload, problem = no refresh list files after upload (automatically).

denmorru, 2017-03-29:
I fixed IE issue with this modification of printer.tpl:
[printer.zip](https://github.com/luc-github/ESP3D/files/878839/printer.zip)
(I do not have push access to this repository, so, replace yourself)

luc-github, 2017-03-29:
Marlin support only shortnames the display is same as doing M20
because M28 / M29 do not upload in subdirectory

I fixed also IE due to javascript compatibility - was fixed on my private repo but I did not pushed on github my bad
 https://github.com/luc-github/ESP3D/commit/38d86c5f8ab312813defdb34d8e25fb4c0153126

luc-github, 2017-03-29:
I tested latest commit on Firefox / chrome / Internet explorer 
Print button / delete buttons are working on my side

luc-github, 2017-03-29:
well javascript support default parameter - but not all browser are following the https://developer.mozilla.org/en/docs/Web/JavaScript/Reference/Functions/default_parameters

denmorru, 2017-03-29:
Main problem is what happens after `onclick="Delete('CAT.GCO','line3')"` ,
while function `void handle_web_command_silent()` running, which returned "ok", but deletion not happened.

Another problem is that "Error" section in /PRINTER page is always empty, so I have no data about LOG events happened in that function...

luc-github, 2017-03-29:
If printer board do not send any line stating with `Error:` it wont display anything on Error log

does `M30 CAT.GCO` give any output in command part?

to debug I usually add output message using console.log("Debug message"); it will appear in developer console of your browser

did you changed anything in code like `serialsdmode` value?

you can add debug message here
https://github.com/luc-github/ESP3D/blob/master/esp3d/data/printer.tpl#L771-L783

and here 
https://github.com/luc-github/ESP3D/blob/master/esp3d/data/printer.tpl#L388-L404

luc-github, 2017-03-29:
LOG  is a function that need to be enabled in FW you must define what output for the LOG
https://github.com/luc-github/ESP3D/blob/master/esp3d/config.h#L94-L99
->do not output LOG to serial if your board/printer is connected it will mess up with all communications

denmorru, 2017-03-29:
M20 mentioned above is sent immediately after M30, before M30 returned result.
In case M20 "ok" is really necessary, this gcode must wait for M30 result: implement synchronious AJAX

luc-github, 2017-03-29:
yes this UI is not really asynchronous that is why I rewriting all 
anyway commands are queued by printer so should not be an issue
and this should not prevent M30 to be executed by printer

denmorru, 2017-03-29:
topic problem is in webinterface.cpp
You should decode URL string before `cmd.trim();`
because string has `%20`

luc-github, 2017-03-29:
>It needs to decode URL string before cmd.trim();
>because string has %20
this is done by webserver parsing arguments automaticaly

>With new empty SD card Serial returned me Error:Line Number is not Last Line Number+1, Last Line: 0
this may due the fact you use an host in same time as ESP, host (at least repetier host) use line number with command- ESP does not

denmorru, 2017-03-29:
No, there was no any other client connected to printer , only ESP3D with Serial0.
Another problem: ESP3D does not listen for GCODE result from Serial , 
so, we cannot find mentioned error in LOG.
ESP3D only do               `  Serial.println(cmd);    ` and "forget" about this , returning "ok". 
But, in this case, result is far from "ok". Result is: "Error:Line Number is not Last Line Number+1, Last Line: 0".
So, this is bug of blindness. 
ESP3D should get result from port, parse, decide the status - ad inform web UI user correspondingly.

luc-github, 2017-03-29:
that is why I have asked : does M30 CAT.GCO give any output in command part?
why your marlin FW says ?
`Error:Line Number is not Last Line Number+1, Last Line: 0` 
when your send M30 CAT.GCO

luc-github, 2017-03-29:
but there is a glich as Error:  line should be catched and so should be displayed  in Error log

denmorru, 2017-03-29:
Why Marlin return error I will research on their side, but here we talk about proper handling of this erroneous result by ESP3D, which not happened. This is my log
[esp3d_log.txt](https://github.com/luc-github/ESP3D/files/880115/esp3d_log.txt)
,

luc-github, 2017-03-29:
Yes normaly with silent command no answer is catched 
so everything in serial should be catched by normal process
can you get the complete serial output when you type the command ?

denmorru, 2017-03-30:
So, I found that:
file names are shown wrong in web interface list: they are CAPITALIZED,
but real files named in lowercase.
So, `M30 cat.gco` will do the job, 
but `M30 CAT.GCO` will fail. Bin button sends as shown in UI = wrong.

luc-github, 2017-03-30:
well the issue is in marlin FW 
as you see M20 command show capitalized char - this is done by marlin not ESP

luc-github, 2017-03-30:
this can be workaround but it means Marlin 1.02 is not compatible with 1.1 rc8

denmorru, 2017-03-30:
Updated to Marlin 1.1.0-RC8 version.
Uppercase problem is still there, 
but deletion makes no error any more, and deletes file as expected, both with uppercase filename in M30 and lowercase name. So, they fixed it badly, instead of turning off decapitalization, they turned on case-insensitive mode.

denmorru, 2017-03-30:
So, how are you going to fix subfolder issue? Long filenames? Have you reported bug to Marlin?

luc-github, 2017-03-30:
You say 
>Uppercase problem is still there,
and
>deletion makes no error any more, and deletes file as expected,
this is confusing 

I have no plan to push any ticket as Marlin / Marlin kimbra / Repetier FW all works in same way - this is a limitation that I live with

Blue-Marlin, 2017-03-30:
https://en.wikipedia.org/wiki/8.3_filename

luc-github, 2017-03-30:
I did not tested latests rc, so seems I am wrong for Marlin 1.01rc8 and repetier
 print can be done in subfolders -  no idea why it was not working when I tested before - so will add subfolders support later

luc-github, 2017-03-30:
Actually Using M20
Repetier display files / folders and files in subfolders 
so UI juste need to hide folders as standalone => UI problem
[image]
Deletion M30 seems not working on subfolder files`
`M30 MYFOLDER/M3.G` failed and crash the filesystem => Repetier issue (TBA)

Marlin just list filename with path when have a G inside but ESP3D FW set sub files as folders when there is no folders but only files => ESP3D FW problem
[image]
M30 deletion seems working on subfolder files

Under smoothieware it is more complexe:
M20 list only root files and root folders on onboard SD
M23/M24 failed on sub folder files
so need to use shell instead: `ls /sd` for onboard SD and `ls /ext` for screen SD or any subfolder
like `ls /sd/luc/'
and use play '<path to file>` to print 
to delete M30/rm is working only on onboard SD not on /ext => FW limitation per smoothieware developer

luc-github, 2017-03-30:
For MarlinKimbra 
List Files, Folders and subfiles and subfolders like repetier but all with long name support

M30 / M23/M24 works with sub folders

luc-github, 2017-03-31:
will take time to revert to Marlin 1.02 to check why Error is not catched by normal flow 
>but there is a glich as Error: line should be catched and so should be displayed in Error log

luc-github, 2017-03-31:
when I try to delete a file 
in /luc/m3.g 
which display /LUC/M3.G when using M20
I type `M30  /LUC/M3.G` and got Ok from Marlin but file is still there which is observed behavior
but I cannot see the :
`Error:Line Number is not Last Line Number+1, Last Line: 0` 
message 
Any step by step reproduce way ?

denmorru, 2017-03-31:
`Marlin_main.cpp` shows M30 usage with lowercase: 
Delete file from SD: "M30 /path/file.gco".
```
if (card.cardOK) {
      card.closefile();
      card.removeFile(current_command_args);
}
```
I would say error message was caught sometimes, in web UI, sometimes not.

luc-github, 2017-03-31:
well if I cannot reproduce issue - I won't be able to figure out to fix it
I did 5 times the command and never got such message and I do not use the webUI  but direct Serial output of mega so I suspect a communication issue between your board and ESP bringing some garbage and misinterpreted as command with line number

denmorru, 2017-04-01:
What exactly do you want to fix?
I believe this is correct way for ESP3D: send in M30 exactly the same filename with path, which you get with M20. 
The only useful fix pending I mentioned above: add LOG level, when all serial incoming data, even if it were "silent" command, will be saved to file.

luc-github, 2017-04-01:
you said you have error message :
`Error:Line Number is not Last Line Number+1, Last Line: 0`
and it does not appear in log file when it should 
`dd LOG level, when all serial incoming data, even if it were "silent" command, will be saved to file.` this is already the case, all lines go to log so there is nothing to add in log level

so I try to reproduce your issue to see why it is happening - all incoming data are logged in debug level but your error message is not  
because if error message do not display in debug which catch all - no error message can be catched in any way

denmorru, 2017-04-01:
[M3STO_v4.zip](https://github.com/luc-github/ESP3D/files/887537/M3STO_v4.zip)
This is exacly that Marlin tested with error. I used UART0 (AUX-1 inner pin row). Because this port paralleled with Mega2560 USB chip, you can send the same command from PC Serial monitor and compare.

luc-github, 2017-04-22:
Ok if error is catched so actually it is Ok 
if no issue with latest Marlin then I close issue 
Others issues have a separate tracker now 
New UI / FW impementation which is fully catching all response should be ready early next month

closed 2017-04-22 by luc-github