luc-github/ESP3D #160: Join web user interface tabs "AP" and "Station" into one

denmorru, 2017-03-29:
Pending rework of user interface would be offered here.
Network connectivity settings are not often used, and they are minor in ESP3D usage time. 
So, move those network tabs into one small tab in the end of the menu list.
I suggest ESP3D software to be targeted to 3D printer functions, which must go on first tab, default tab.
Network connectivity is well implemented in other public free libraries, like <WiFiManager.h>, so, they must not take time from developers in this project, until proved opposite.

luc-github, 2017-03-29:
current devt fw does not have anymore the /HOME /PRINTER /etc page but only one 
Wifi Config is one tab - Printer config is one tab - Camera is one tab - printer page is default tab
  
[image]
[image]
Printer tab is inspired from imrahil UI for smoothieware
[image]

 and I do a second one with repetier host UI 
[image]

SPIFFS and web update are just a dialog box launched from menu when needed

So all you describe is already done from ESP FW point of view - I am still working on UI part to get good asynch reactivity

closed 2017-03-30 by luc-github

denmorru, 2017-03-30:
So, where is actual version of web user interface for test?
https://github.com/luc-github/ESP3D/tree/master/UI is this working with current firmware?
So, tabs will be:
1) Printer control page, including SD-card browser
2) Camera and system information, including ex-HOME
3) Printer and system config, including SPIFFS browser , firmware update, ex-EXTRA
4) WiFi config 
?

denmorru, 2017-03-30:
One more suggest on how to arrange master branch:
move https://github.com/luc-github/ESP3D/tree/master/docs  to https://github.com/luc-github/ESP3D/wiki
So, targeting to clean distributive

luc-github, 2017-03-30:
No it won't be compatible as no more template engine - which break everything in current UI but save some memory to add more query features - so I won't share before FW has finished UI to work with