luc-github/ESP3D #161: ESP3D not starting WiFi on ESP-01S module

denmorru, 2017-03-31:
Upload ESP3D firmware to ESP-01S module (4MB flash memory). Upload both ESP3D.ino.bin and esp3d.spiffs.bin finished successfully. But it does not start, even Serial.

In case upload to this module some other firmware, like https://github.com/denmorru/ESP8266DM_SM (using <WiFiManager.h>) - it works normally.

luc-github, 2017-03-31:
I do know this module where did you bought it?

openhardwarecoza, 2017-03-31:
Module is fine, i've used it with esp3d. Check your spiffs settings in tools, and NB validate that with the checkspiffs sketch from esp8266, examples. If the spiffs setup in tools doesnt match the module exactly, this symptom happens.

openhardwarecoza, 2017-03-31:
See https://github.com/luc-github/ESP3D/wiki/Flash-Size

denmorru, 2017-03-31:
This does not matter, if you choose 128KB SPIFFS or 512KB on ESP-01 (1MB) both worked the same correct for me. When you have 4MB (like ESP-01S), it is worth setting 3MB SPIFFS for ESP3D.
ESP3D.ino.bin has size: 05b360 (364KB).

Useful test with CheckFlashConfig.ino happened, result:
```
Flash real id:   001440E0
Flash real size: 1048576
Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration wrong!
```
So, this looks like there is readable memory limit 1MB. It was just misinformation from module seller that it has 4MB. After I set 512KB+512KB ESP3D is startable with WiFi.
Exception: CAPTIVE_PORTAL_FEATURE

luc-github, 2017-03-31:
did you check the FAQ for the SPIFFs size on 1MB flash?

luc-github, 2017-03-31:
for what I can see ESP01-S is not 4MB but 1MB flash size 
http://www.electrodragon.com/product/esp-01-esp8266-wifi-module/

this is actually the new generation of ESP01 the black one - with 1MB flash, the older one has only 512K of flash - and yes so far no issue if you follow the SPIFFS size I have described in FAQ 

So if you upload using using wrong partition it won't work - IDE do not prevent that unfortunatly

So I close issue

closed 2017-03-31 by luc-github

denmorru, 2017-03-31:
with CAPTIVE_PORTAL_FEATURE ESP3D is not working on ESP-01 1MB, the same as https://github.com/luc-github/ESP3D/issues/42 - you close issue when problem is not solved

luc-github, 2017-03-31:
1 - CAPTIVE portal is ESP core library it is just added 
2 - Captive portal is not the issue described in ticket but wrong setting usage so yes I close issue 
3 - Open a new issue and please give as much information how to reproduce this 'CAPTIVE PORTAL' issue  and not another one as I cannot reproduce it on my side