luc-github/ESP3D #165: serial communication

semvis123, 2017-04-30:
Hi I have a problem with the rx and the tx pins,
I am using a wemos d1
this is what I got
above the Arduino UNO serial monitor with serial reading sketch uploaded hooked up to the wemos d1
and below the wemos d1 serial monitor
[image]
here are my settings
[image]

luc-github, 2017-04-30:
I do not see any communication error 
the garbage you see on begining is bigger than on my ESP but it come from board itseft and some from bootloader - currently there is nothing to prevent this
Fortunatly this happen at cold boot when printer do not listen yet the serial port so it is ok 
when doing restart I swap the serial output to avoid this garbage to go to printer - but at cold boot nothing can be done

semvis123, 2017-04-30:
the printer is also not responding on everyting
status:offline and is not chancing to online

are there other boot methots like cold boot?

luc-github, 2017-04-30:
well you did not mentionned this before
what is your printer Board? 
how do you connect esp to printer?
are printer and esp using same baud rate?
did you invert rx and tx to see if it is a wire issue ?

semvis123, 2017-04-30:
my printer board is a mks _gen v1.0
i used the aux 1 port __#__#__rx__tx__
                                   __#__#__#__#__
i got power trough usb and a power adapter 5v
my printer uses 250000 baud rate and the esp 250000
yes i inverted the rx and tx cant see any difference

luc-github, 2017-04-30:
esp and board should share the ground
why you do not use the 5v of aux 1?

semvis123, 2017-04-30:
i will test it .
my printer board uses marlin

semvis123, 2017-04-30:
is there a 3.3v port on the mks gen_l v1.0?
https://ae01.alicdn.com/kf/HTB1mODbQXXXXXbGXFXXq6xXFXXX2/220234939/HTB1mODbQXXXXXbGXFXXq6xXFXXX2.jpg

semvis123, 2017-04-30:
5v is too much i think

luc-github, 2017-04-30:
you can use the 5v of D1 or use a tension divider like https://github.com/luc-github/ESP3D/wiki/ESP8266-12E-F

semvis123, 2017-04-30:
okay sharing the ground did the trick thanks the problem is solved

semvis123, 2017-04-30:
thanks for your fast comments

luc-github, 2017-04-30:
Good - have fun - I close issue then

closed 2017-04-30 by luc-github