luc-github/ESP3D #166: PlatformIO compilation issue

Sparkxxx, 2017-05-03:
Hi Luc,

I got this error 
src\webinterface.cpp: In function 'void handle_web_interface_home()':
src\webinterface.cpp:874:21: error: cannot convert 'station_info::<anonymous struct>' to 'station_info*' in assignment
station = station->next; 

when trying to compile with Platform IO, any idea what could be wrong? Both failed for staging and normal version of espressif8266

[05/03/17 12:03:56] Processing esp12e (platform: espressif8266_stage; upload_speed: 256000, # Using library Name, ;EspSaveCrash; board: esp12e; board_f_cpu: 160000000L; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Converting esp3d.ino
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph

|-- <DNSServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266SSDP>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <EEPROM> v1.0
|-- <ESP8266NetBIOS> v1.0
|   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs\esp12e\src\esp3d.ino.o
Compiling .pioenvs\esp12e\src\webinterface.o
Compiling .pioenvs\esp12e\FrameworkArduino\IPAddress.o
Compiling .pioenvs\esp12e\FrameworkArduino\MD5Builder.o
Compiling .pioenvs\esp12e\FrameworkArduino\Print.o
Compiling .pioenvs\esp12e\FrameworkArduino\Schedule.o
Compiling .pioenvs\esp12e\FrameworkArduino\Stream.o
Compiling .pioenvs\esp12e\FrameworkArduino\StreamString.o
Compiling .pioenvs\esp12e\FrameworkArduino\Tone.o
Compiling .pioenvs\esp12e\FrameworkArduino\Updater.o
Compiling .pioenvs\esp12e\FrameworkArduino\WMath.o
src\webinterface.cpp: In function 'void handle_web_interface_home()':
src\webinterface.cpp:874:21: error: cannot convert 'station_info::<anonymous struct>' to 'station_info*' in assignment
station = station->next;
^
Compiling .pioenvs\esp12e\FrameworkArduino\WString.o
Compiling .pioenvs\esp12e\FrameworkArduino\abi.o
Compiling .pioenvs\esp12e\FrameworkArduino\base64.o
Compiling .pioenvs\esp12e\FrameworkArduino\cbuf.o
Compiling .pioenvs\esp12e\FrameworkArduino\cont.o
*** [.pioenvs\esp12e\src\webinterface.o] Error 1
Compiling .pioenvs\esp12e\FrameworkArduino\cont_util.o
 [ERROR] Took 2.14 seconds

Thank you.

luc-github, 2017-05-03:
Well I do not use PlatformIO so not sure what esp core version they use
 but if you do not use git version of esp core you cannot use my master but the released version only as stated in readme:

Stable version:
Arduino ide 1.6.5 with stable 2.0.0 from ESP8266, please use https://github.com/luc-github/ESP3D/releases/tag/v0.5.1
Arduino ide 1.6.8 with stable 2.2.0 from ESP8266, please use https://github.com/luc-github/ESP3D/releases/tag/v0.6.2

Development version:
Arduino ide 1.8.0 with git from ESP8266 : 

My master use a lot of change present in Esp core master which is unfortunatly not yet released

Sparkxxx, 2017-05-05:
Solved it, posting for others who face the same issues.

In platformio.ini instead of platform = espressif8266 use platform = espressif8266_stage
In the ide at PlatformIO menu go and click on Update platforms, packages and libraries.

luc-github, 2017-05-05:
Ok good - thanks - I close issue then ^_^

closed 2017-05-05 by luc-github