luc-github/ESP3D #167: Strange movement of Prusa i4

DemoN3579, 2017-06-20:
Hello
I've connect ESP12F to my Prusa i4 everything is OK. 
I can connect to the web interface, printer is online an a can move axis and control temperature.
But then I connect to the printer from Pronterface to the printer through WiFi an try to print model. Printer do strange movement and try to print something else but nit my model.
Gcode file that I try to print is OK. I can print my model if connect to the printer through USB cord. 
Version of firmware for the ESP is v0.6.2
Version of printer firmware (that printer answer on M115 command) in  sunhokey PanTech 1.0

So what I should do with this?

luc-github, 2017-06-20:
Hi what is the fw based ? Marlin ?
What do you mean by it try to print something else ?

DemoN3579, 2017-06-20:
FW is Marlin based, but I'm not sure
"printing something else" mean that printer execute mix of g-commands from my model and random movement on X and Y axis
And yes, I have the web interface open in same time as using pronterface by wifi
May be this is the problem?

luc-github, 2017-06-20:
well I guess it could - 
imagine 2 flows of commands are sent to serial one from WEB UI and  one from TCP/IP pronterface, they can mess up the printer
if you print from pronterface better not use web UI which is more  for monitoring SD card print which is the safer IMHO

you may try the development version which use another UI and give more time of execution for tcpip pipe

DemoN3579, 2017-06-21:
I try to print using only connection from pronterface not using WEB UI at same time. 
The problem still exists
I try to use firmware for the ESP from the master git branch. The same result.
I've share to video 
Then I try to print using usb cable https://cloud.centre-it.com/cloud/index.php/s/sA9IEUYjoVctlhx/download
And then I connect trough ESP  https://cloud.centre-it.com/cloud/index.php/s/pScoegh6TXDIV5Y/download

Besides I find that I can't print model from SD card using WEB UI. I just push button near filename, but nothing happened.

May be this is problem of capability with printer firmware? Or incorrect connection between printer and ESP?
Any idea for diagnostic?

luc-github, 2017-06-21:
Hi about SD print from UI  current master has some limitation for Marlin based on old revision , there are some issue due to upper/lower case filename 

Please try the devt branch (not the master) which is far better and fix such issue, depending on your printer FW but need also to update UI  as I mentioned earlier

Do you have a link pointing  to your printer FW sources ? to identify what kind of FW it is. as you must specify it to make it works properly

About direct connection pronterface / printer issue seems it do not get correct command, you should have a setting to echo every command the printer receive, you should enable it to see them in pronterface log - it could help to understand by comparing what pronterface send and what printer receive it

DemoN3579, 2017-06-21:
I've upgrade my printer firmware to Marlin v1.1.3 the result is the same...
I still able to print through the USB cord but can't print through ESP
It seems for me taht the problem is with the connection ESP to the printer board. My board is MKS base v1.5
How to connect ESP right?

And that about "you should have a setting to echo every command the printer receive" how I can do this?

luc-github, 2017-06-21:
did you updated ESP FW and UI ? - Marlin 1.1.3 is fine on this one

to connect the wiki said it is MKS for smoothieware but should work also on Marlin version : https://github.com/luc-github/ESP3D/wiki/MKS-Smoothieware-compatible

what ESP are you using ?

DemoN3579, 2017-06-22:
yes, I've try both Master and Devt branch with Marline 1.1.3
The result is the same
In the Devt WEB UI I don't know how to print from SD it look like this https://cloud.centre-it.com/cloud/index.php/s/sx6k4GKjo5iewEi/download There is no button to print model near the file name

And abount connection ESP to the printer. There is inforamtion abount MKS SBASE but I have MKS Base v1.5 it is look like this https://cloud.centre-it.com/cloud/index.php/s/QCdtPPXeqjgR59Y/download
For this moment ESP connected to the pin D2 and D1 of MKS TFT LCD slot

luc-github, 2017-06-22:
it seems it does not see the GCO extension because there is number in name and so does not allow it to be printed weird

can you do a screen shot of esp FW settings in UI ? I want to see if target FW is correct

Yes Aux1 is actually the TFT slot

DemoN3579, 2017-06-22:
In the TFT Slot two pair of data pins D1,D2 and D57,D58 with of them correct?

luc-github, 2017-06-22:
Should be D1 D2 for serial, but your connection seems Ok as M115 is working

luc-github, 2017-06-22:
can you do a screen shot of esp FW settings in UI ? I want to see if target FW is correct

also can you show the result of M20 command ?

DemoN3579, 2017-06-22:
Here is screenshots of ESP settings
https://cloud.centre-it.com/cloud/index.php/s/OmOUlt7oPaPnKNh/download

luc-github, 2017-06-22:
I see your setting for Fw is wrong :
[image]

it is for repetier - change it for marlin - it should solve your issue

DemoN3579, 2017-06-22:
I've change target FW to Merlin
M20 say:

M20
Begin file list
TEST~1.GCO 3328106
End file list

luc-github, 2017-06-22:
Ok this is weird my version of Marlin does not display file size - but I am 1.1.0Rc8  and I just see https://github.com/MarlinFirmware/Marlin/pull/6325

I will update code to support this new feature tomorrow - s I need to update my test machine first with latest Marlin  FW

Thanks for the catch

luc-github, 2017-06-22:
I have created https://github.com/luc-github/ESP3D-WEBUI/issues/3 to track issue as it is in UI not ESP FW itself

luc-github, 2017-06-23:
please update https://github.com/luc-github/ESP3D-WEBUI/blob/master/index.html.gz using web UI 
issue is fixed - now Marlin 1.1.3 can print from UI

DemoN3579, 2017-06-23:
I've update index.html.gz
Now I can print correct model from SD card through WEB UI
But then I print model from pronterface through ESP the situation is the same

luc-github, 2017-06-24:
what is the log of M111 S1 when you print compare to actual commands ?
what settings do you use in pronterface for communication ?

DemoN3579, 2017-06-24:
It's look like pronterface push stream of Gcode command to the EPS
And EPS didn't have time even to answer
Log here https://cloud.centre-it.com/cloud/index.php/s/N6X4OfVD81Fj68q/download

Besides, ESP work extremely slow when I try to upload file to the SD
I need about 5 min tu upload file about 3 Mb

luc-github, 2017-06-24:
what settings do you use in pronterface for communication ?

>Besides, ESP work extremely slow when I try to upload file to the SD
>I need about 5 min tu upload file about 3 Mb

Actually the  serial file transfert protocole is slow like hell - it is not ESP - I have added this feature for compatibility but  I use Wifi SD card as it is faster

DemoN3579, 2017-06-24:
There are not a lot of setting for prontorface communication
just connect string 192.168.7.234:8888
and port speed 115200

luc-github, 2017-06-24:
did you uncheck the TCP streaming mode ? seems this one speed the communication but may overload esp
did you tried to decrease baud rate ?

DemoN3579, 2017-06-24:
Yes, I try decrease to 2400
The same result. The log is here https://cloud.centre-it.com/cloud/index.php/s/U5DT568SOBPZTC1/download

What do you mean to "uncheck the TCP streaming mode"?

DemoN3579, 2017-06-24:
Hm.... I didn't have this options :)

DemoN3579, 2017-06-24:
I've upgraded the pronterface to 2014.08.01
Now it looks that everything works properly

luc-github, 2017-06-24:
good - happy printing then 😄

luc-github, 2017-06-29:
I close issue if no update

closed 2017-06-29 by luc-github