luc-github/ESP3D #168: Feedback welcome on Devt branch

luc-github, 2017-07-16:


benji720, 2017-07-17:
Hi - I've done a few changes to incorporate % complete and planning IFTTT integration.  I've got no idea how to merge into GitHub (I don't normally use it).  Do you have that in your branch?  Do you want it (% complete is trivial - just issue a M27 every so often and include the output in the STATUS JSON)?

I was also going to update the web interface - I think it can look much nicer.

luc-github, 2017-07-17:
Yes ESP3D-WEBUI will become the new UI but need Devt branch to work - it will be merged in master end of this month

Sparkxxx, 2017-08-22:
Hello Mr. Lebosse, 

While testing and playing with the development version I enabled AUTHENTICATION_FEATURE in config.h and since then the web interface is just displaying Connection failed! is your FW correct? 
Digging deper in the code I stumbled on the browser console log that sais 
Error 403 : Not allowed, log in first! 
so my assumption is that something wrong is happening since there's no way I can login at this step since I am just loading for the first time.
Disabling the authentification renders the page normally.

luc-github, 2017-08-22:
Hi Thank you for this feedback - yes, you are correct, current UI does not have any UI for authentication but FW itself is in theory ready -  I will implement soon a login page and a change password page for user and admin

I am currently working on first setup steps pages then will work on authentication, I wanted to get full process working without authentication before enabling it

apphead, 2017-08-31:
Using devt version, when these feature are uncommented in config.h, they throw errors:
- AUTHENTICATION_FEATURE: as reported above by Sparkxxx
- SDCARD_FEATURE: compilation fails due to
....arduino_build_743302\sketch\config.h:169:19: fatal error: SdFat.h: No such file or directory
 #include "SdFat.h"
File is present in _[ESP3D-devt]\libraries\SdFat\SdFat\src\_

luc-github, 2017-08-31:
Thanks for the feedback 

SDCARD_FEATURE is not available yet, that is why it is currently commented
to get SdFat.h you need to install the library manually https://www.arduino.cc/en/Guide/Libraries#toc5
But I did not documented it yet as SDCARD_FEATURE  is not available - but will be available soon (code is ready but not public)

luc-github, 2017-09-06:
Authentication is now implemented in FW and UI : https://github.com/luc-github/ESP3D/commit/798f6ea149c8e5877891531a94d352765646d400

eduard-sukharev, 2017-09-27:
Seems like Sd card code is here now: i've tried building from master and overall firmware looks and works ok, but uploading files to sd card works wrong. I created a simple gcode file with only G28 command in it and tried to upload it, but the printer behaves as if it was given direct gcode commands, three times in a row! It homes all axes, then homes them again and then again.
Is it a bug and should i report it as a separate issue?

It's Marlin fw, fresh esp8266 and esp3d.

luc-github, 2017-09-27:
you mean serial sd upload ?

eduard-sukharev, 2017-09-27:
@luc-github i mean the ability to transfer files to printer's SD card over wifi.
Here's the settings page of my FW:
[image]

My setup is following: I have MKS Gen v1.4 board (RAMPS 1.4 compatible) with attached ULTIPANEL (which is also an MKS MINI12864 LCD screen with built-in SD card reader). The esp is connected to printer via Tx/Rx, with no additional commutation between Esp module and SD card reader.

luc-github, 2017-09-27:
Ok So yes it is Serial file transfert
Upload send file to ESP which foward to RAMPS  by serial which write data to the  MKS MINI12864 LCD screen

Does content of SD card is well displayed in UI before upload ?

eduard-sukharev, 2017-09-27:
Well, kind of. Filenames are in 8.3 DOS-like format with tildas, but i could tell them apart.

luc-github, 2017-09-27:
yes 8.3 format is the marlin way to display files, you will have same output when doing M20 command in command panel

luc-github, 2017-09-27:
what file name is used for upload ?

eduard-sukharev, 2017-09-27:
The filename is 'homing.gcode', the contents is single line "G28" and a newline at the end.

eduard-sukharev, 2017-09-27:
Is there any debug info i could provide? Or probably some simple firmware sketch to replicate the file uploading to marlin? Is there something in marlin that should be enabled for serial sd card upload?

eduard-sukharev, 2017-09-27:
I've tested a bit more with different file contents. It does not depend on the new line characters: neither CRLF, nor CR, nor LF work. Comments in gcode lines are also play no role.
The "triple" execution is only for G28. The fllowing file contents:
```
G90 ; use absolute positioning for the XYZ axes
G1 X100

G28 ;

```
makes my printer move the caret once, but then it homes and rehomes and rehomes and rehomes.

After everything, LCD screen shows "SD upload failed".

luc-github, 2017-09-28:
I suspect the filename - I cannot reproduce now asI am travelling ut this this my guess

the filename is not 8.3 so the command to start writing to SD including filename is not taken in account 
and the content of file is no more see as content but as commands itself.

please try renaming the file in 8.3 like homing.gco and try upload

eduard-sukharev, 2017-09-28:
The naming of files indeed matters. I've tried renaming a file into 'homing.gco' and 'homing.g' and 'home.g' as you proposed and all uploaded well!

But now i've just found out that i can't start printing of any of the files on sd card!
I have both non-8.3 files and 8.3 files on sd card and when i click the priner button near them the printer does nothing, lcd shows nothing and "Information" logs show following:
```
:Now fresh file: /HOME.G 21
:Now fresh file: /HOMING.GCO 21
:Now fresh file: /HOME-8.GCO 21
:Now fresh file: /HOMING.GCO 21
```
Each line resembles one button click, one print attempt. Any ideas what's wrong?

eduard-sukharev, 2017-09-28:
Is there any reason why filesize is not split away from filename for Marlin firnware? For repetier and others you seem to split the contents of json by space character, while maelin filenames are left intact with numbers after space. I suspect that might be the reason for prints not starting.

luc-github, 2017-09-28:
humm this has been fixed in master if I remember well, I may missed to update devt, I will check tonight and add a check for file name lenght

luc-github, 2017-09-28:
yes issue is in webui and seems I wiped the fix...
21cb4bb701afa95e522c5a91d30b9233d0eefc40

eduard-sukharev, 2017-09-28:
ah, so to fix this i have to upload web-ui files with current esp3d fw? ok then.
How often do you update UI files in this repo from WEB-UI repo?
And one more question - why file sizes are not split from filenames for marlin?

luc-github, 2017-09-28:
The fix is to upload the new espwebui when I put fix back
I update webui when there are issues but I am currentling traveling so ...
Before Marlin do not have file size, then they introduced it like others FW but I do not use Marlin so I do not follow all update as soon they are out

eduard-sukharev, 2017-09-28:
hm, so you didn't wipe the fix. it's there. The problem seems to be that i haven't used espwebui files at all and just uploaded esp3d files as is. Will try espwebui this evening and will give the feedback.
Btw, regarding espwebui - is installation all about uploading `index.html.gz` and that's it? Should no other files be uploaded? No js files? no css or other files?

luc-github, 2017-09-28:
yes that is all index.html.gz is the only needed file for devt branch

luc-github, 2017-09-28:
please be sure you use devt branch : https://github.com/luc-github/ESP3D/tree/devt/esp3d

mzeve, 2017-10-04:
Hey, I cant compile the devt branch as it fails with the error: 'class HardwareSerial' has no member named 'setRxBufferSize' . I dont understand why thats happening

luc-github, 2017-10-04:
@mzeve devt branch need master git from ESP core not released version, setRxBufferSize is not in current released version