luc-github/ESP3D #170: FirmwareTargets for Repetier are the wrong way around?!

apphead, 2017-08-31:
Checking the code in config.cpp, I found 
    if ( CONFIG::FirmwareTarget == REPETIER4DV)response = F("Repetier");
    else if ( CONFIG::FirmwareTarget == REPETIER)response = F("Repetier for Davinci");

I don't know if the programming logic is right, but the naming is certainly misleading. Also, before looking into the code, I assumend that DV stands for dev version of Repetier....

luc-github, 2017-08-31:
Yes these are upside down you are correct - I fixed on my private repo and did not pushed the fix on public github yet

REPETIER4DV stand for Repetier for Davinci - it is my flavor of repetier for Davinci Printer, with customized UI and more feature from printer itself, more settings in EEPROM, manual bed leveling, specific menu item for Davinci printer, but this works also for DUE/RADDS

luc-github, 2017-09-06:
Fixed in devt branch : https://github.com/luc-github/ESP3D/commit/798f6ea149c8e5877891531a94d352765646d400

closed 2017-09-06 by luc-github