luc-github/ESP3D #171: Printer restarts, when opening printer page

blackus3r, 2017-09-11:
Hello, 

First I would like to say that your work is great :-) 

But at the moment, I have the problem, that the printer always restarts, when I open the printer control page. I am using the current Marlin 1.1.5 
I have testet ESP3D 0.9.7 and 0.9.851. After I have started a print, I don't use the WebUi again, because a reload of the page will force a restart of my printer. 

Do you have any tips for me? 

Thanks a lot.

luc-github, 2017-09-12:
Hi that is weird,
Do you print from Host or from SD ?
How do you connect the ESP ?

Can you check the devt branch : https://github.com/luc-github/ESP3D/tree/devt
(you need also to upload the SPIFFS content) it does not query the printer at start unlike the stable version

blackus3r, 2017-09-12:
Hi, 

I'm printing from SD. 
ESP is connected via 12V, stepdown to 3.3V and TX RX on my RAMPS (AUX1). 

It's the same behaviour, when you press connect in "Pronterface - Printrun". The Atmega2560 restarts immediately when loading the printer page (only the printer page). All other pages are running fine without restarting the printer. 

Does the serial connection is closed after some time? Is it a problem that printer and esp are powered on on the same time? 

I'm using the current version from devt #798f6ea

luc-github, 2017-09-12:
well I know when pronterface  connect marlin, it restart board by default - but ESP is establishing connection at boot so when you open page it should be fine 
today commit do not send the M20 at start so no printer command is send when opening webui
I cannot reproduce on repetier -  I will see if I can setup a marlin but I am not sure as I go to  trip today

blackus3r, 2017-09-12:
I will test  again, maybe I can fix it or give you more debug information. 
I'm currently working on a better translation for german. I will send you a pull request later.

luc-github, 2017-09-12:
Awsome thanks a lot

blackus3r, 2017-09-12:
This looks like the same problem with my board: https://github.com/MarlinFirmware/Marlin/issues/4041

blackus3r, 2017-09-12:
I think, the only solution is to always power the esp chip... But that is not a problem for me, because I already had the idea to power on/shut down the printers 12v via esp.

luc-github, 2017-09-12:
I am sceptical about the root cause described in issue and agreed ESP is already powerd , you just reload the page which does not create new serail connection

can you confirm the following:
1 - webui load page 
2 - launch a sd print from page or from printer
3 - wait 10 s 
4 - reload web page

Does Printer restart ?

blackus3r, 2017-09-12:
Yes.

luc-github, 2017-09-12:
I have installed on Mega/ramps (with 2 temp sensors 1 bed/1 extruder) 1.1RC8 and 1.1.5
I do not have full system but only board right now
both FW init when opening arduino serial monitor
none reset when I reload web page
I cannot heat as no full system right now

Can you confirm the behavior when not printing but refresh web page ?

blackus3r, 2017-09-12:
I will test it... I have four other atmegas for testing.. maybe theres something bad with it.. 

Heres the problem I had yesterday serveral times: 

I start my printer. In the same time, the esp starts. 
After the bootscreen disappears, the esp prints via Serial "Connectin..." on the LCD of the ramps board. After some seconds it prints the ip address. 

I open the webpage and start a print. I close the window and after some time, I want to check the state of the print. I open the browser an go to the printer control page and suddenly the boot screen appears. 
Maybe I can film it, would this helps you? 

Does Marlin or the ESP close the serial connection? And after reloading the webpage the esp tries to rebuild the serial connection? 
Maybe we have to send a heartbeat or a keepalive for the serial connection.

luc-github, 2017-09-12:
normally when setting baud rate it create serial connection, and this is done at boot time, no connection is closedby ESP

can you try to see what happen when restarting ESP ? you have a reboot button in interface

On my side it does not restart the printer when restarting  ESP

blackus3r, 2017-09-12:
I will try to log the serial connection.... 
does the esp waits until the serial connection is successfully done? 

Maybe its a problem, that printer and esp are powered on on the same time. 
I will give it a try :)

blackus3r, 2017-09-12:
We can close this. I have found out, that Marlin crashes, when SD List is done. There were other files too on my SD card, after I have removed them, everything is running well except the sd upload. I always getting the error "SD upload failed"

luc-github, 2017-09-12:
Ok sounds good for restart issue 
the serial file copy is very slow and subject to error - but it works normally - you can try same with pronter face

I suggest you to format the Sd card using https://www.sdcard.org/downloads/formatter_4/index.html
it usually solve many issues

blackus3r, 2017-09-13:
Thanks, I will give it a try.
You can close this issue, if you want. 

I have yesterday implemented a nice feature. (screenshot) I will make a Pull Request when its finally finished. 
[image]

luc-github, 2017-09-13:
looks really nice

luc-github, 2017-09-13:
Ok so I close issue - you can update also your progress in web ui for progression
https://github.com/luc-github/ESP3D-WEBUI/issues/5

closed 2017-09-13 by luc-github