luc-github/ESP3D #173: EEprom  mapping

peekpt, 2017-09-17:
Hi, this is not an issue whatsoever, but IDK if you are aware that you can create a typedef struct and put a single variable (structure) on the EEprom instead of making memory maps  and counting bytes.  Then you load all config  it to the struct you have defined . And it's progressive if you don't switch the places of the struct.

You will be ending with vars something like     eeprom.myInt eeprom.mychar

Nice Project btw, I am still trying to fire this up

luc-github, 2017-09-19:
Hi I guess you talk about the get and put functions ?
These were added long time after I have started the project and did not took time to rewrite program accordingly as the current behaviour was ok

peekpt, 2017-09-19:
Thank you lucky for the reply , I understand if it works don't touch it :D , but  defining a type of kind struct, with all variables in one struct is very straight forward. You still use get and put but instead of a char[] or a int you use one struct only. 
Advantages: easy to implement,aggregatable, easy to use , don't need to count bytes make memory maps. Disadvantages: you need to load all the bytes at once. If you need constant access it will slow down.  
https://github.com/esp8266/Arduino/issues/1539
---------------------------------------

About the project,  If I put in station mode I can't access via web on my network, but I am able to discover the printer in Cura. I clueless how to connect to an computer app, I am on a mac and Repetier-host is an old version (dev is off) which doesn't support TCP/IP connection

luc-github, 2017-09-19:
About code
 yes I agree, there is always a way to make better but there is a wording about modifing code : 
Do not try to fix what it is not broken, it works very well with Murphy law 😉 

About your issue
What version of ESP3D are you using ? 
You try to access by web using IP ?

peekpt, 2017-09-19:
I used the v0.6.2 , yes using web ip on port 80 (station), I tried dhcp and static ip, M117 sent to the printer , but I can't get it working. Maybe my esp is damage,  I haven't problems connecting on AP mode.

luc-github, 2017-09-19:
is your signal good enough to connect AP?

peekpt, 2017-09-19:
Connecting is not the problem, all good, the router is 1-2m away the problem is I got web on AP mode but not on Station mode. But the data server is working, at least is detected

luc-github, 2017-09-20:
Yes when have signal issue with ESP in station mode it can connect but it look like ESP do not transmit any web output 
Can you try the devt branch ? it has more commands to diagnose issues:
https://github.com/luc-github/ESP3D/tree/devt
 but with this one you need to update web files as well

peekpt, 2017-09-22:
I finally installed devt branch and I'm just blown away... jeezas man good job!

luc-github, 2017-09-22:
did you managed to connect to web ui when in client mode ?

peekpt, 2017-09-22:
Yes all working now!  I am trying  on an MKS board 1.4 (Tevo tarantula) 

The MKS board has 100R resistors in series until the ATMEGA2560 is it enough or should I build an voltage level adapter

luc-github, 2017-09-22:
ESP is 3.3v and sensitive, so depending of your board you may need voltage adapter

peekpt, 2017-09-22:
Ok, I inserted a voltage divider on RX and a resistor on TX , it is working fine.

luc-github, 2017-09-23:
Great - I close issue if everything is Ok then
Have fun

closed 2017-09-23 by luc-github