luc-github/ESP3D #174: Allow printable ASCII in SSID name

ligius-, 2017-09-30:
Currently the firmware rejects any non-alphanumeric letters in the SSID name (when acting as a client), I haven't studied this extensively, but I believe most printable characters should be supported. By commenting out some code inside "isSSIDValid" I can get the firmware to successfully connect to a "Fritz!Box".

luc-github, 2017-10-01:
what fw version are you using? looking at master there is no more check already

luc-github, 2017-10-01:
https://github.com/luc-github/ESP3D/blob/master/esp3d/config.cpp#L62

luc-github, 2017-10-01:
and devt branch use isPrintable, so it should work

ligius-, 2017-10-04:
Sorry for the late reply. I was testing on the release branch (0.6.2), as that's what is currently most accessible to potential users. Now it would be your choice on how the handle this: reject this as "not applicable" or mark it as fixed when a new release is made.

luc-github, 2017-10-04:
Hi Ok noted - so it is already fixed in master - I plan to do a release in coming days of master and move devet to master as stable now

closed 2017-10-04 by luc-github