luc-github/ESP3D #176: Problem with Wemos D1 Mini & Marlin 1.1.1

mpsdskd, 2017-10-14:
I'm having trouble getting my printer to communicate with my esp.

Installing worked without issues, the wifi connection works flawlessly, I connect the esp to my home wifi, I connect it to my printer and then can access the esp website and i can control my printer. But I seem to not get any information back from my printer, the page looks like it does in #150.

I installed **esp3d v0.9.78** on my **Wemos D1 Mini** using Arduino 1.8.5 on Kubuntu. 
Arduino Settings: Flash Size 4M (3M SPIFFS), CPU 160 MHz, esp8266 v2.4.0-rc1
Marlin is Version 1.1.1 using 115200 Baud.
I attached a picture of my wiring, but that seems to be correct.

**I Have no idea where the actual problem lies. How can I debug my problem? #150 is talking about seeing the serial information, how do I do this?**

Thank you very much for any help

I tried 

- enabling "#define Bluetooth" option in marlin -> no effect
- lower baudrate of 9600 -> no effect
- In my stupidity in my first try I was too stupid to wire my voltage divider correctly, so the first esp was connected directly to 5V in parallel to a 2kOhm connection to ground -> In case this should have killed the ESP, I bought another esp and connected this one correctly with a voltage divider, multimeter shows something 3V with nothing connected to the pin -> nothing
- the last "release" version of ESP3D with esp8266 V2.3 & V2.2 -> no effect
- Enabling another serial connection in marlin -> nothing works anymore

luc-github, 2017-10-14:
how do you wire wemos to your board ?
if your marlin use 115200 in ESP3D you must set baud rate 115200
did you tried to invert Tx/Rx ? bad wiring happen 

I suggest you to use the devt branch instead of master - it have better support for marlin

mpsdskd, 2017-10-14:
Rx/Tx should be okay, as the communication TO the printer works flawlessly.
Baud was set on both devices.
Tried devt-branch, was quite quick, as it kept all my wifi settings, but makes no difference -> I still can talk to my printer, but it does not talk back (or I can't hear it).

luc-github, 2017-10-14:
with devt branch you must also update UI
I do not understand:
>I still can talk to my printer, but it does not talk back
>communication TO the printer works flawlessly
what happen if you send M115 in command ?

mpsdskd, 2017-10-14:
I have now connected the esp to the printer & the printer and the esp connected to my computer, so that I can see the serial communication.
The devt branch seems to be working flawlessly on the ESP and it seems to have reset all the settings, as I had to connect to the WiFi "esp8266" again and put in all the settings.

If I put in "M115" into the web interface command window, it shows 
`FIRMWARE_NAME:Marlin 1.1.1 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
ok`
in the printer's serial monitor.

luc-github, 2017-10-14:
what is your pinter board ?

mpsdskd, 2017-10-14:
Ramps 1.4 on an Arduino Mega 2560 clone

luc-github, 2017-10-14:
so esp use same rx /tx as USBport? be sure usb is disconnected when starting both boards

luc-github, 2017-10-14:
both cannot be used at once

mpsdskd, 2017-10-14:
They are disconnected, I only connected them for testing purposes with the serial monitor.

luc-github, 2017-10-14:
well I am dry...
try to use putty or any tcp/ip monitor and connect in raw using port 8888, it is a simple bridge so communication works in both way

mpsdskd, 2017-10-14:
Okay, I installed putty, connected raw to the printer IP. If I put in "G28 Y0" it does home, but M115, for example, gives no output.

mpsdskd, 2017-10-14:
So, is it correct, that both the printer's TX pin and the ESP's RX pin have to be working for a USB connection to those devices? So that I can rule out defective pins.

luc-github, 2017-10-14:
yes in that case you have wire problem or the ramps board has issue

mpsdskd, 2017-10-14:
The connection on the ramps board seems to be in working order, I could beep the connection from aux to the arduino with my multimeter.
Also I measured 3.1V on the rx pin and 3.5V on the TX pin of the ESP, is that how it should be?

I will try making some new wiring tomorrow, perhaps that could help.

Thank you very much for your time

luc-github, 2017-10-15:
sorry not be able to help more - good luck

mpsdskd, 2017-10-15:
Okay, so today I build a little test for my ESP: I made the Mega send "Test" periodically on its serial port and wrote a little program for an arduino uno and my esp to receive the "Test" and show it on the serial monitor, and connected both to my computer so I could see the communication on the serial monitor. 
Arduino Uno connected directly to the printer's RX showed the data, ESP connected over my voltage divider did not show it.
So I went on and connected my Uno to the voltage divider - no data again.
Then I connected the ESP to the printer with an 1k in line resistor to prevent the esp from frying with the 5V - still no data.
But neither did the Uno show anything with the in line resistor.
Then I connected the ESP directly to the printer AND: There was my connection. 

As I'm a mechanical engineer with little electrical knowledge I can only guess what's the problem, but it could have something to do with the printer's pull ups perhaps?
Hope I don't fry the ESP in this direct setup (meh, 2,5€ from china^^).

Thank you very much for your help, without your ideas I wold not have continued after my failures.
Perhaps this thread can help others with similar problems.

luc-github, 2017-10-15:
l tested originaly according what is explained in Wiki (https://github.com/luc-github/ESP3D/wiki/ESP8266-12E-F) and never had problem - I am only software guy with some electronic knowledge so I cannot explain all, your wemos may already has a voltage divider so adding a second one will cut communication 

At least you make it working - have fun ^_^

closed 2017-10-15 by mpsdskd