luc-github/ESP3D #177: Error opening: /home.tpl

vMeph, 2017-10-20:
i was trying connect esp32 with out conecting yet to printer to see platform but when putiing ip on browser i get this error
Error opening: /home.tpl

any ideas

vMeph, 2017-10-20:
@luc-github  got it to work now
just one question when i turn on ESP with my ramps it shows on lcd the ip all is ok
but then if i try conect from a device to the ESP the lcd screen light starts to blink and disapears data from LCD
trying to figuer out what could be, i did a voltage divider for the ESP rx to tx of arduino ramps
baud rate is at 250000 like marlim firmware
any ideas what could be happening?

thanks

luc-github, 2017-10-20:
what version of FW do you use ?
Did you tried to connect to ESP directly connected to your Computer to see the serial output when you try to reach ESP ?

vMeph, 2017-10-20:
im using version FW: V0.6.2
i finaly got it to work, i conect to machine then says conected but no comands work no jog moves, no bed set nothing
im using Marlin-bugfix-2.0.x firmware

luc-github, 2017-10-20:
Hmm the version 0.6.2 was done when 1.0rc8 was active so I doubt it will work with 2.0x
I do not follow marlin but current devt of ESP3D support 1.1.x
Your  best chance today it to use it - I will try to test this week end to install the 2.x to see if some adjustment are necessary

luc-github, 2017-10-20:
for what I read : Marlin 2.0 Bugfix Branch : Not for production use. Use with caution! 

It is for 32-bit processor boards - what board do you use ?

vMeph, 2017-10-20:
I use ramps 1.4, marlim is where they make new adjustments,add new funcionalities,etc..it suports 8bit and now also 32 bit.
Is basikly like a arduino nightly build before release , if u see closely now they organizing the structure, instead of having all files in same fol

luc-github, 2017-10-20:
so you use mega/ramps 1.4?

vMeph, 2017-10-20:
yes

vMeph, 2017-10-20:
maybe not working cause  now is  all files organize in folders structure othen then be all files in same folder

luc-github, 2017-10-20:
seems they use platformio for compiling all - I am not familiar with it - so need time / documentation to do it on my side

vMeph, 2017-10-20:
it using platformio cause it seems that arduino ide is having a problem dealing with the file directories and folders, arduino side is working on it
https://github.com/arduino/Arduino/issues/6751

platformio ide its very nice and easy to use  ide once you set up

vMeph, 2017-10-20:
the current official version is 1.6 but from talks i seen they about to release the 2.0 version, in witch it will come with the new folders structures with directories that all isl organize by sections for better organization

luc-github, 2017-10-21:
Ok I have painfully installed and configured platformio to workwith  Marlin 2.X bug fix on Mega/Ramps1.4 + GLCD using platformio - unfortunatly I was not able to use it on RE-ARM/Ramps 1.4 + GLCD but it looks promising

I can confirm the firmware is still in alpha - first command always failed, RE-ARM is not fully functional, and some other small thing during compilation 

2.X Bugfix is actually 1.2 so commands are compatible with 1.1 and works with Devt branch for what I have tested

luc-github, 2017-10-21:
So I close issue as Devt FW is working with current Marlin 2.x -  I did not tested long name support  but reading at code it is still not complete - need to use M33 command to get long name -  so 8.3 limitation is still necessary

closed 2017-10-21 by luc-github

vMeph, 2017-10-21:
Thanks for looking by the way im using normal ramps .1.4 not arm version 32 bit

luc-github, 2017-10-21:
Yes I know - but I took opportunity to test it as RE-ARM is supposed to be the 32b working reference and I have one ^_^

vMeph, 2017-10-21:
Cool

luc-github, 2017-10-22:
Just one note about re-ARM and Marlin 2.X:
I finally succeded to compile and install Marlin FW using Viki 2 screen instead of GLCD (which need a modified cable for 5V)
Unfortunatliy USB/Serial is not yet working so in this case ESP3D won't work properly  as ESP3D cannot communicate with Marlin - it is work in progress on Marlin side