luc-github/ESP3D #178: ESP3D 1.0 release time ?

luc-github, 2017-10-27:
It will be based on Devt branch which is stable now and will be when ESP core 2.4 is available currently we are in 2.4rc2

Squadzone, 2017-11-09:
hello, i like to test using arduino mega ramps 1.4, can i use repetier firmware from repetier? or do i need to use repetier firmware from this repo? thanks

luc-github, 2017-11-09:
Yes you can 
 there is no repetier FW in this repo ^_^ this is ESP8266 / ESP8285 / ESP32 repo

Squadzone, 2017-11-09:
so i dont need this https://github.com/luc-github/Repetier-Firmware-0.92?

Squadzone, 2017-11-09:
looks like you made some change in there, is there any patch to apply in repetier firmware from repetier?

luc-github, 2017-11-09:
sorry i do not understand, can you elaborate ?

luc-github, 2017-11-10:
Ho I missed previous message : this repetier flavor is for Davinci printer based on same MCU as Due - it has some extra but nothing is mandatory for ESP3D 

I test ESP3D with original repetier as well

Squadzone, 2017-11-10:
great, i will try it, and report back

luc-github, 2017-11-13:
Devt branch has been merged to master - this will avoid confusion

luc-github, 2017-11-13:
As there are no time frame for ESP8286 2.4 and no staging / stable JSON for RC2 but git download - master git will now be the reference for ESP Core ESP8266 and ESP32 and for ESP3D repo

1.0 will be released when ESP8286 2.4 is released, waiting for that code will be maintained for issue onl , no big new feature will be added - so I can focus on 2.0 to add features

luc-github, 2018-04-26:
Ok done with delay - but done ESP3D 1.0 is now out
https://github.com/luc-github/ESP3D/releases

closed 2018-04-26 by luc-github