luc-github/ESP3D #179: ESP 3D 2.0 Status

luc-github, 2017-10-27:
Ok as ESP3D 1.0 will be released soon (I hope) time to prepare 2.0 

What is the plan ?
- it will definitly  use ASync WebServer to have better both way communication between server and client and avoid some unecessary poiling which can disturb printer:
 - follow all printer output in terminal like
 - temperatures update when heating
 - follow disconnection in authenticate page
 - allow template back ? (TBD)
 - better following of printing progress without querying printer for some FW
 - Support Websocket/Serial - I still do not understand why but it will be there ...
As Now ESP32 is supported and give a lot of opportunity 
 - Add screen support like [Nextion](https://nextion.itead.cc/) as it can be fun

Any suggestion is welcome and can be discussed here

luc-github, 2017-10-27:
Also will support this kind of screen (0.96 inch oled 12864 ) which is cheap and popular : 
[image]

kafftass, 2017-11-01:
If have added support for ILI9341 SPI Display for me , just a test, works fine. with 320x240/2.4 '' Resolution its better for my eyes. ;) At the momemt i only display sprites on it. With the optional Touch these display are multifunctional usable, no more worry about buttons. They are available from china for less than 6 €.
I also use MQTT for status messages and control, could be a choice for the next Version.

luc-github, 2017-11-02:
I am not familiar with MQTT protocole so sorry if my question looks weird:

It  seems mqtt is a  subscribe/puplish protocole so it needs to be implemented with a goal to discuss with specific target,  also the mqtt client for browser looks using websocket.
I understand it is used on home server like Jeedom or Domoticz but main usage of ESP3D is WebBrowser
So for sending some data to web browser need tcp/ip + websocket + mqtt - when today ESP3D use http+ JSON also based on query - answer way. 

Can you explain / share to me what is the gain to use mqtt for ESP3D ? I would like to learn and understand.

kafftass, 2017-11-02:
Hi Luc ,

i use ESPs in my Homeautomation , controlling my planthouse, my 
heater,my garden pump,  my light and everything which has a switch/cable ;)

Some things dont make sense , some are really good.

MQTT is a message broker where messages could be stored in "channels". 
Devices can subscribe to this channels and will be notified when new 
messages(values,conditions) arrive.

It is not so easy to control all devices via the included WEB-Interface 
so i have a general controling instances , in my case OPENHAB on a PI. 
There you can define rules what should happen on which event.

In my projects i use MQTT for send/receive values or conditions from the 
devices in a bidirektional way. So i can control the devices not only by 
WEB-Interface. In Arduino environment i have good results with the 
"PubSubClient".

In ESP3D it would be possible to send Status messages ( is printing, is 
finished) to the defineable channels to display the Status in the 
homeautomation. Also bed/head temp .

Needed entries (in Flash) were :

MQTT Server : IP or FQDN

Channel Root : path/Name of the channel e.g : /MYCHANNELS/ESP3D

List of values to transmit and a interval time

Hope it is understandable what i wrote.

I have a quick and dirty implemention of mqtt running an my ESP32 with a 
collido DIY printer. I have had the effect that ESP32 DHCP clients 
always gets a new IP-Adresse , not the ESP8266. With a quick look in 
MQTT Server i got the IP sended via MQTT after connected.

Now i have static IP for the ESP32´s.

Dont know if any other will need something like that.

luc-github, 2017-11-02:
@kafftass thanks so this confirm this is only in purpose to link to home automation server.
 I have already added a ticket for that : https://github.com/luc-github/ESP3D/issues/148

I am quite new in home automation but all servers I have tested Jeedom, Domoticz, openhab support support different protocol including JSON for event signal 
https://community.openhab.org/t/how-to-recieve-json-data/3720

khseal, 2017-11-11:
There is an interesting device based on ESP32
https://ru.aliexpress.com/item/M5Stack-Extensible-micro-control-module-WiFi-Bluetooth-ESP32-development-kit-Built-in-2-inch-LCD-for/32827969217.html?spm=a2g0v.search0204.3.1.gOUnx4&ws_ab_test=searchweb0_0,searchweb201602_0_10152_10065_10151_10068_10344_10345_10342_10343_10340_10341_10543_10541_10307_10301_10060_10155_10154_10056_10055_10539_10537_10536_10059_10534_10533_100031_10577_10103_10578_10102_10169_10142_10107_10084_10562_10083_10312_10313_10314_10211_10550_10073_10551_10128_10552_10129_10553_10554_10555_10557_10125,searchweb201603_25,ppcSwitch_0_ppcChannel&btsid=e46b4bc2-4b16-4d77-bdce-c6e39be7252d&algo_expid=dd5edb72-9a17-4f5d-9946-2519d3b5b8a5-0&algo_pvid=dd5edb72-9a17-4f5d-9946-2519d3b5b8a5
It also has buttons, a display and a memory card.

luc-github, 2017-11-11:
@khseal m5stack have their github https://github.com/m5stack - I have ordered one to play with 
Thanks for sharing ^_^

shaddow501, 2017-11-18:
Suggestion.
1. Add some GPIO control for turn on/off additional LEDs.
2. Option to Connect / Disconnect to the printer if using both Wireless and USB connection that one will not work when other is connected.
3. Option to see the IP address and connection status in the OLED screen.
4. Option to control PTZ wireless camera

[image]

luc-github, 2017-11-18:
1 -Add some GPIO control for turn on/off additional LEDs.
[luc]This is already available : 
```
*Get/Set pin value
[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES]
if no V<value> get P<pin> value
if V<value> 0/1 set INPUT_PULLUP value, but for GPIO16 INPUT_PULLDOWN_16
GPIO1 and GPIO3 cannot be used as they are used for serial
if PULLUP=YES set input pull up, if not set input
if RAW=YES do not set pinmode just read value
```
and put the according command in UI using macro

2 - Option to Connect / Disconnect to the printer if using both Wireless and USB connection that one will not work when other is connected.
[Luc]How this can be done ? It is handled by printer FW or you need a daughter board that do the job according ESP priority on Serial and cutting USB if ESP is discussing with printer 

3 [Luc] Yes that is the plan

4 - [Luc]Not sure how to do that , do you mean Macro accept url as command ? what camera has such command ?

shaddow501, 2017-11-18:
1 -Add some GPIO control for turn on/off additional LEDs.
[luc]This is already available :

*Get/Set pin value
[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES]
if no V<value> get P<pin> value
if V<value> 0/1 set INPUT_PULLUP value, but for GPIO16 INPUT_PULLDOWN_16
GPIO1 and GPIO3 cannot be used as they are used for serial
if PULLUP=YES set input pull up, if not set input
if RAW=YES do not set pinmode just read value
and put the according command in UI using macro
[shaddow501] -I mean buttons on the web UI

2 - Option to Connect / Disconnect to the printer if using both Wireless and USB connection that one will not work when other is connected.
[Luc]How this can be done ? It is handled by printer FW or you need a daughter board that do the job according ESP priority on Serial and cutting USB if ESP is discussing with printer

[shaddow501] - to add a check mark whether it is controlled by USB or Wireless, than if selected USB than the ESP32 do not transmit TX commands only receives RX.

3 [Luc] Yes that is the plan

4 - [Luc]Not sure how to do that , do you mean Macro accept url as command ? what camera has such command ? ---- I have no idea lol

luc-github, 2017-11-18:
1- Macro buttons are buttons ^_^ no ?
2 - Ok so User set the status not ESP3D automaticaly and UI will be read only for printer commands - is that what you mean ?

shaddow501, 2017-11-18:
1- where can I find it in the web page?
2 - yes. the ESP3D will not send commands only will receive then,

luc-github, 2017-11-18:
1 - [image]
[image]

2 - Currently ESP / ESP-WEBUI do not send anything unless user request info / send command  on UI to printer

shaddow501, 2017-11-18:
How do you write there the command to turn on / off the GPIO can you give an example?

luc-github, 2017-11-18:
Should be  :
`[ESP201]P5 V1` High on GPIO 5
`[ESP201]P5 V0` Low on GPIO 5

You can check the code also : https://github.com/luc-github/ESP3D/blob/master/esp3d/command.cpp#L391-L470

luc-github, 2017-11-26:
I have created a ticket for every topic - thank you everyone

luc-github, 2018-01-11:
ESP3D 2.X topic are now properly tagged : https://github.com/luc-github/ESP3D/issues?q=is%3Aissue+is%3Aopen+label%3AESP-2.X

luc-github, 2018-01-14:
Devt Plan should follow this list:
1 - File upload check with file size (ESP3D + ESP3D-WEBUI)
2 - ESP3D as library (ESP3D)
3 - Oled support (ESP3D)
4 - WSocket + terminal like output (ESP3D + ESP3D-WEBUI) 
5 - GRBL support (ESP3D + ESP3D-WEBUI)
6 - Output to printer configuration (ESP3D + ESP3D-WEBUI)h
7 - Event management (ESP3D + ESP3D-WEBUI)
8 - Nextion Screen support ((ESP3D + Nextion FW)
9 - TBD

duderu-erik, 2018-03-07:
I have been digging the Nextion displays as well and did a POC with the 3.5 inch screen and an ESP32... Check it out at https://github.com/duderu-erik/esp32-nextion

It's rough and is not for the Arduino folks which I don't do. But it has the source nexiton HMI file which provides examples and will get you thinking about how much you want the screen processor to do and what you want the micro-controller to do. It's an interesting give and take.

luc-github, 2018-03-07:
I did also using the lib https://github.com/itead/ITEADLIB_Arduino_Nextion with ESP32 some time ago - an all examples are working so far - I used lib version 0.9. which fingerprint is not big and avoid to handle the Nextion syntax messages.

I have also started to design a UI here some screen using open icons :  
[image]
[image]
[image]
[image]

Because UI actually define the back end code.

 Nextion as a big memory limitation, so displaying File list, EEPROM content or smoothieware config.txt and even ESP EEPROM settings is problematic, because they do not fit memory size (tooooo big for it)  so they cannot be used straight forward, but let say 5 by 5  elements, which would be Ok if ESP itself have enough memory to store also all information : like the 10K+ of smoothieware config or the  4k+ of repetier EEPROM - no mention the SD print file list, which could be huge. 

So need to add some code tricks and may be using SPIFFS as swap location for lists to keep navigation smooth, TBD

duderu-erik, 2018-03-07:
Cool stuff Luc and your graphics skills are way better than mine. I'm using
the Nextion 3.5 inch basic screen which has plenty of memory for  my use. I
don't do Arduino though as most of my code is proprietary due to customer
demands. I'm trying to throw some free snippets out their when I can but it
will be either Assembly, C or C++ straight off the esp-idf.

I'm doing a POC on some hardware as well as the ESP32 IDF which is still
buggy as you probably know. It will be interesting to see how the ESP32 and
Nextion display holds up to 115 degree Phoenix, Arizona weather this
summer. :-]

Cheers!

luc-github, 2018-03-08:
@duderu-erik yes ESP32 is still young -  I have identified even younger one as several library are sill missing  : https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK
If you know other possible plateform like (cost / feature) please share, I would be happy to have a look ^_^

luc-github, 2018-03-15:
I am working on web interface to make it more flexible, to support laser/mill interface removing what is specfic to printer like temperature and extruder and even removing bed/ fan controls for printer which no need. for that there is now a preferences dialog box : 
[image]
Preferences are saved in SPIFFS in a files called preferences.json
The goal to to move what was stored to cookies in this files - for the moment only language is saved in it but I will move all to not be device dependant for the preferences.
I think it make sense to move the settings specifics for UI out of EEPROM and out of ESP settings:
[image]

duderu-erik, 2018-03-15:
Cool stuff. Are you using the JSON component included with the ESP-IDF?  I
have found that it works ok with Chrome, Firefox and Dolphin both on
Windows and Android platforms but the JSON component causes a panic and
reboots the ESP32 using IE on Windows or ANY browser running off an iOS
device. Even Chrome on an iPhone or iPAD does not like the JSON component.

Downloaded Fiddler to try and debug further but haven't had time yet.

luc-github, 2018-03-15:
@duderu-erik no all code for UI is here : https://github.com/luc-github/ESP3D-WEBUI
latest code is still on my private gitlab repo but I will sync soon

The JSON is used by javascript only

luc-github, 2018-03-16:
grbl support is now ready : 
[image]

I also realized that Z controls should be showed/hidden based on preferences as cnc/ laser may not have such feature -  more to put in preferences

luc-github, 2018-03-23:
WEB-UI 2.X is working for the moment with current ESP3D fothe ones who want to try : https://github.com/luc-github/ESP3D-WEBUI/tree/asyncUI
it has : UI preferences for better user experience/ several UI glitch fixes / support for GRBL FW

luc-github, 2018-03-23:
Here the current preferences ;
[image]
Will add mixing extruder support soon

luc-github, 2018-03-30:
the connecting events is now implemented and used to allow only one instance at once : 
[image]
this will prevent to overload the ESP with too many queries at once 

that I found one issue on ESP32 + asyncWebserver which seems not like to be poilled for temperature / position every 3 seconds - No issue on ESP8266 at least -> need to dig in Async code 😟

luc-github, 2018-04-21:
I have just pushed update for Async in ESP3D and ESP3D-WEBUI

I was not able to find solution for ESP32 in async server part so I have rewriten the command processor in ESP-WEBUI to avoid any command overlap - now all commands go to FIFO and give better control to requests
As now only one UI instance  is allowed and commands go to fifo - there is (in theory) no chance a command poluate an upload 

Boot is also more under control 

Debug messages present in console are left until command processor is considered as stable

luc-github, 2018-04-25:
for DTH support I plan to use https://github.com/beegee-tokyo/DHTesp , I have tested it and looks ok on ESP8266 and ESP32
unless someone has better proposal  this is what will be used on ESP2.X

luc-github, 2018-04-30:
API for OLED is now done -  it must be completed by the output configuration soon 
Here some screenshots

Boot splash   
[image]

AP mode  in safe mode  
[image]

Station connecting    
[image]

Station connected
[image]

luc-github, 2018-05-07:
Flashair support is not working as I wish : query flashair using command.cgi and upload.cgi is not allowed, I am facing an *Access-Control-Allow-Origin* issue, seems there is no way to have an full ajax integration - so currently the only solution is to host the flash air web page in an iframe, too bad...

luc-github, 2018-05-24:
DHT support is now done, will push it to repo soon
it is displayed on nav bar as not linked to extruder temperature
[image]

can be show/hide based on preferences
[image]

because it is handled by FW and not UI - settings are in EEPROM
[image]

the refresh interval is for ESP3D to query sensor - result is send to WEB UI using events    

On OLED only temperature is displayed currently to save space    
[image]

luc-github, 2018-05-25:
WSocket  + terminal like output is now implemented upstream  - will be in next push to this repo like others new features

Now all printer serial output can be redirected to websocket 
here repetier output in terminal 
[image]

and of course it can be disabled anytime
[image]

luc-github, 2018-05-28:
Now working on Reprap Native FW support - thanks to @svs072477 and his nice board [ruramps4](http://3dtoday.ru/blogs/svs0724/continuing-the-theme-ruramps/)    

**M115** give : `FIRMWARE_NAME: RepRapFirmware for RURAMPS4D FIRMWARE_VERSION: 1.21 ELECTRONICS: RADDS 1.5 FIRMWARE_DATE: 2018-03-21`     

I have already filled the needed info in https://github.com/luc-github/ESP3D/wiki/Firmware--support
looks like the answers are always one line ended  by **\r\n** but no **Ok** ack like others FW

The support of M408 will be another part as no more specific to reprap/duet3d

cyberhasse, 2018-05-31:
Hi!
Great work :)
I would like to try the GRBL functionality but i can't find the sources to compile.
Maybe i'm dumb or blind..
Regards.

luc-github, 2018-05-31:
GRBL is on development  version for 2.0 branch which as mentioned on read me is :
<u>[Development version for 2.0 (asyncwebserver branch)](https://github.com/luc-github/ESP3D/tree/asyncwebserver) & [ESP-WEBUI (asyncUI branch)](https://github.com/luc-github/ESP3D-WEBUI/tree/asyncUI):</u> 
 🙈 😸

cyberhasse, 2018-05-31:
Thank you for the quick reply.
Yes i found that but how do i compile the source?
V1 was easy when loading ino file in arduino ide..
Regards.

luc-github, 2018-05-31:
Still in read me : https://github.com/luc-github/ESP3D/tree/asyncwebserver#installation
Necessary libraries are in libraries folder : https://github.com/luc-github/ESP3D/tree/asyncwebserver/libraries

latest webui is in git also : https://github.com/luc-github/ESP3D-WEBUI/blob/asyncUI/index.html.gz

cyberhasse, 2018-05-31:
Great thank you!
Regards.

luc-github, 2018-06-03:
Just pushed latest update for DHT / Oled support / Websocket: 
https://github.com/luc-github/ESP3D/tree/asyncwebserver

luc-github, 2018-06-12:
Mixed extruders feature is now ready (up to 10 mixed extruders, still 2 extruders max for not mixed)  
In preferences:     
[image]
In UI:    
[image]

Read only feature is also ready:
In preferences:
[image]

In UI if enabled:
Only control that do not affect printer are allowed
[image]

Progress button is the only exception because it is the one that allows to check print status

I will push these update to git when authentication is done

luc-github, 2018-06-18:
22 is. On esp32,  on esp8266 you can use  16 or any free pin you have

luc-github, 2018-06-20:
On ESP8266 I got several data lost issue from serial when quantity is more than 10K, several middle parts are missing - I did not test ESP32 as much as ESP8266 so not sure about behavior on ESP32

I am not sure where the issue come from as even using TCP commands, the bridge is loosing data ....
As the only big difference is the webserver - I will fork async branch and use sync web server for ESP8266  to see if it solve issues on ESP8266  - I wil use https://github.com/Links2004/arduinoWebSockets to keeps same feature to be able to compare behavior properly
this is not a little change so need time to work on it

luc-github, 2018-06-25:
Ok it is confirmed: using same core but just put webserver sync and reading 24KB text file - got 0 data lost using serial/telnet bridge (which is not related to Webserver)  , and 0 data lost with Serial / Webserver bridge.
So issues come from Async webserver somehow - but no idea why... 

New ESP3D  2.0c is based on ESP3D  2.0b allowing ESP32/ESP8266 and Webserver Sync or Async , actions to be done are:    
*  I currently focus on sync/esp8266 to bring back all features - still miss event / websockets but should be done soon
* After I will check sync ESP32 to be sure everything is Ok

* If Ok, I will push code to github

* Then I will complete ESP3D 2.0c todo list

So test on Async will be postopned until issue root cause is found - to find out I will do a Minimal code to reproduce Async + Serial/TCP bridge , but currently it is not a priority vs the ESP3D 2.0c

luc-github, 2018-06-26:
Ok 3 small issues moving to sync webserver : 
*  Post arguments are not available untill all files download are processed, this is needed to check files are properly downloaded on SPIFFS and even FW update file.
=> solution : do a PR for ESP8266 Webserver allowing query to temporary post arguments array,
this should not break anything in existing code, so hope PR will be accepted when I push it

* Websocket library is not part of Webserver unlike async one, so need to use a different port number than webserver, 
=> solution : Because  I am not sure this is a necessary flexible parameter, I am planning to set automaticaly websocket port as web port +1, eg Webport = 80, Websocket will be 80 +1= 81, so not need to have a configuration for it

* There is no event class / support in Websocket library 
=> solution Use a second instance of Websocket adding `[label]` in message to discriminate messages, using web port + 2 (eg:80 +2 = 82) as need a new port for new instance, so not need to have a configuration for it neither

This will need a small modification of Web UI but all features will be there.

One drawback : this need now to forward 3 ports instead of 1port on routers for external access

luc-github, 2018-06-28:
Update : actually use 1 additional port is enough : 
*  use Text message for events 
* use Binary message for Serial to Websocket bridge
So only one instance of websocket of websocket is neccessary

luc-github, 2018-06-30:
Smoothieware config-override is now supported :cat:   
Default page:    
[image]

When no config-override file:    
[image]

If there is a config-override file:    
[image]

Will test this week end and fine tune some parts, then next week 2.0 sync for ESP8266 will be pushed to repo, I will then test if ESP32 is still Ok

luc-github, 2018-06-30:
I will change the /sd/config.txt and /sd/config-override push button by radio button for UI consistency vs  ESP3D Settings
[image]

luc-github, 2018-07-04:
Code pushed on https://github.com/luc-github/ESP3D/tree/2.0
Compatible ESP8266/ESP32 Async/Sync Webserver:
For best performances use : 
ESP8266 with Sync webserver
ESP32 with Async Webserver  

I did not tested all configurations yet, but it is quite stable,  feedback are welcome here https://github.com/luc-github/ESP3D/issues/238

For open topics I move them to https://github.com/luc-github/ESP3D/issues/239

So this topic can be closed as it become to be long

closed 2018-07-04 by luc-github

DemosfenArsen, 2019-01-06:
А экран oled 128*32