luc-github/ESP3D #180: ESP8266 to TTL

shaddow501, 2017-11-04:
Dear Sir

I have seen you made a nice option to control the 3D printer,
I wonder if it is possible to set something like:
From the printer side ESP8266 to send/get  the data from the TTL (of the printer )to other ESP8266 connected by USB Serial to the PC.
Than I can also control the printer from lest say Cura, or Repetier host that will connect trough the (wireless) serial port.

luc-github, 2017-11-04:
What you mean by TTL ?
Currently you already can connect from Repetier Host using ipaddress:8888 directly in TCP/IP
[image]

shaddow501, 2017-11-04:
I mean that you connect the ESP8266 to the printer board RX/TX (TTL) (sorry I didnt mention that)

I would like also to get/send  this data to other ESP8266 that will be converted to USB to serial.
Some programs of 3D require direct serial connection to the printer for example Cura.

Also I dont know if the setting in the webpage will fit the Anycubic i3 mega printer...

luc-github, 2017-11-04:
ESP3D is for the ESP connected to the printer

If you are looking for something that help any application to transform a serial to wifi to connect to another wifi device like cura - you need a different FW than ESP3D for this part
Cura has some plugin for that and a guy did an ESP FW for that : https://github.com/probonopd/WirelessPrinting

shaddow501, 2017-11-04:
That is nice but how to integrate both worlds :)?

luc-github, 2017-11-04:
the Link I gave you do not use ESP3D FW at all -

If you really want to use ESP3D on printer -  you need to write a FW for the ESP which do the bridge between PC serial and act as  a wifi client connecting to the wifi server of ESP3D

there are code example for wifi client : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino
You just need a page to configure the wifi configuration, there is a lib for that https://github.com/tzapu/WiFiManager and a page to configure  the Wifi server address or you can do the configuration by serial - nothing difficult I think

after will this connection reliable for printing ? I have no Idea - I always print from SD never from host for  safety

shaddow501, 2017-11-04:
Ok, How did you add support to 250000 baud rate? what should be done?

luc-github, 2017-11-04:
I just put 250000 in supported list of baud rate :smile_cat:

shaddow501, 2017-11-04:
Where which file?

luc-github, 2017-11-04:
https://github.com/luc-github/ESP3D/search?utf8=%E2%9C%93&q=250000&type=

shaddow501, 2017-11-04:
thanks a lot

shaddow501, 2017-11-05:
I am trying to compile and get this error.

exit status 1
'class HardwareSerial' has no member named 'setRxBufferSize'

shaddow501, 2017-11-05:
Something strange, when I use this lib 2.4.0-nightly+20170222 I can compile your code without any issues but have errors on the other https://github.com/probonopd/WirelessPrinting/ and cannot compile it, I get a lot of errors.

When I change the hardware lib  (remove 2.4.0-nightly+20170222 lib and add lib esp8266-2.4.0-rc2 ) 
I am able to compile the https://github.com/probonopd/WirelessPrinting/ and not able to compile yours....
I get this error

sketch\webinterface.cpp: In function 'void handle_web_interface_home()':

webinterface.cpp:874: error: cannot convert 'station_info::<anonymous struct>' to 'station_info*' in assignment

             station = station->next;

                     ^

exit status 1
cannot convert 'station_info::<anonymous struct>' to 'station_info*' in assignment

What is going on with this? Do i need to change hardware libs every time I want to compile?

luc-github, 2017-11-05:
My master is outdated due to recent update of SDK but is ok with esp core you mentioned

My Development branch follow all esp core update and actually is much more stable than master which is 7 month old

I should update the master but did not have time yet - I am waiting ESP core do 2.4 release to release the devt branch as master and also as release

I suggest you to use my devt branch with the latest git master of ESP core to get the best and stable version of ESP3D

for the other project issue looks same : he use old version (july) of ESPcore as mentioned in his README

shaddow501, 2017-11-05:
The best option we can have is your code that will also support octoprint and sd card :)

Ilan B. (Mobile)

https://github.com/luc-github/ESP3D/issues/180#issuecomment-341988356

luc-github, 2017-11-08:
I already support SD card - but currently it is on private repo waiting to be released
For octoprint ESP8266 won't fit for sure - but may be on ESP32 - but for this you should request this to octoprint developers team because if octoprint fit ESP32 then ESP3D is useless

shaddow501, 2017-11-08:
Dear Luc

 

In the original ESP3D Do you support the printer SD card or a SD card that connected directly to the printer motherboard?

If I buy ESP32 can I program your original code to it? Or does it need some changes in order to fit?

Isnt it possible that you can add octoprint as well to your code?

 

The printer I am going to use it is Anycubic i3 Mega.

luc-github, 2017-11-08:
Current ESP3D support Serial SD upload - using M28/M29 commands but this is nightmare
Direct Upload to SD need a hardware rework to connect SD card reader to ESP
Devt branch support ESP8266/ESp8285/ESP32

I do not use octoprint, I always print from SD because it is safer IMHO - I do not know octoprint code so porting it to ESP is out of my scope

shaddow501, 2017-11-08:
Dear Luc

 

I did try to upload the index.html.gz to the module using the webgui upload feature but I get an error when trying to connect some kind of FW error?

luc-github, 2017-11-08:
you use which upload SD or ESP3D Filesystem ? index.html.gz go to ESP3D Filesystem

shaddow501, 2017-11-08:
Dear Luc

 

I have not connected the esp to the board yet so I upload it to the local flash.

luc-github, 2017-11-08:
can you screen shot the error ?

shaddow501, 2017-11-08:
Sure

 

Here attached.

luc-github, 2017-11-08:
I do not see any screenshot - It may not work to attach by email

shaddow501, 2017-11-08:
I try here
[image]

luc-github, 2017-11-08:
this looks you do not use Devt branch to build the ESP3D fw but master or released ESP3D with ESP-WEBUI
what `http://10.0.0.110?forcefallback=yes`  show ?

shaddow501, 2017-11-08:
I will able to provide more information much later today since I took the unit to work and now I cannot set it back to AP mode... so I cannot enter the website.

shaddow501, 2017-11-08:
the http://10.0.0.110?forcefallback=yes,  show the same thing - Connection failed! is your FW correct?

luc-github, 2017-11-08:
so it confirm it is not proper FW for ESP-WEBUI it is not the devt branch FW
it is easy to check : the data directory contain what UI is supported
home.tpl file for master branch or and released version  or index.html.gz for devt branch

shaddow501, 2017-11-08:
So what do I need to program that it will work which code? can you provide a link?

luc-github, 2017-11-08:
just what I wrote https://github.com/luc-github/ESP3D/tree/devt
with git version of ESP8266 https://github.com/esp8266/Arduino/tree/master

shaddow501, 2017-11-08:
Thanks now it worked.
Now my issue how to convert the USB port (cp210x) that come from the printer back to TTL but leave the cp210x available, basically to build a dongle that will connect to the USB port of the printer and give a wireless support.

have any clue?

luc-github, 2017-11-09:
nope sorry 
Closing issue as no more FW related

closed 2017-11-09 by luc-github

shaddow501, 2017-11-09:
Ok, thanks

Ilan B. (Mobile)

https://github.com/luc-github/ESP3D/issues/180#issuecomment-343082387

m8dhouse, 2019-01-01:
@shaddow501 I'm also using a Anycubic i3 mega (8bit). But I can't seem to get it to work :-/ 
It uses modified 1.1.9 marlin version (https://www.thingiverse.com/thing:3249319/files). ESP is an ESP12e I had lying around (Doit version). I've tried the v1 version now of ESP3D - all compiles and uploads fine. When I connect to Arduino serial monitor I see the the temperatures coming in. But as soon as I use the display for something the serial connection is gone. Also from the ESP3D interface I get nothing via serial or no commands go through. I'm using the RX/TX on the trigorilla board. I've seen some of your posts and from what I understand you got it to work on an i3. Could you share what versions you are using and where you connected the RX/TX to?  Thanks

shaddow501, 2019-01-02:
Dear m8dhouse.
First of all I use the original firmware that came from the manufacturer, second I didnt have the time to install it on my printer.
So at the moment I have nothing new to tell you.

I also do not know why you have decided to use modified firmware, what do you gain from it? are there any visible improvements from the original firmware to the custom one?

On my printer I have replaced the motor drivers to TMC2208 and reverse the connections. I am using 0.3 nozzle size and print at 0.2/0.1 quality. 

Also I print at about 200% speed setting and getting fast printing without losing quality on ABS.

My settings are:
[image]