luc-github/ESP3D #181: d1 mini connection schema for 5v board and kbf2.0 board pins both for wiki

exilaus, 2017-11-12:
https://photos.app.goo.gl/RnZmNcFNCWmF4ISE2 to add in your wiki  !!!  
amazing work!

luc-github, 2017-11-13:
Thank you for sharing - what is KBF2.0 board - what fw goes on it ? I did not find lot of information about it

exilaus, 2017-11-13:
Nothing, thanks you for that masteripiece of software !!!   kbf2.0 are ultra low china board are 100% ramps compatible  board same pinout can use marlin or kimbra without issue, not open but  have more order pinout and have labels on board for all pin have label on back of board , and permit to use no steptick drivers without converter or fly pins..   https://it.aliexpress.com/item/BIQU-KFB2-0-Stampante-3D-scheda-controller-Ramps1-4-Mega2560-R3-a4988-DRV8825-TMC2100-Reprap-Mendel/32835310377.html

luc-github, 2017-11-13:
Is it ok I split picture in 2 parts?
* 1 for D1 mini
* 1 for kbf2.0 board

luc-github, 2017-11-13:
Forget my question there are already splited ... orz

luc-github, 2017-11-13:
Pages added - thank you - I have increased Font size on KFB picteure for better visibility

closed 2017-11-13 by luc-github