luc-github/ESP3D #183: ESP32 to RAM

alequeiroz, 2017-11-15:


luc-github, 2017-11-15:
Did you check the wiki ? RAMPS is described

alequeiroz, 2017-11-15:
If I say I've only seen it now, nobody believes it, but still what changes do I have to make at Marlin? I have an ESP32 DEVKIT V1, which pins should I use? Sorry for the questions, I'm starting this area.

luc-github, 2017-11-15:
No I am sure - no one read the docs - I get used to :cry: 
Nothing to do so far - these are Serial connection same as USB connector - just be sure baud rate is same on Marlin and ESP3D and define Marlin FW as target FW in UI in settings or use the setup menu

alequeiroz, 2017-11-15:
Ok, it's already working perfectly, just when I try to upload a gcode it gives me the message:
[image]
Is it normal?

luc-github, 2017-11-15:
what is the name of the file ?

alequeiroz, 2017-11-15:
maxilar.gcode, queixo.gcode, dedo.gcode, dedao.gcode, and others they all give me that error

agonysfate, 2017-11-15:
Try the .gco extension name..

luc-github, 2017-11-15:
so yes it is working properly ^_^ As error message says only 8.3 filename on marlin
these are not

alequeiroz, 2017-11-15:
Thank you guys! 
[image]

luc-github, 2017-11-15:
good luck , serial upload is slow like hell...
feel free to feedback, I did not tested esp32 as much as I wished

agonysfate, 2017-11-15:
Luc, i have two new esp32 but didn't try your code on those boards yet only esp12e
How can i contact you direct? Also need your documentation link, you use facebook?

luc-github, 2017-11-15:
there is gitter in private or in room for chatting, documentation is the read me and the wiki, there is no other link sorry

alequeiroz, 2017-11-15:
Yeah, i already notice the upload rate, even in esp32 is that slow? rlly? But for testing is ok, and i wanna know if i can command my 3d printer beyond the home´s wifi, it´s connected as client and from my work internet i can command it, is it possible?

luc-github, 2017-11-15:
yes this is the protocol, cannot be improved unfortunatly
yes of course, why not ? It is just a network configuration not depending of ESP3D itself

agonysfate, 2017-11-15:
Its because serial baud rate is slow specially for speed like 115200 or 250000
I upload my codes on 900000 baudrate for esp12e and even wait long until it finishes uploading..
Direct upload like when you use OTA is way faster.. But you need to keep your runtime implementation on your code so arduino IDE  can detect its ip address

agonysfate, 2017-11-15:
I meant when you use OTA on wifi to upload codes,
Same principle if its possible to upload files to esp12e or 32 using OTA.
ESP can be programmed over wifi so fast using ota. Faster than 900000 baudrate on serial transfer.

alequeiroz, 2017-11-15:
@luc-github But i have to clear the ip on my router?
@agonysfate i think i´ll test arduino due soon, and i´ll give a feedback with the rates, arduino mega is to slow for this kind of things.
And btw u guys had seen the implementation of ESP32 as a completly board controler? https://github.com/MarlinFirmware/Marlin/issues/6524

luc-github, 2017-11-15:
yes it is on going, it use ramps 1.4

luc-github, 2017-11-16:
OTA is not implemented as OTA on ESP3D, it does not need arduino IDE - it is web update : just upload  the binary using web UI and ESP3D will restart using new FW - faster than flash upload for sure, and as fast as OTA as using same commands.

The Serial Upload speed to SD is currently limited by 2 things:
1 - ESP8266 has 128 byte hardware serial buffer so can only send 128Byte at once
2 - the M28/M29 transfert protocole is actually to send each line one by one to verify integrity - it is not a real file transfert - when it transfert data as it is on Marlin - Repetier even check and convert each command to binary. So you guess it is very slow...

luc-github, 2017-11-26:
If no update I close issue

closed 2017-11-26 by luc-github

el-shweikh, 2018-02-01:
Hi sorry for my question but can I use the ESP32 as the controller instead of arduino meaga?

luc-github, 2018-02-01:
there is a project for this, but currently there are several open issues:
1 - Marlin FW is not yet ready - ongoing
2- Numbers ofpPins is not enough so need to do some trade off and use GPIO expander - ongoing
3 - ADC for reading temperature is not yet accurate - on going

Please check Simon project: https://github.com/simon-jouet/Marlin and https://github.com/simon-jouet/ESPRamps

Simon allowed me to be part of this great project 😄