luc-github/ESP3D #184: ESP3D for CNC milling - please help

luc-github, 2017-11-16:
Hi I would like to add support for CNC milling - I have almost 0 knowledge on this so I want to build one to understand what are the needs, how it works and of course test ESP3D.
I have read a lot on the net but did not found a clear way to go
 
I would like to use aluminium extruded and 3D printed parts. 

- Any suggestion ?
- Anyone built one already ?

Every suggestion is welcomed

3d-gussner, 2017-11-17:
@luc-github Checkout 
ttps://www.youtube.com/channel/UCItcVCx_yzg59jxuRzcyImg
http://www.maslowcnc.com/
http://www.instructables.com/id/Converting-CNC-Electronics-to-Arduino/

https://www.youtube.com/watch?v=2IoiSy1r2uU

As CNC needs more torque i guess lead screws are the way to go. Depending what material you want to CNC you may need to go even with NEMA23 or bigger.

luc-github, 2017-11-18:
Yes for milling need Nema 23 or bigger and good drivers for what I saw - and no way for belt, only lead screws

luc-github, 2017-11-18:
For milling I plan to use this Kress 06082206 1050W as it looks good 
an may be based on this project : http://www.mon-fablab.fr/openmakermachinepro/machine/descriptif
sorry it is in french

3d-gussner, 2017-11-18:
You build it already? So why not just add the laser to that one?

3d-gussner, 2017-11-18:
My French wasn't that bad in the past, so i can read it. Have to practice bit more :-)

luc-github, 2017-11-18:
no this is the one that match my expectations - the axis are strongs  - they explain each steps and use aluminium + printer part
but some aluminium parts are not generic - so need to find a way to use generic product

vMeph, 2017-11-18:
In my opinion belts works fine, check shapeoko or x - crave machines

luc-github, 2017-11-18:
@vMeph do you have one ? 
I always thought belt are fast and cheap, Ok for soft milling (PCB/soft wood) but getting damaged very quickly for metal / aluminium milling., 
For me timing belts are OK as long as there are no forces involved. 
I can be wrong that is why I open this ticket - discussion on internet that I have read says what and I have wrote, and the pro belt - just say it work but without such explaination or need to use very expensive belt, not usual ones which could make sense

vMeph, 2017-11-18:
you will need to decide what system you want to work with, there is CNC that works LPT parallel port conection that works with mach 3, and there is arduino GRBL, i never tryed work with LPT parallel port but from what i understand it works with direct signal vs arduino that works with buffers,
sense you want to build a CNC machine and a Laser machine why not do 2 in 1?
for lasers machine most likely belts will be a better option cause of speeds, a cheap way to start if you will be using arduino would be get a CNC shield v3 that costs like 3 dollars, but attention that cnc shield v3 was done for versions of grbl 0.8 below, above versions there was a modification on pins12 and 11 that was swaped cause of the PWM , that can be easy fixed on the CNC shield v3 with some cables, there more recent versions of the CNC shield that it suports the GRBL 0.9 and above versions but the ones you find is like 20 dollars, cheaper to just use the CNC shield v3 and just swap the pins, or you can use with out the shield and use external drivers, i have some DM542 drivers, in start i used the shield, but i was having alot of problems with EMF noise so i opted for external drivers and also for the fact that i have some nema 23 3A steep motors and the cnc shield drivers are the A4988 and DRV8825 that can suport so many Amps, about the lead screws or belts i dont see what can be so diferent about, with you be cutting material most like you will not be cutting at huge feedrates, you will need to adjust feedrates vs spindle speed, maybe leadscrews have less blacklash?
but if you keep motors always enable that you most like will have no problems on skipping steeps etc..
im no expert on this so forgive me if i saed anything wrong

im no professional i did this machine with scrap material i had arround i uses windows rails with bearings etc...
there is somethings that i could have improved if i would it build now, one of the things is that the steal bearings that raids on windows rails could have some kind of plastic cover that could be replace from time to time to avoid scratch the rails, if you look at X crave or shapeopko they use plastic covers on bearings
[image]
at same time it sounds like you looking to build something more professional,so you would need to find some good mechanical plans for the mcahine
if i was you i would probably start by doing a simpler machine that could do the CNC milling and laser at same time to see how all works and how it goes and then jump into something more heavy dutty, at the end of the day youl end up with a nice laser machine anyway lol

sorry my english

luc-github, 2017-11-19:
Thanks a lot for sharing ^_^ really appreciate
well original goal is to understand and make one ESP3D for laser and one for CNC - not yet sure it make sense to use ESP3D on those machine, so I need to build each of them to see.
As it will be for ESP3D the electronic will use Smoothieware or repetier as board FW (I am not sure Marlin has CNC/laser capabilities)
I have some grbl boards and shields but as ESP3D does not support them yet it was not my first idea to use them.

About 2/1 well I prefer to get separate machines - luckily I have space in my workspace  - also I prefere not changing the tool holder each time I need to use the machine - finaly use Nema 23/24 for laser I just feel weird ^_^ even it is not an issue

About feedrate I think won't be high feedrate, aluminium is not butter ^_^, and looking at laser cut videos, it need to give time to burn/melt so speed is also low, so you are right

About belt vs rod I guess I will use belt as I would like to have a big working area 1M 
I already meet EMF issue with end stops in the past so now I ground/shield cables

About external drivers for CNC - yes I read that to get high AMP use external is better - like 
[image]
Is it  the same kind that you used ?

Yes do simple then complex would be the smartest choice -but I am waiting more than one year to build them so  I am just afraid I won't do it before next century :stuck_out_tongue_winking_eye: 

May I ask what software do you use ? Because the quality also depend on software part a lot and no plan to use a CNC studio neither Mach 3 or alike (at least at the begining)
I was planned to use the list provided by smoothieware : 
   http://smoothieware.org/software

Is your cnc on picture ?Loks very nice, I read Xengrave It is not the one you build from scrach right ? the casing is very neat,

vMeph, 2017-11-19:
the machine in picture is the machine i have built the X- Engrave sticker is just some Stiker i did to look good lol

the drivers i use looks like the on in pictures i use the DM542 drivers

the software i use for machine control is UGS platform looks like this
[image]
https://winder.ci.cloudbees.com/job/UGS/

for design i use Fusion 360 that is free for Startups and have CAM tools and post processor for GRBL

luc-github, 2017-11-19:
Nice Machine indeed - congratz ^_^
Thanks a lot for these feedbacks - excellent

luc-github, 2018-12-09:
I am still trying to build my system but as now project is part of grbl_esp32 (https://github.com/bdring/Grbl_Esp32) request are more clear for me and process also

SO I think this topic can be closed as no updated for a while

closed 2018-12-09 by luc-github