luc-github/ESP3D #185: ESP3D laser support - please help

luc-github, 2017-11-16:
Hi I would like to add support for Laser engraving - I have 0 knowledge on this so I want to build one to understand what are the needs, how it works and of course test ESP3D.

I have 500mA laser but still have concern about safety. Is belt better than threaded rods ?

I have read a lot on the net but did not found a clear way to go, I have checked http://openbuilds.com but so many model and did not find a full walkthrough 
 
I would like to use aluminium extruded and 3D printed parts. 

- Any suggestion ?
- Anyone built one already ?

Every suggestion is welcome

3d-gussner, 2017-11-17:
@luc-github Hi, i don't have a laser engraver but here some interesting links may help you:
https://www.youtube.com/watch?v=TV6GonFDmSs
https://www.gearbest.com/3d-printers-3d-printer-kits/pp_597310.html?wid=21

https://www.youtube.com/watch?v=cbDS_ba_6j0&t=19s
https://www.youtube.com/watch?v=1Yp_5Utti_A&t=1s
https://www.youtube.com/watch?v=VXOupF-J-6E
https://www.youtube.com/watch?v=BErBHOE05vg
https://www.youtube.com/watch?v=2MfYN7sUF7w
http://lasergrbl.com/en/

luc-github, 2017-11-18:
Thank you, I will watch study all these, very interresting about focus, Tech2C is bery nice chanel - EleksMaker A3 Pro 2500mW seems not expensive for such laser compare to what I saw in France

3d-gussner, 2017-11-18:
The 2500mW looks kind of okay for the price. The A5 with 5000mW is more expensive, but also the laser module is much expensive.
Like the approach from Tech2C using one motion system to 3d print, laser engrave or plot things.

luc-github, 2017-11-29:
Ok I have just  ordered the laser as I have lot of spare parts - so I bought a Focusable 450nm 5500mW laser  with Cooling Fan ^_^ https://www.gearbest.com/3d-printer-parts/pp_490731.html,  I took the chinese plug instead of EU plug and saved 60 euros !!!
I will use it to cut the acrylic / medium parts I will need to build the milling cnc later.

It will be a cartesian cnc with z spindle  also as video show using z movement allow to cut deeper 

Frame is done using 20x20 extruder aluminium (L= 60cm W=50cm H=28cm) - I will use pulley/belt for X/Y and a lead screws for Z because very short distance should be necessary ~10/15 cm maximum, 
Drylin bearing because never used them yet ^_^ and 8mm rods,  
Nema 17 steppers -  I  will see for the board but it will be a Repetier or Smoothie FW , 
I guess smoothieware because it has more users today.

Now designing / 3D printing  the axis parts ... hope done this week end to start wire 

Once all is assembled (hope next week if I got the laser) ,  I will do a complete casing + wifi cam to control for safety purpose

Not sure what can be done to avoid laser to start fire so casing must not be able to burn, I  need to check what cheap material can be used for this purpose (painted aluminium sheets ?)

victorpv, 2017-12-02:
I recently bought a 3.5W laser, 450nm. Have used it successfully with Marlin, by using the Fan control.
 just replaced the head in my 3d printer with the laser.
I am using a metal plate in the bed, the laser can't burn thru it. Any aluminum or steel thin sheet should work.
Talking on a different subject with someone else, he has a 100W laser and can't burn thru any kind of metal, so a 3-5W laser should be safe with any sheet metal as a base.

luc-github, 2017-12-03:
@victorpv thank you for the feedback - yes I think use metal sheets for the sides 
currently base support is MDF due to cost constraint,  but I  hope to change soon for set of aluminium extruded profile which will allow a easier way to maintain mdf / acrylic plates to cut / engrave

May I know what is your optimal focal distance with your 3.5W laser ?

victorpv, 2017-12-03:
The laser lense can be focused, I set it about 40mm over the pieze I'm engraving, but I think it will depend on the adjustment range in yours. A few mm up or down doesn't seem to change the focus much in mine.
Unrelated, do you mind to give me some pointer how to use the tcp to serial bridge? I'm using windows, and tried creating a virtual com port with trueperle. Have tried raw connection with putty to port 8888 and the putty window closes right away. Just a link to some page is more than enough. I'll edit and remove this part later since it's unrelated to the topic, but didn't want to open a new issue for just that question.

victorpv, 2017-12-03:
Forgot to comment about the MDF. My laser can cut wood even at 20% power if it stays in the same spot a few seconds, so if you are cutting anything over the MDF, it will likely burn it. If you can place any kind of metal sheet on top when you are cutting thru, it should protect it.

luc-github, 2017-12-04:
Thank you - I just wanted to know the distance because now working on Z part and wanted to be sure the distance is not too far, now I have a 0.5W laser focusable and the 5.5W is also but not yet arrived.
The 5.5W documentation mention de cutting depth is 3mm so focus should be optimale within tthis range.

About serial bridge it depend how you want to use it - in putty use raw, `<ip:data port>` and you can use serial commanda to query your ESP or the device on ESP, for 3D printer repetier host use directly ip instead of serial, for laser purpose your application may need the serial/tcpip converter you mention.
I have started to play with laserweb (https://github.com/LaserWeb/LaserWeb4/wiki)  which is handy and use websockets, but I did not yet implemented them in ESP3D (should be soon) but there is already a FW available.
In the past I used https://virtualhere.com (it was free at this time) to control my 3D printers and use web cams to monitors prints, but now, I only use SD card printing which is safer  IMHO.

Yes MDF is temporary and plan to protect with painted metal sheet.
Thanks again for the tips

victorpv, 2017-12-05:
Thanks for all the tips. I have been using sd print always, but just recently started designing my own printer board, around an stm32f1, and have yet to get the LCD and sdcard connected and working. SDcard should work, but without LCD I can't do much. But with your esp3d I can :) I even added a connector specifically for an ESP01 since I have wanted to use esp3d for a few months now, but the board I had didn't have any serial pin available.

luc-github, 2017-12-05:
Well this is out of topic but for printer without extra serial available I was thinking of ESP connected to usb port and acting as pass through if need to connect to PC or as Wifi bridge, the power of ESP would come from the in USB (pc or ac adaptor or cable from printer itself(- if you can do you own printer board this should be easy for you  ^_^

I envy electronician guys like you - you can do a lot :smile: 

Back to the topic : as you have laser machine - any idea of benefit/feature  of ESP3D for laser ?

victorpv, 2017-12-05:
Perhaps an option to turn on and off the laser, similar to the current buttons to move the axis 1mm, 10mm, etc?
As far as I have seen so far, most printers control the laser with M106 and M107 which are for the Fan, but real CNC use M3-M5. Marlin includes support for M3-M5, but I still need to test that out.
Or perhaps instead Fan speed control (using M106 and M107), that would allow to control a Fan, or a Laser if the pin is setup as Fan.

luc-github, 2017-12-12:
Ok machine is mostly done - used RE-ARM + ramps 1.4 + Viki2 lcd ( I was not able to make the azteeg mini  stepper to work 😢 )
I use smoothieware cnc version - I tried the cheap laser 500mW I had, but unfortunatly it worked 10 min then died...
  Need to wait for the big one to do more test now...

I think I need to build another small machine to test marlin/repetier board or even grbl arduino uno as behaviour is easier to see

Homing on Smoothie forCNC is very different than 3D printing need to use `$H` to move then `G28` to set origin and this work only for X Y not for Z - Laser use `fire nnn` and `G1 Sxxx` 
I have a some learning curve to follow
And I guess Marlin / Repetier / grbl arduino uno will have diffrenet behaviour...
Good challenge indeed 😸

luc-github, 2017-12-17:
@victorpv - thank you for sharing 
I just finished re-build an i3 prusa I had in spare parts - using ramps1.4 + Mega + Graphical LCD - I will wire all and test the way you describe
The inkscape generation is done by a plugin - right ?

victorpv, 2017-12-17:
Right, I think one plugin is included and the other I had to add.
One is to cut outlines, it will move the head around the outlines, and can be set for a number of passes, like 1, 5, 10 passes, depending it you want to cut thru, and how thick the material is. Can be used to cut shapes.
The other is a rasterizer, the head will move side to side turning on and off, then moves the bed one line further, and the head moves again side to side. That's the one for engraving.
There was a video in youtube that showed both, and is the one I followed, if I can find it I'll link it.
The outline plugin is called J Tech photonics. The one to rasterize is from 305 Engineering.

There is another one to generate puzles, and another to generate boxes, but haven't used either yet.

One point of caution, dont pause or kill you print midway, cause it can leave the laser on in the spot where it stops, better to have a kill switch just for the laser so you can turn it off if something happens.
I think it would be nice if Marlin had some safeguards for that, but at the moment it doesn't.

If you want an STMF1 printer board, let me know and I'll send you one from my next rev. The current rev works fine except I placed the LCD adn SD card connectors too close and you can't plug normal ribbon cables. I use it with dupont wires without an issue though. But everything works, including the ESP01 port for ESP3D ;)

luc-github, 2017-12-17:
Thank you, will check that, I have just installed https://github.com/cnc-club/gcodetools but did not used it yet, will check your solution first. 

For the pause you can add a macro when you pause there is a macro you can use as you use repetier host 
go to Config / Printer Settings / Scripts and select `run on Pause`  you can add a command to stop laser

For the board I  do not want to bother you but thank you if you do

Actually as you use Marlin 2.x I have noticed an issue but it is with RE-ARM + Ramps
can you confirm you can (or cannot  like me) delete a file using M30 command on your STMF1?

With Re-ARM it always failed when it is ok with Mega/ramps

victorpv, 2017-12-17:
I'll check it up. I did notice that the filenames show as 8.3 in serial, while the LCD shows the full name. Did I miss some config option to get full names in serial?

luc-github, 2017-12-17:
Well Marlin actually does not really support long file name unlike other FW (including Marlin Kimbra)

`M20` list file name in 8.3, Gcode commands use 8.3 not long nam,e but you can display long file using `M33 <path>` (http://marlinfw.org/docs/gcode/M033.html) for each path generated by `M20` command which is really time/serial consuming when it is not used by FW actually - so I currently stick to 8.3 for Marlin for consistency and save communication bandwitdth.

victorpv, 2017-12-22:
Sorry I forgot to test the M30 command. Just turned on the printer, deleted a file, and the next listing shows the file is no longer there. Rebooted the printer to doublecheck and the file is gone for good.

openbuilds-engineer, 2018-01-17:
Luc, you can use Github.com/LaserWeb/LaserWeb4-Binaries as CAM, fully Smoothieware compatible (;

Also, see http://openbuildspartstore.com/openbuilds-acro-system/ next time you need a kit (;  (In reference to your first post in this thread at the very top hehe)

luc-github, 2018-01-17:
@openbuilds-engineer thank you for sharing

yes I use Laserweb4 for smoothieware but I still wait for the laser from gearbeast ....

luc-github, 2018-02-19:
Thanks for everyone feedback was very helpful - I think I can close issue now as feature will be in 2.0 which has a TODO list : I close issue as it is in TODO list #179

closed 2018-02-19 by luc-github