luc-github/ESP3D #186: OLED support

luc-github, 2017-11-26:
Now Oled LSC can be very cheap (2$) it can be nice to display IP / wifi status / signal on it - if printer do not have screen or to have always information available

I plan to use https://github.com/squix78/esp8266-oled-ssd1306 because I tested it - but open to another if exists

luc-github, 2017-11-26:
[image]

luc-github, 2017-11-26:
Here a possible UI : 
Line 1 : SSID used and signal if station mode - SSID and access Point icon if AP mode
Line 2: SSID
Line 3: Message
Line 4: Message or progression bar
[image]

shaddow501, 2018-01-11:
Dear Luc

I have made a case for Lolin ESP32 OLED if you are interested with 3D print.

https://www.thingiverse.com/thing:2752734

luc-github, 2018-01-14:
Note: 
Heltec wifi kit 32 with embeded oled screen need extra init : 
```
 //For Embeded OLED on Wifi kit 32
   pinMode(16,OUTPUT);
   digitalWrite(16, LOW);   // turn the LED on (HIGH is the voltage level)
   delay(100);                       // wait for a second
   digitalWrite(16, HIGH);    // turn the LED off by making the voltage LOW
```
the standalone oled screen I have do not need such extra initialization but better to plan it with a flag

luc-github, 2018-01-17:
Test skech to evaluate rendering show that SSID can over load screen so need a banner rendering function that do not rely on delay / yield to be used with async webserver - it should not be linked but better plan it
So IP and SSID have exchanged positions to meet the screen size constraint:
[image]
[image]

luc-github, 2018-02-19:
I close issue as it is in TODO list #179

closed 2018-02-19 by luc-github