luc-github/ESP3D #193: Diagram of HY_BASE_L V1.0

arcidodo, 2017-12-12:
Hi there,

i bought an Infitary m508 printer but cant find any information about the board (HY_BASE_L V1.0). can someone tell me where the RX/TX is located? 

[image]

https://www.alunar.net/m508-self-assembly-3d-printer.html

luc-github, 2017-12-12:
Looks like MKS base depopulated - what are the 8 pins close to the Z end stop cable ? looks like the Aux 1 connector

arcidodo, 2017-12-12:
thats D12, D11 & A12, A11

ekaggrat, 2017-12-12:
i dont think they have broken it out.. you will need to solder it directly from the chip if your soldering skills are good enough

luc-github, 2017-12-12:
[image]
what about the others free connectors ?

arcidodo, 2017-12-12:
there are 2 ICSP ports on it, the 2 connectors below the USB connection. and the pins above the LCD screen port are D12, D11 & A12, A11

luc-github, 2017-12-17:
did find out ?

luc-github, 2017-12-25:
Close it as no update - feel free to reopen if new info

closed 2017-12-25 by luc-github

henri98, 2018-11-06:
Any updates on this?

azoul21, 2020-02-04:
moi j'ai un problème avec cette carte mère car quand j'imprime des grand fichier d'une dure de 20H la moindre  coupure de courant l'imprimante revient a imprimer a zero il n’y pas de solution a ça merci