luc-github/ESP3D #194: ESP32

shaddow501, 2017-12-13:
Hello

I am trying to implement your code with the ESP32 lolin module.
I have a few questions:

1. Which one should I program? should I download the https://github.com/luc-github/ESP3D/tree/master/esp3d, and just upload the ino file? cause the last time I did that with the ESP8266 and uploaded the new Web_GUI files the module stopped working... So I am a bit confused.
2. I am tring also to implement the writing on the screen, that will write all the info that it write on the serial port on screen.

Where in your code can I find the location of how you send the serial.print data ? at which file is it located?
I only want it to print the Wifi station that it is connected to, the IP address and status.

Can you give me some directions?

luc-github, 2017-12-13:
Sorry it is not clear, you talk about ESP32 and you said it block on ESP8266, I am not sure what you mean

ESP32 + EP8266 support is on https://github.com/luc-github/ESP3D/tree/devt
when ESP8266 only is on  https://github.com/luc-github/ESP3D/tree/master

shaddow501, 2017-12-13:
Yes, last time I used the esp8266 and when i tried to upload the web ui archive it didnt let me login anymore, than you said I need to program version 0.62 or something.
Now I have switched to esp32 lolin module and I am not sure what to program.

How can I implement this :

Line 1 : SSID used and signal if station mode - SSID and access Point icon if AP mode
Line 2: IP Address
Line 3: Message
Line 4: Message or progression bar
[image]

luc-github, 2017-12-14:
this is https://github.com/luc-github/ESP3D/issues/186 and did not thinked much about it - but I guess need to be implemented in bridge class with a way to select the targeted lines (0 1 2 3 ) also some wrapping function like updateSSID() updateSignal() updateIP() UpdateStatus()   

Did not really spend time on it yet

luc-github, 2017-12-25:
Close it as not issue

closed 2017-12-25 by luc-github