luc-github/ESP3D #196: Add support for grbl FW

luc-github, 2017-12-19:
For CNC purpose better to support https://github.com/gnea/grbl

vMeph, 2017-12-19:
would be awsome control grbl arduino wifi mode

vMeph, 2017-12-19:
@luc-github is there any way esp could be adapted to GRBL the most used software to parse Gcode to GRBL
https://winder.ci.cloudbees.com/job/UGS/
wonder if CNCs gcode jobs could be send from UGS into ESP connected to GRBL arduino uno wireless

luc-github, 2017-12-20:
you mean to put the universal GCODE sender to ESP firmware ?

I have no idea as grbl fully new for me

About using UNO Wifi not sure it is possible according : https://github.com/esp8266/Arduino/issues/3004

I need to check anyway - I have uno + cnc shield but no wifi uno - I need to find one or a clone like https://robotdyn.com/catalog/boards/uno_wifi_r3_atmega328p_esp8266_usb_ttl_ch340g_micro_usb/

vMeph, 2017-12-20:
Yes universal gcode send gcode to esp plug to the grbl tx rx lines,
Not wifi arduino, a normal arduino uno, they do this with bluetooth module so i would say you could do with esp

vMeph, 2017-12-20:
What cnc shield you have?

luc-github, 2017-12-20:
A standard one I think (cnc shield)

I found a wifi uno clone [here](https://www.aliexpress.com/item/UNO-WiFi-R3-ATmega328P-ESP8266-32Mb-memory-USB-TTL-CH340G-Compatible-for-Arduino-Uno-NodeMCU-WeMos/32843050202.html?spm=a2g0s.13010208.99999999.262.wRxXK8)

And I see there is an ESP32 pinout compatible version of  uno : [here](https://www.aliexpress.com/item/For-Wemos-D1-TTgo-ESP32-WiFi-Bluetooth-4MB-Flash-UNO-D1-R32-Board-Module-CH340-CH340G/32839196190.html?spm=2114.search0104.3.194.iNVfkH&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10065_5000015_10151_10344_10068_10345_10342_10343_51102_10340_10341_5130015_10609_10541_10084_10083_10305_10304_10307_10306_10302_10610_5870020_5080015_10312_10059_10313_10314_10184_10534_100031_10604_10603_10103_10605_10594_5060015_10142_10107,searchweb201603_32,ppcSwitch_5&algo_expid=f61e0d55-f1b1-4b70-adb3-c535d6650372-29&algo_pvid=f61e0d55-f1b1-4b70-adb3-c535d6650372&rmStoreLevelAB=5)

It may be interresting to test

vMeph, 2017-12-20:
If you have the cnc shield v3 and you want to use grbl 1.1 with the new features you have to hack the shield, that shueldield was done for versions grbl 0.8 and the new version there was a swap of pins 11 and 12, to adapt can be easy just bend pins 11 and 12 of the shield the solder to cables on the bottom of arduino pin 11 and 12 

https://github.com/grbl/grbl/wiki/Connecting-Grbl

luc-github, 2017-12-20:
Thank you for the information - seems I have 2 V3.00 from protoneer

luc-github, 2017-12-20:
May I ask why doing an hardware rework ? why not just invert values here : https://github.com/gnea/grbl/blob/master/grbl/cpu_map.h#L65-L68

vMeph, 2017-12-20:
they change cause the spindle was on pin 12 that dont have PWM so they change to pin 11 for have PWM

luc-github, 2017-12-20:
Yeah my bad I misread the board I saw all digital are PWM when actually only the ~ pins are ... orz

luc-github, 2017-12-20:
for what I understand reading internet docs the best is : 
* ESP is connected to CNC shield using rx/tx 
* ESP has SDCard reader connected
* ESP stream SD card content using Streaming Protocol on Rx/T: Simple Send-Response as suggested by
https://github.com/grbl/grbl/wiki/Interfacing-with-Grb
* Monitor Gbrl is also described in previous link

Hosting UGS which size is 12 MB and java is not really the way to go I think  - current ESP-WEBUI is lighter and can be modified to fit the need

vMeph, 2017-12-20:
looks good i just mention interaction  with UGS sense is a platform with  alot tools like autolevel, probing etc..
and many many people use it,

luc-github, 2017-12-20:
I think UGS should be used from PC using Serial ->TCP drivers (I do not see network feature on it) and then access to ESP3D using data port which forward to CNC shield

vMeph, 2017-12-20:
I wonder if @winder could add network features on UGS

luc-github, 2018-01-01:
As note for the printer settings tabs:
the `$$` command display all settings
```
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
```
to change value 
`$0=<new value>` will set new value

luc-github, 2018-01-10:
I have CNC V3 + uno but  I see there are also CNC shield for Nano used for laser cnc - so I have ordered one to check

luc-github, 2018-01-11:
I am thinking to add host feature on ESP32 for GRBL FW
Currently there is no SD card support for what I can see on GRBL 
so by connecting ESP32 + SD card reader and ESP reading SDCard (I prefer no using SPIFFS currently) and even a screen and an encoder and the machine do not need to be connected to a computer anymore 
would be a cheap solution

luc-github, 2018-02-19:
I close issue as it is in TODO list https://github.com/luc-github/ESP3D/issues/179

closed 2018-02-19 by luc-github