luc-github/ESP3D #198: EEPROM read error

jamesarm97, 2018-01-03:
I am trying to get this working on the Wemos D1 mini Pro version with 16mb of flash data. Where would I look if I enabled debug and the firmware is outputting an EEPROM read error and some other stuff then resets over and over with the same error? Is there something I need to do to get EEPROM working? Flash real size: 16777216  16MB (15MB SPIFFS)

jamesarm97, 2018-01-03:
This is with it not connected to anything. Is it trying to read EEPROM from the host or on the ESP itself?

Debug Serial set
**Error no EEPROM access**
M117 ESP EEPROM reset
Chip ID: 8426313
CPU Frequency: 80Mhz
Free memory: 38.71 KB
SDK: 2.1.0(deb1901)
Flash Size: 0 B
Available Size for update: 4.00 GB(Ok)
Available Size for SPIFFS: 1.00 GB
Baud rate: 115200
Sleep mode: Modem
Channel: 1
Phy Mode: 11g
Web port: 80
Data port: 8888
Active Mode: Wifi Off
Captive portal: Enabled
SSDP: Enabled
NetBios: Enabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: ???
Debug: Debug Enabled :serial",FW version: 0.9.99
**Error write byte**
Restarting

Debug Serial set

luc-github, 2018-01-04:
I do not have this board so I cannot duplicate the issue 
The EEPROM is a  4KB area in flash shared with SPIFFS and FW
May be there is a problem with the 16MB ?
I guess you use the 2.4 core, can you try the EEPROM samples ?

https://github.com/esp8266/Arduino/tree/master/libraries/EEPROM/examples

Another reason would be another EEPROM library is used instead of ESP8266 one

jamesarm97, 2018-01-08:
I will check the eeprom library used. The sample code to check for the reported flash size shows the numbers correctly (all but the IDE expected size). could it be a variable size overrun on the 4gb / 1gb issue?

luc-github, 2018-01-08:
Flash size is get by : `ESP.getFlashChipSize()` and it seems getting 0 so this is wrong

Available for update is get by : 
```
uint32_t  flashsize = ESP.getFlashChipSize();
if (flashsize > 1024 * 1024) flashsize = 1024 * 1024;
BRIDGE::print(formatBytes(flashsize - ESP.getSketchSize()).c_str(), output);
```
something is wrong as flashsize is seen as 0 , max size for flashing is 1M so cannot be 4 GB => this part is definitly an overflow

also SPIFFS is get by : 
```
 fs::FSInfo info;
 SPIFFS.info(info);
 BRIDGE::print(formatBytes(info.totalBytes).c_str(), output);
```
and give 1 GB but you have 16 MB max of flash so her is wrong also from SPIFFS function

jamesarm97, 2018-01-08:
Thanks. I am digging deeper. Looks like the totalBytes and usedBytes are not being filled out properly. I'm guessing due to the SPIFF error:

SPIFFS_info: rc=-10024, err=-10024
TB: 4277137406
UB: 4277137406
BS: 8192
PS: 256
MO: 5
MP: 32
Flash real id:   001840C8
Flash real size: 16777216

Flash ide  size: 16777216
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

snippet:
    fs::FSInfo info;
    SPIFFS.info(info);
    Serial.printf("TB: %u\r\n",info.totalBytes);
    Serial.printf("UB: %u\r\n",info.usedBytes);
    Serial.printf("BS: %u\r\n",info.blockSize);
    Serial.printf("PS: %u\r\n",info.pageSize);
    Serial.printf("MO: %u\r\n",info.maxOpenFiles);
    Serial.printf("MP: %u\r\n",info.maxPathLength);

jamesarm97, 2018-01-08:
Ok, numbers look ok now with the test program. I didn't call the SPIFFS.begin. With that I now get good numbers so I will see if the ESP3d program wasn't calling SPIFFS.begin for some reason.

SPIFFS Opened
TB: **14900866**
UB: 502
BS: 8192
PS: 256
MO: 5
MP: 32
Flash real id:   001840C8
Flash real size: 16777216

Flash ide  size: 16777216
Flash ide speed: 40000000

luc-github, 2018-01-09:
SPIFFS.begin is done here : https://github.com/luc-github/ESP3D/blob/13ee5e19e834fbfebd8652d5ee08e73d3b06eefd/esp3d/esp3d.ino#L143-L147

So nothing should prevent to launch it .
Unfortualy I do not have ETA when I will get the board to reproduce issue -I  hope will get it within 2 weeks.

luc-github, 2018-01-23:
I just received my board today, I have updated ESP core with today git 
I got several WDT reset at the begining - not yet checked the root cause but size and uploading SPIFFS is working : 
[image]

jamesarm97, 2018-01-23:
I believe I found the same thing, it resets a few times then settles out.

luc-github, 2018-01-23:
I disabled the serial debug and Ido not have any reset - I will keep watching

I am guessing the long boot and original reset was due to SPIFFS first format

luc-github, 2018-01-25:
Root cause seems the long time of formating SPIFFS and currently there is no API to track it - may be in next version of SPIFFS.

So I guess @jamesarm97 Issue can be closed ?

jamesarm97, 2018-01-25:
Seems to be working ok now and know to just except a long boot / WDT on a new install.

luc-github, 2018-01-26:
I think it is same as  https://github.com/esp8266/Arduino/issues/4237 and no fix currently just wait until format is complet

luc-github, 2018-02-19:
I think issue can be closed now

closed 2018-02-19 by luc-github