luc-github/ESP3D #202: Possible to connect ESP32 directly to USB port on printer?

cymen, 2018-01-15:
I have an ESP32 (Lolin32 Lite from Wemos: https://www.aliexpress.com/item/WEMOS-LOLIN32-Lite-V1-0-0-wifi-bluetooth-board-based-ESP-32-esp32-Rev1-MicroPython-4MB/32831394824.html). Can I hook that up to my Marlin-based printer via USB and use ESP3D to talk to it? Or does it need a direct serial connection?

Thanks!

luc-github, 2018-01-16:
I do not think 
1 - Printer is not PC so if printer USB is not an USB host it won't communicate
2 - The printer does not provide power on USB, because not USB host, so ESP won't be powered.

luc-github, 2018-01-16:
I guess they read the wiki : https://github.com/luc-github/ESP3D/wiki 😄

cymen, 2018-01-16:
I did read the wiki. I get it now though -- the basic idea is the ESP is connected to the boards serial directly, right? The wiki is missing the general pattern and dives into each board but I don't have one of those boards. Thus why I am asking here!

But it makes sense now I think.

cymen, 2018-01-16:
I have a Creality CR-10S which has a variant of the Melzi board. It looks like this is how to get access to the serial:

http://reprap.org/wiki/Melzi#Wiring

I am going to open mine up again and compare to the photos to see if it is the same. Thank you! I was just confused a bit about how people were connecting this but now it makes a lot more sense.

For firmware, I am running Marlin 1.1.8.

closed 2018-01-16 by cymen

nonflammable, 2018-11-26:
I think it is possible with MAX3421E. You can use ready boards available here:
https://www.aliexpress.com/af/usb-host-arduino.html?SearchText=usb+host+arduino
There is library:
https://github.com/felis/USB_Host_Shield_2.0
Now it remains to combine one with the other ;)
Hereby I am proposing an enhancement.

schorhr, 2019-05-19:
Hello :-)
Sorry if I missed some detail (English isn't my first language), but is it possible or even planned  to implement something like this USB host (https://github.com/tobozo/ESP32-3DPrinter-Bridge) to avoid having to solder to Creality boards? 

I have a low budget school workshop and it would rock to just plug and play instead of having to solder on each printer.

Anyway, thank you for your awesome work! <3

luc-github, 2019-05-19:
Could be a good addition indeed

luc-github, 2019-05-19:
I have added it to 3.0 todo list (https://github.com/luc-github/ESP3D/issues/239#) thank you

schorhr, 2019-05-20:
No, thank You ;-) That would make it so much easier to get the cheap printers into the network <3 Can't solder to vias or SMT pins with my shaking hands very well.