luc-github/ESP3D #203: Support to DHT11/22 temp.-humidity sensor

Gi0cas, 2018-01-20:
Hi, i really like ESP3D, but i ask if it is possible to add support to temp/humidity sensor DHT11 or DHT22.
It would be good a button in the UI that shows real time temperature and humidity in the printer box (if i decide to close the printer inside a box, to print ABS for example).
Arduino has a some library, and sensor is really simple and use only 1 GPIO pin. 

It is a doable request?

luc-github, 2018-01-20:
Yes I know these sensor - I have some so should be Ok for 2.X

Gi0cas, 2018-01-20:
Thanks Luc, do you have an approx ETA for esp3d 2.X ?

luc-github, 2018-01-20:
hummm this year - but feature will be available one by one 
I have added DHT sensor to list : https://github.com/luc-github/ESP3D/issues/179

luc-github, 2018-01-20:
I put after the Wsocket to see if we can use it instead of polling temperature or even push button to get the  data

luc-github, 2018-02-19:
I close issue as it is now in Todo list https://github.com/luc-github/ESP3D/issues/179

closed 2018-02-19 by luc-github