luc-github/ESP3D #204: ESP3D access point issue

dshrpv, 2018-01-31:
Dear Luc,

First of all I want to say great thank to you for this **amazing project!**
I flashed ESP8266 ESP-01 1Mb in accordance with your instruction and video on YouTube.
But I can`t connect to ESP3D access point for initial configuration.
It not accepting the password specified in config.h tab.
Moreover I`m found that after flashing FW and FS the access point not changes SSID.
ESP shows previous AP name (from sample sketch flashed before) if it is different with specified in config.h tab.
Tried different ESP8266 ESP-01 1Mb modules, on 3 different PC`s, different AP names, etc. with no result ((( 
Win7 and Win10, Arduino IDE 1.85, ESP8266 Library 2.4,  ESP3D SW V0.9.99

Please advise how to fix it.

Thank you very much in advance!

luc-github, 2018-01-31:
the first boot detect EEPROM settings are not valid so it save all default settings to EEPROM, this settings are used on next boot to configure ESP3D, so even you flash with different settings after it is still the EEPROM settings which are used.

To change after you have 2 ways using https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands as you cannot connect to web server:
* `[ESP444]RESET` then restart this will use the defaults settings you modified to re-create EEPROM
* `[ESP105]<SSID>` then restart - it will change only SSID
* `[ESP106]<Password>` will only change password - need also a restart to apply 

I have noticed sometimes  windows do not allows to connect, I never found why - but it always works from Android, Linux, OSX 
You can try the test AP sketch to verify you can connect : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

But if you cannot, you can use the https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands to configure ESP3D as Station client joining and AP using ESP 100 / 101 / 103 / 104 commands

dshrpv, 2018-02-01:
Dear Luc,

The issue solved.
I passed AP authentication and made initial setup in web interface.
I will try connect the ESP module to RAMPS this weekend. 
Thank you very much for this project and quick support!

Wish you all the best!

BR
Denis

closed 2018-02-01 by dshrpv