luc-github/ESP3D #205: Support JSON M408/RRF

svs072477, 2018-02-14:
Is it planned to support the work through the JSON, and RepapFirmware?

luc-github, 2018-02-14:
What do you mean ? can you elaborate ?

Currently ESP3D do not handle almost any GCODE code it is just a bridge between printer and host/UI

Current Web UI do not use M408 to querry printer but nothing prevent to send M408 command and get answer

About reprap FW support, it support natively ESP8266 with its own UI as Duet wifi has an esp8266 embedded - so not sure what the point for ESP3D but I can be wrong

Version 2.0 is ongoing and any suggestion is welcome to discuss

svs072477, 2018-02-15:
It was meant to add the ability to exchange information with the printer by sending the M408, and display the response in the web interface after parsing JSON.
I just use RepPapFirmvare not on Duet3d but on this board:
http://3dtoday.ru/blogs/svs0724/ruramps4d-ramps-for-adruino-due-the-pilot-shipment-available-for-zack/
The standard web interface used on Duet 3D does not work on it.
Now I use ESPRINTER 3D, but the project has not been developing for two years.
In addition, the support of JSON is already included in Repetriere Firmware and MK4Duo

svs072477, 2018-02-15:
And the fact that you have the support of ESP32 is generally wonderful

luc-github, 2018-02-15:
Ok So actually the issue is here : https://github.com/luc-github/ESP3D-WEBUI
In the ESP3D FW I have tried to removed 99% of GCODE and GCODE management, 
M408 command should work but it is not currently used by UI, I did not used because it is not implemented Marlin so current UI cannot rely on it to be generic,
That said nothing prevent to use it for ESP3D-WEBUI 2.0 for all but Marlin as there are lot of information to set initial values only. Because query often M408 when not heating  / not moving may not be an optimal solution to avoid to overload ESP / printer.

About reprap support reading at GCODE list some answer format are not recognise like M105 for temperature but this should be easily fixed so I can add it to ESP3D/ ESP3D-WEB UI 2.0 list : https://github.com/luc-github/ESP3D/issues/179

I will have to build a reprap system -  but RADDS is little bit expensive

The RURAMPS4D board looks definlty nice, may I know what is the price ? You designed it  from what I can understand (my russian is 0 so I used google 😉,  it need to order from you - right ?
Do you send out of Russia (France) ?

svs072477, 2018-02-15:
Yes, I am a developer of RURAMPS4D, at the end of the month a new boards will be released from production. In Europe, the experience of sending was. I sent it by air mail.
The board is now assembled in 2 versions:
With the ability to connect two thermocouples k-type (installed AD8495) - 60 euros
And without this option - 50 euros

**The RURAMPS4D board provides the following connectivity’s:**
 
```
Work with Arduino Due only
Supply voltage: 12 to 25V, on-board DC-DC converter output 9V 3A, and 5V 0,8A for Displays.
Separate power for the Heated Bed and the rest of the electronics
6 Stepper motors connectors on-board : X,Y,Z, E0, E1, E2. 
4 connectors for extrnal stepper driver: X, Y, Z, E3
7  MOSFET`s:  2 for Fans, 3 High power (IRLR2905) for Hot End Heaters, 2 Very High power (IRLS3034) for Heated Bed (when connecting two mk2b to one mosfet, it remains cold)
Multizone Heated Bed support (like this http://www.cheap3d.ru/shop/Zapchasti-dlya-printerov/Aevuhzonnaya-plata-nagreva-Cheap3D-360Vt-poaeogrevnoi-stol-300h300mm--1327.html)
LCD: standart Reprap Discount Smatr/FullGraphic Display, MKS TFT Dislpaly, Nextion Display. 
6 endstops (Xmin,Ymin,Zmin,Xmax,Ymax;Zmax)
Connector for Inductivity or capacitor Z-Probe sensor
5 thermistors 
optional 2 termocouples k-type
2 servomotors
Bl-touch (connecting to servo and endstop connectors)
I2C, SPI.
Connector for Wifi module ESP-01
 
Features:
Barier terminal connectors for Power supply (30A)
Catch-diodes on the MOSFET`s
Car-fuses instead of thermo fuses
EEPROM
 
Firmware
MK4Duo
Repetier Firmware
Marlin 2.0
RepRap firmware (now porting)
for wifi module: ESP3D, ESPRINTER.

```
In the future, a version with a microcontroller and an Esp32 on the board is planned.
I will wait for the release of ESP3D-WEBUI 2.0
My English is also not very good, and I also use Google translate :)

luc-github, 2018-02-15:
Nice - what step driver resolution/type is supported ? 
50 euros is including air mail, I am correct ?

svs072477, 2018-02-15:
Step drivers:
drv4899
drv8825
lv8729 (1/128) microstepping
tmc21xx
Connection SPI to the TMC2130 while only wires, in the future I think I will place it on the board.
Air Mail cost 6 euros, but to you as a developer of Esp3d I will make free shipping :)
If you have any other questions, write on email svs072477@yandex.ru

luc-github, 2018-02-15:
Great, excellent  - thank you  - I will contact you

luc-github, 2018-02-19:
I close issue as it is in TODO list https://github.com/luc-github/ESP3D/issues/179

closed 2018-02-19 by luc-github