luc-github/ESP3D #206: Using with MKS TFT Wifi

vader333, 2018-02-18:
I've setup the custom firmware with my MKS TFT Wifi (ESP-12S). The web UI shows up nicely, but I can't get it to interact with the board. The baudrate is the same (115200) and it's using data port 8080 as per the manufacturer's specification. Is there anything else I can try to get it to work?

luc-github, 2018-02-18:
How did you connected MKS TFT  and ESP ?
MKS use serial RX and TX and ESP need also RX and TX serial to communicate with board so I wonder how did you connected all

luc-github, 2018-02-18:
I am guessing you use the WiFI connector of the MKS TFT - which is not the serial of the printer board
 for what I can see the MKS TFT try to establish communication with the wifi module using AT command:
`at+net_wanip=?` to get the IP address, but as ESP do not answer to this command the TFT says is disconnected and/or generate exception and so to not forward any command from WiFi serial to Printer Serial.
That is why ESP cannot communicate with printer
Possible solutions are: 
1-  to use same serial pins as TFT, not sure it is wise to have 2 host at once
2 - simulate the mks wifi behavior in ESP3D to cheat TFT board hoping there is only few commands - MKS FW is not open so need to spy the data
3 - check TFT settings to see if there is way to force the Serial forward

vader333, 2018-02-18:
I see. I'm afraid I'm not very good at hacking. Are there any questions I can ask the manufacturer (other than asking for the source code of their wifi module) that can help you develop a solution? I can speak chinese, and they are more responsive on the chinese channels.

Another idea would be to flash the stock firmware to the mks wifi module, and connect it to a debugger and see what kind of output it produces. Would that work?

luc-github, 2018-02-18:
The issue is not in the wifi module but on TFT part which do not bridge the Serial WiFi to Serial Printer if the  TFT think there is not WiFI module

The only question is the the last point I suggested - find a setting which allow Serial Bridge without WiFi module present.

Currently the only solution is to duplicate TFT RX/TX pins to be used by TFT and ESP

I have MKS-TFT + MKS-WiFi I could check what need to be added in ESP3D FW to Cheat TFT but I am little bit busy - so I can add it to ESP3D 2.0 Todo list but cannot say if/when it will be added - Sorry

vader333, 2018-02-19:
It's not urgent, as I'm moving away from MKS Sbase to Duet Wifi or Azteeg now. But thanks for helping me understand!

closed 2018-02-19 by vader333

jamesarm97, 2019-04-10:
I would like to know the same. I have used the ESP3d interface before and love it but now I have the MKS TFT with Wifi and would love if it would work on it some day. I guess the main issue from what I have read is the ESP3d expects to just send gcode directly to the main board but the MKS TFT wifi is sending special commands instead.

luc-github, 2019-04-10:
latest status is here : https://github.com/luc-github/ESP3D/issues/249
still ongoing as I am busy  other projects also