luc-github/ESP3D #207: no

Gi0cas, 2018-02-24:
Hi, i flashed ESP3D on my sonoff basic.
I want to create a button to put ON/OFF my printer (puttin pin12 HIGH or LOW).
I created a button, with a macro, for example "/macro1.g".
I created a macro with notepad, with only the text [ESP201]P12 V1, but when i press the button nothing happens.
On the bottom side of the screen, in the "commands tab" appears : [ESP700]/macro1.g, and "ok" on the next new line. But it not works !!

I noticed that if i write this command in the "commands" tab : [ESP201]P12 V1, it works. If i write it on serial, the message that appears is "M117 Incorrect Cmd". How i can set my button to put a pin high or low?

luc-github, 2018-02-24:
Which version of ESP3D are you using ?

Gi0cas, 2018-02-24:
esp 0.9.99

luc-github, 2018-02-24:
where do you save the file ?

Gi0cas, 2018-02-24:
i upload it on spiffs, it is a simple text created with notepad, with only the text 
[ESP201]P12 V1

luc-github, 2018-02-24:
this is weird  -  it should work  
I will investigate

Gi0cas, 2018-02-24:
Ok, it sounds like this command can't be understood by serial. When i write the command in the "command" tab, it works, if i write the same command in the serial, serial tells me that the command is incorrect. It seems like the serial not recognize some commands, because other commands like [ESP800]<header answer>, works also in serial.

luc-github, 2018-02-24:
I found out there is issue reading file line by line - looks API changed because it was working before...
I am working to a fix - should be ok tomorrow.

In serial it depend how you enter it and need return cariage to be catched

Gi0cas, 2018-02-24:
Okok, you are awesome... fixed this error in no time, great !!

luc-github, 2018-02-24:
should be ok fixed with https://github.com/luc-github/ESP3D/commit/2714904c52a4ed3b735928f9af0e317a352d4730

luc-github, 2018-03-01:
@Gi0cas  can you confirm the fix solve your issue ?

Gi0cas, 2018-03-02:
No, the issue persist.
I attach some photos on the bottom of this reply, this is the exact procedure i followed to create/use a button:
1- create macro button, name it M1, link to /M1.g
2- create a new M1 text file with only "[ESP201] P13 V1"
3- Upload m1.g to spiffs
4- Click on button, nothing happen. In the Command section appears "ok", but nothing works really.
5- Checked to send [ESP201] P13 V1 from Arduino IDE serial, it tells me that not understand command

[image]
[image]
[image]

luc-github, 2018-03-02:
I have connected a led on pin 13 of one nodemcu 0.9  - which have UI: 0.9.81LE / FW: 0.9.100

I am not able to reproduce the terminal issue - when I type [ESP201]P13 V1 or [ESP201]P13 V0 it is working and no error

But I see what is wrong on your macro - SPIFFS is case sensitive - you upload m1.g but you use M1.g
I agree the error management is not good but your issue come from this - change to same filename that you upload and it should work - as it work on my side

Gi0cas, 2018-03-02:
I corrected it, but not works. If you want i can make a video of complete process and the issue.

luc-github, 2018-03-02:
are you sure of your pin mapping ? first you said 12 then now you say 13

Gi0cas, 2018-03-02:
Yes, now i tried with pin 13 that on Sonoff rapresent the LED. I am taking a video of complete issue.

luc-github, 2018-03-02:
I just soldered connector to a sonoff module and flashed it
it works on my side with one comment : 
[ESP201]P13 V1 make led off
[ESP201]P13 V0 make led on

Gi0cas, 2018-03-02:
Yes, so if all works on your sonoff there is something wrong in my procedure. I am uploading (sadly slow) a video of my complete process with relative error.

luc-github, 2018-03-02:
did you tried : [ESP201]P13 V0 ?

Gi0cas, 2018-03-02:
[ESP201]P13 V0 and [ESP201]P13 V1 works good only if i type expressly in the "commands tab", but if i create a file to launch with macro button not works.
In the video i am uploading is visible.

luc-github, 2018-03-02:
can you try uploading my files ?
Please do not edit them in notepad+
[macro.zip](https://github.com/luc-github/ESP3D/files/1775981/macro.zip)

Gi0cas, 2018-03-02:
Uploaded to spiffs your file and LED on pin13 not works when triggered by macro button.

Gi0cas, 2018-03-02:
Please wait the link to the video, it shows clearly the problem

Gi0cas, 2018-03-02:
See there : https://youtu.be/AIQAJpjCiXM

luc-github, 2018-03-02:
can you share the version of FW UI ?
[image]

luc-github, 2018-03-02:
and which browser do you use ?

Gi0cas, 2018-03-02:
[image]

Google Chrome
Windows 10, 64bit

luc-github, 2018-03-02:
Ok Latest version of UI is here : https://github.com/luc-github/ESP3D-WEBUI/blob/master/index.html.gz
But seems not the problem

Gi0cas, 2018-03-02:
I've updated now the UI, but the error persist. If you try to create a button to switch on/off your sonoff led it works? or the problem is it only on my sonoff ?

luc-github, 2018-03-02:
yes I did all like you 
[image]

adding space in command / changing web port to 14006 / use only NL in terminal / using Chrome and win 10

I should say I am dry ..... I do not reproduce your terminal issue neither led issue

luc-github, 2018-03-02:
may can we have clue in debug mode ?
in config.h enable DEBUG and Serial output
```
//DEBUG Flag do not do this when connected to printer !!!
//be noted all upload may failed if enabled
#define DEBUG_ESP3D
//#define DEBUG_OUTPUT_SPIFFS
#define DEBUG_OUTPUT_SERIAL
//#define DEBUG_OUTPUT_TCP
```
on terminal i get : 
```

Debug Serial set

Set STA mode
M117 SSID NETGEAR_2GEXT_OFFICE2
SSID NETGEAR_2GEXT_OFFICE2
M117 Connecting.   
M117 Connecting..  
M117 Connecting... 
M117 Connecting....
M117 Connecting.   
M117 Connecting..  
M117 192.168.1.47
Setup Done
1 Web command
commandText:[ESP700]/macro1.g
Web Command:[ESP700]/macro1.g
Execute Command
Execute Command
Pin:13
Set:1
1 Web command
commandText:[ESP700]/macro2.g
Web Command:[ESP700]/macro2.g
Execute Command
Execute Command
Pin:13
Set:0

```
Cann you do same and share the output ?

Gi0cas, 2018-03-02:
Tomorrow i'll try on another ESP board. I have a Wemos D1 mini, i'll try on it. 
The only thing possible is that there is an error in my config file (or another file). When you corrected the bug at the beginning, i copied only the modifies you made to the code (command.cpp and config.h), because the official release available is yet the 0.9.99 and not 0.9.100 (bug-free).
Thanks for your help.

Gi0cas, 2018-03-02:
Debug Serial set

Set STA mode
M117 SSID FASTWEB-VH4B4U
SSID FASTWEB-VH4B4U
M117 Connecting.   
M117 Connecting..  
M117 Connecting... 
M117 Connecting....
M117 192.168.1.140
Setup Done
1 Web command
plain:[ESP800]
Web Command:[ESP800]
Not allowed
Not allowed
Execute Command
1 Web command
plain:[ESP400]
Web Command:[ESP400]
Not allowed
Not allowed
Execute Command
request:/macrocfg.json
type:application/octet-stream
1 Web command
commandText:[ESP700]/m1.g
Web Command:[ESP700]/m1.g
Not allowed
Not allowed
Execute Command
Not allowed
Not allowed
Execute Command
1 Web command
commandText:[ESP700]/m2.g
Web Command:[ESP700]/m2.g
Not allowed
Not allowed
Execute Command
Not allowed
Not allowed
Execute Command
Check Command:[ESP201] P13 V0
Not allowed
Not allowed
Execute Command
M117 Incorrect Cmd

luc-github, 2018-03-02:
Did you enabled Authentication ?

Gi0cas, 2018-03-02:
Yes, and i logged in with "admin" account

luc-github, 2018-03-02:
It show Authentication is enabled and password is wrong

Gi0cas, 2018-03-02:
It is strange, i logged in correctly

luc-github, 2018-03-02:
I will redo test with Authentication enabled - I did not know that setting was on

luc-github, 2018-03-02:
Did you changed admin password or use default one ?

Gi0cas, 2018-03-02:
I use a custom password, i am testing now without authentication

luc-github, 2018-03-02:
Ok Now I reproduce issue - looks like Authentication is not passed to macro command and so fail
it is same for terminal I guess - it is rejected because no password
[ESP201]P13 V0 pwd=admin works 
I will see what is the bug with macro authentication 
now the issue is more clear ^_^

Gi0cas, 2018-03-02:
YES !!! i disabled auth and all works ok!
So the problem is in the authentication...okok, i'm not crazy :-)

luc-github, 2018-03-02:
No but every information is important - so now I know where so check ^_^

shaddow501, 2018-03-02:
Dear Luc,
 I am experiencing the same problem when using the macro and upload to the "ESP" the pin value 
-----[ESP201] P12 V0----- not change but if I change "ESP" to "SD" the command works.

In the bottom of this page there is a line to send commands directly and there this command works ([ESP201] P12 V0)

I dont know if it help or not..

luc-github, 2018-03-02:
File on SD do not care about Authentication it just ask printer FW to print file
Fix is now on repo - it needed 2 commit because I have broken the no Authentication orz

But now it is solved

Gi0cas, 2018-03-02:
Thanks Luc, a simple question. If i want to connect sonoff serial to my MKS GEN 1.4 i have to solder a logic level converter 5-3.3v or i can connect directly to the board?

shaddow501, 2018-03-02:
Dear Luc

Since I have changed a bit of your code to support other functions that I have, could you please tell us where to change and implement the fix?

luc-github, 2018-03-02:
@shaddow501 check the 2 last commits on master branch

luc-github, 2018-03-02:
@Gi0cas sonoof is ESP8266, which is 3.3v when MKS Gen 1.4 is 5V - so apply voltage divider like I suggest or logic converter is safer

shaddow501, 2018-03-02:
Is it possible to get a link...??

luc-github, 2018-03-02:
@Gi0cas here what I did before : https://github.com/luc-github/ESP3D/blob/master/images/HW/Wires.png
If communication is not good - remove the voltage divider - it means - no need

luc-github, 2018-03-02:
@shaddow501 the links are in github interface this is the purpose of this tool:
[image]
[image]

shaddow501, 2018-03-02:
Thanks, finally found that tab ... sorry.

luc-github, 2018-03-03:
@Gi0cas let me know if issue can be closed - thank you

Gi0cas, 2018-03-04:
Yes, i think you can close it.
Thanks