luc-github/ESP3D #208: ESP3D on Prusa MK2S

vsky279, 2018-02-25:
Hi, I find ESP3D very interesting especially when it is relatively easy to build. Experimentally I made it run on the Prusa MK2S. What I see as a limitation is the use of the 8.3 filenames. Is there any chance to support long filenames or is there any technical limitation that would be hard to overcome?

vsky279, 2018-02-25:
Thank you Luc.
So it seems that Prusa has the Marlin limitation. The M20 command prints the 8.3 style. M33 command is unknown.
So the workaround is to implement the M33 command?
What needs to be changed to be able to upload long filenames?

luc-github, 2018-02-25:
As wrote Marlin does not support long file name - M33 option need to be enable at compilation but it is not a solution
Only 2 solutions : 
- push Marlin dev to implement real long filename but they won't for board compatibility reason 
 [http://github.com/MarlinFirmware/Marlin/issues/1263](http://github.com/MarlinFirmware/Marlin/issues/1263)
- move to another Firmware like Marlin Kimbra which support long filename

luc-github, 2018-02-25:
@vsky279  Your Prusa Mk2S use which board ?

vsky279, 2018-02-25:
Prusa MK2S uses MiniRambo1.3a.

I am not familiar with different firmwares. I was more thinking more about modifying the existing Prusa's fw to list long filenames on serial interface.

Haha, I can see that it was implemented but has reverted probably for Octoprint compatibility reasons (https://github.com/prusa3d/Prusa-Firmware/commit/18c107ba77e795376b05747526f0a58c5ef0dbe8).

luc-github, 2018-02-25:
I did not know there is a prusa FW - too many flavor  but looks like same as Marlin - and so check what have been done in MarlinKimbra for long file name it may apply
FYI Marlin Kimbra looks supporting MiniRambo

vsky279, 2018-02-25:
Thanks. I will have a look.

luc-github, 2018-03-01:
I guess we can close issue now

closed 2018-03-01 by luc-github

Lino77, 2018-09-25:
@vsky279 witch port did you use @ the mini rambo for the esp

vsky279, 2018-09-25:
@Lino77 I use the unused serial port 2. I think this one is used for example for the multimaterial update.
I needed to do the following changes to the firmware: https://github.com/vsky279/Prusa-Firmware/commit/5e65b47fb70401677b07164aa6a418c056fdcfe7.

3d-gussner, 2018-09-25:
I use just an USB cable from my Wemos D1or wemos lolin to the miniRambo on the mk2. Only thing you need is to compile the mk2 branch as the communication fix was merged in Dec 2017 but the hex files never been updated since Nov 2017. You also can try my mk2 release https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM if you like. If you do please read the notes as some user had issues with not connected filament run out sensor.

AustinSaintAubin, 2018-10-16:
Tested [ESP3D v1.0](https://github.com/luc-github/ESP3D/releases/tag/1.0) & [ESP3D-WEBUI v1.0](https://github.com/luc-github/ESP3D-WEBUI/releases/tag/v1.0) on [Prusa i3 MK3 firmware v3.4.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.4.1). ESP3D configured for Marlin.

I am having issues with commands and reading not working as expected. Cant read from SD card. Cant read temperature. Can only home, and set temperature. cant seem to be able to read anything. Tested by flashing to an ESP32, and ESP8266.

AustinSaintAubin, 2018-10-17:
@luc-github looks like its an issue with serial communication for the [Wemos® Nodemcu ESP8266](https://www.banggood.com/Wemos-Nodemcu-Wifi-And-ESP8266-NodeMCU-1_3-Inch-OLED-Board-White-p-1160048.html?rmmds=myorder&cur_warehouse=CN) and [Wemos® ESP32](https://www.banggood.com/Wemos-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?rmmds=myorder&cur_warehouse=CN) that i am using. Serial communication is slow to send a receive, even at high bauds like 115200. Tested hardware with simple serial sketches... does not seem to be a hardware issue... not sure what I am doing wrong. 

Moved my issue to new ticket [Slow & Intermittent Serial Communication #269]