luc-github/ESP3D #209: Werror=unused-value

shaddow501, 2018-02-28:
Dear Luc

When trying to compile on ESP32 Lolin I am starting to get this error.

command.cpp: In static member function 'static void COMMAND::read_buffer_serial(uint8_t*, size_t)':

command.cpp:1506: error: value computed is not used [-Werror=unused-value]

         *b++;

             ^

What should I do to fix that?

luc-github, 2018-02-28:
you can disable this compilation flag which is anoying - I am guessing you are using command line idf compilation mode with arduino as component  ?
or you can change the function like this:

```
//read a buffer in an array
void COMMAND::read_buffer_serial(uint8_t *b, size_t len)
{
    for (long i = 0; i< len; i++) {
        read_buffer_serial(b[i]);
       //*b++;
    }
}
```
it looks like I did not merged this part from my private repo yet

luc-github, 2018-02-28:
seems have a lot gap - need to sync later
here the error solving : 
```
 if (PhyMode == (WIFI_PHY_MODE_11G))BRIDGE::print(F("11g"), output);
    else if (PhyMode == (WIFI_PHY_MODE_11B))BRIDGE::print(F("11b"), output);
    else if (PhyMode == (WIFI_PHY_MODE_11N))BRIDGE::print(F("11n"), output);
```

shaddow501, 2018-02-28:
Thanks, Now it compiles

luc-github, 2018-02-28:
I will sync master branch tomorrow

luc-github, 2018-03-01:
done with https://github.com/luc-github/ESP3D/commit/1697f68f10fc5c1b3ee5a03c11f690290da1b9d9

luc-github, 2018-03-01:
thank you for pointing this -  I close issue now

closed 2018-03-01 by luc-github