luc-github/ESP3D #210: Disable ESP3D Serial output

shaddow501, 2018-03-02:
Dear Luc 

First, it is not an issue but more of help needed.

Since my printer have only one USB port (it uses USB to serial on its controller to the Arduino RX / TX lines) 
And I would like to add your Wireless support but still have the USB function as well I need to know how to disable all the commands from the ESP3D sent by serial to the controller pins.

My Scenario is that on the ESP32 I use the second serial port (HardwareSerial Serial(2)) to get the commands that comes from the Onboard USB to serial controller into the ESP second serial port and than to send them to the serial that is connected to the printer board RX/TX lines. (which is the ESP_SERIAL_OUT).

Long to short, getting commands from Serial2 and forwarding them to ESP_SERIAL.OUT.

This toggling between ESP3D commands and the Serial2 commands are done by the GPIO state.

I dont care about the commands that are sending from the printer to the ESP3D module/Serial2 since they do not bother the parameters that are being sent.
Only interested to block the TX commands from the ESP3D software.

And if I mark in the ESP3D check status check box, it gets the Serial2 commands routed to the ESP_SERIAL_OUT to very slow,
I would like to be able to see what is going with the printer via wireless but not have any delay or sluggish reaction from the Serial2 to the ESP_SERIAL_OUT process.

With this code I have two issues:

1. Commands like "Auto check every" still sending M105, M114 to the ESP_SERIAL_OUT even though it is not suppose to.
2. When the Pin Is low (means that the Serial2 should be now active) if I set "Auto check every" it gets the commands come from the Serial2 to be delayed and not at real time.

How can I fix it?
-
My code looks like that: (well part of it lol)

CODE:

      //be sure wifi is on to proceed wifi function
       //Wifi mirrring to serial code
     if (WiFi.getMode()!=WIFI_OFF ) {
#ifdef CAPTIVE_PORTAL_FEATURE
        if (WiFi.getMode()!=WIFI_STA ) {
            dnsServer.processNextRequest();
        }
#endif

 //web requests
        web_interface->web_server.handleClient();

//Converting between Serial2 to ESP3D Commands

  digitalRead(12);
  if (digitalRead(12) == LOW)
      {
  while (Serial2.available() >0) {
    char inByte = Serial2.read();
    ESP_SERIAL_OUT.write(inByte);
  }
    }
    else
    {

#ifdef TCP_IP_DATA_FEATURE
        BRIDGE::processFromTCP2Serial();
#endif
    }

        BRIDGE::processFromSerial2TCP();
    //in case of restart requested
    if (web_interface->restartmodule) {
        CONFIG::esp_restart();
    }
     }
}

luc-github, 2018-03-02:
there is a flag you can use to block output to serial:
`web_interface->blockserial` set to true it not allow output

about delay on serial - may be need to flush to force output

shaddow501, 2018-03-04:
Dear Luc

Even after I set web_interface->blockserial = true; 
I still get commands in the serial output of M114 / M105.

And in the web display I get  - Error 503 :Communication locked!
how can I block the serial output?

luc-github, 2018-03-04:
yes you get `Error 503 :Communication locked!` because you set `web_interface->blockserial = true`
As you see if you set the flag no command is sent : 
https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L1282
https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L1476

now I do not know how you handle this flag but be sure it is not handled in other part of the code as today is it , because some part disable it and unlock serial so you must search in the code

shaddow501, 2018-03-04:
So what is relate to the M114 & M105?

luc-github, 2018-03-04:
they are like any web_command

shaddow501, 2018-03-04:
Ok, where do I find the place that I should modify?

by the way do I write this command wrong?
[ESP444]RESET pwd=<admin password>

shaddow501, 2018-03-04:
By the way I am trying to run the macro on the ESP (as target) - [ESP444]RESTART, but nothing happens.
If I write the same command at the command line the module performs reset.

luc-github, 2018-03-04:
macro code has been updated recently - reading macro file has a bug then authentication issue
also there is already this command in ESP tab
[ESP444]RESET pwd=admin_password is necessary only for serial or tcp command - for macro is it not as you are already identified using UI

shaddow501, 2018-03-04:
Ok so at the moment this shouldnt work I guess till the next fix
[ESP700][ESP201]P12 V0

luc-github, 2018-03-04:
[ESP700] use file name not esp command

shaddow501, 2018-03-04:
When I add it to the shortcuts 
[image]
the command is like that [ESP201]P12 V0.
but on the monitor webpage its output like this [ESP700][ESP201]P12 V0.

luc-github, 2018-03-04:
yes because you need to create file containing the file containing the command not the command itself
watch how Giorgio did : https://www.youtube.com/watch?v=AIQAJpjCiXM&feature=youtu.be

shaddow501, 2018-03-04:
Oh now it is clear

shaddow501, 2018-03-04:
I dont know if it is an issue but when when I send the file to be processed it takes a long time to respond till I get OK. on the terminal view.

auto-check position every..  does block everything else.  if I press several times to send the macro eventually somehow it get processed.
I think that the auto-check position every... & auto-check every.. holding the process time (maybe a delay in the code is the problem since there is no response.)
the funny thing that the check box is not checked but it still send those commands...

M114: Get Current Position; 
M105: Get Extruder Temperature

So it works without any request that it will work. (again the check box isnt marked)

shaddow501, 2018-03-09:
Dear Luc

Did you check this issue?

luc-github, 2018-03-09:
I am able to reproduce this issue : https://github.com/luc-github/ESP3D-WEBUI/issues/20
I will work on it today

luc-github, 2018-03-17:
this feature is already added to Todo listfor 2.X, so I guess issue can be closed now

closed 2018-03-17 by luc-github