luc-github/ESP3D #212: ESP8266 + RAMPS 1.4 doesn't work

luxarts, 2018-03-24:
Hi!

When connect RX and TX from wifi board (ESP-01) to TX0 and RX0 from Arduino Mega I haven't response. The webpage works and I can send commands but the Arduino don't receive it. I tested the wifi board with a USB-Serial converter and it works fine. Also tested the Arduino serial port and works fine too.
I wonder if it's because the serial port number 0 (D0 and D1) are used by the USB-Serial converter from Arduino board. If that's the cause the connection to RAMPS 1.4 are wrong.

Anyone has this problem?
Thanks!

luc-github, 2018-03-24:
Does your mega is also connected by USB ?
you cannot have mega connected by USB and use TX0 and RX0 in same time because they are identical if you want to use arduino mega USB you must connect ESP to Serial 1 / 2 / 3 I forget which one is available

luxarts, 2018-03-24:
I'm using a an Arduino DUE and I found the problem:
The Serial-USB converter in DUE that is in programming port seem like is not the same that Arduino Mega has. The TX pin (from the ARM) works great but the RX don't and it's because the USB-Serial converter onboard uses the RX line even without the cable connected. I tried changing the USB port to native one but didn't work either.
The solution I found it's configure the firmware (Repetier in my case) to use the Serial1 (Z-Min and Z-Max pins in RAMPS 1.4) as Bluetooth connection. In this way you can use the USB cable and the WIFI board at same time.
Please delete the image from the wiki, it will confuse people.

I'll try to implement a software serial in port in D57 and D58 so the AUX-1 port can be used.

luc-github, 2018-03-24:
sorry what is not working as described ? due or ramps ?
I use both  as  described  without. issue

luxarts, 2018-03-24:
RAMPS 1.4(hacked to work at 3.3v) + Arduino DUE with ESP-01 connected to D0 and D1 (AUX-1) using Repetier 1.0.1.

luc-github, 2018-03-24:
well both works as described on my side 
looks like your rx/tx connection is noisy -
What Firmware are you using ?
What Printer Firmware are you using ? 
Are you sure you set right baud rate on both side ?

luxarts, 2018-03-24:
Firmware (ESP): v0.9.99 with lastest WEB-UI
Firmware (Printer): Repetier v1.0.1
ESP board: ESP-01
Yes, both devices are in the same baudrate (115200).

As I said, I connected to Serial 1 and configured that port like a Bluetooth in the printer's firmware and now it works. So I doubt that the problem is noise. I'll try later with shielded cables anyways.

luc-github, 2018-03-24:
Well I do not know where the issue is, I share wiring I am using on due/radds and mega/ramps and I know others people use it also.

luc-github, 2018-03-28:
I think issue can be closed as not FW releated

closed 2018-03-28 by luc-github