luc-github/ESP3D #213: Reset ESP8266 wirelessly

vader333, 2018-03-24:
I have a Tranceboard Ozone (new smoothieware clone) and it has an esp8266. It used a modified version of ESP3D. I tried flashing the newest ESP3D to it, but somehow it screwed up and it displays a broken webpage. How can I reset/format SPIFFS without a TTL to USB cable?

luc-github, 2018-03-24:
connect your ESP using the data port (8888 by default) using putty in raw mode for example and use the 
https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands
format command 
Format SPIFFS
[ESP710]FORMAT pwd=<admin password>

luc-github, 2018-03-24:
@vader333  I saw the board and wifi connector here http://singularidea.cn/tranceboard/ - 
Please can you share the wifi module picture ? I do not it see on website but I see connection description so I guess it is sells with board ?

luc-github, 2018-03-24:
@vader333 is this board better than your MKS + TFT ?

vader333, 2018-03-24:
[image]

This is the wifi module. 

It's not a bad board. Includes many features like auto shutdown, nice screw terminals for power, blade fused circuitry for heaters, and pins dedicated to Octopi. It requires external stepper motors though, which I think is a good design choice, but is a little messy to implement.

It's much much much better than an MKS Sbase. I'm using MKS LV8729 drivers, and it's extremely quiet on 128 microsteps. Draw back is that you can't use their pretty touch screen with wifi, as they both use the same pins.

[image]

luc-github, 2018-03-24:
But I see several connectors on their TranceScreenOne 
Is there no way to hook wifi module on it like MKS-TFT ?

vader333, 2018-03-24:
According to the manufacturers, it's possible, but you wouldn't be accessing the sd card on the main board. The TFT acts like a gcode machine. And yea, both 13 and 18 share branches of the same pins. Still, it's not a bad arrangement to hook the esp8266 to the TFT, you just don't get to change the config as easily.

vader333, 2018-03-24:
I'm afraid they don't have anything of the sort. Could probably ask them, but it isn't opensource or anything like that.

luc-github, 2018-03-24:
沒問題- thanks for sharing - always interresting to learn new stuff ^_^

luc-github, 2018-03-24:
Is your ESP board ok now ? Format solved your issue ?

vader333, 2018-03-24:
Haha, I haven't tried yet =P. I'll do it now

luc-github, 2018-03-24:
Hi ok that is more clear ^_^
Yes they must use online resources to help to display like smoothie UI do and this won't work if you are nor connected to internet - that is why I made ESP3D web UI independant of external resource and so working in AP mode.
You said they use a modified version of my FW ? 

you can format using this command on ESP3D: [ESP710]FORMAT
the password is only necessary if you have enabled login feature which is disabled by default

but formating is not necessary if you already flashed their board with my FW
if board has my  recent FW just go to embedded uploader
`<your IP>?forcefallback=YES`
you should get a UI allowing to delete/upload files : [image]
So delete files and upload latest index.html.gz

vader333, 2018-03-25:
OMG IT WORKS!!!!! The format did the trick! It went to your fallback, and then I could upload the correct files to the SPIFFS! Thank You so much! If you have a donate link, I'd be very happy to donate!

Although, it can't access the SD Card though. But I guess I'll have to ask the manufacturers about that haha. I'm very happy, I was about to send this back to China to have them reflash their interface.

luc-github, 2018-03-25:
Good to see you make progress
Donate link is in read me : https://github.com/luc-github/ESP3D/blob/master/README.md#donate

About SD card it is weird 
What is M20 command result in terminal

vader333, 2018-03-26:
[image]
It looks ok... I'll try reflashing the firmware.

luc-github, 2018-03-26:
Did you select the FW target as smoothieware in webUI ?
can you screenshot the UI ?

vader333, 2018-03-26:
Yup, I selected Smoothieware.

[image]

luc-github, 2018-03-26:
there is an issue but I cannot reproduce it, update with latest UI https://github.com/luc-github/ESP3D-WEBUI/blob/master/index.html.gz, also there some bug fixes on FW - you do not have the latest one 

[image]

vader333, 2018-03-26:
[image]

I've updated, but still can't though.

[image]
[image]
[image]
[image]

luc-github, 2018-03-26:
yes any SD upload will fail as SD is not seen - did you do any modification in FW before flashing ?

vader333, 2018-03-26:
Nope. It seems to be purely a UI error. I can make a macro that references /sd/on_boot.gcode, and it runs.
[image]

luc-github, 2018-03-26:
can you type this command in manual command and share the output : 
`[ESP800]` I suspect EEPROM data is corrupted by previous FW

vader333, 2018-03-26:
[image]
Here you go

luc-github, 2018-03-26:
So yes this is the issue, correct output is:

```
[ESP800]
FW version:0.9.99 # FW target:smoothieware # FW HW:Serial SD # primary sd:/ext/ # secondary sd:/sd/ # authentication:no
```
as you see SD is not defined

luc-github, 2018-03-26:
there 2 ways 
1 -  use RESET command to clean EEPROM with proper data
2 - set manually the values

luc-github, 2018-03-26:
as you used a previous FW the cleanest would be to do an EEPROM reset to be sure no other data is missing

luc-github, 2018-03-26:
but I can give you the manual command to test first

vader333, 2018-03-26:
I'll do a reset then.

luc-github, 2018-03-26:
`[ESP401]P=851 T=B V=2`
and
`[ESP401]P=852 T=B V=1`

becareful these update directly the EEPROM

luc-github, 2018-03-26:
try these command first - to confirm the fix

luc-github, 2018-03-26:
you need to refresh page after of course to see the update of `[ESP800]` output

vader333, 2018-03-26:
I've used the commands, and the interface is fixed, but I still can't access the SD card

[image]

luc-github, 2018-03-26:
press the button /SD nothing happen ?

luc-github, 2018-03-26:
what the command `ls /` and `ls -s /sd` give as output ?

[image]

vader333, 2018-03-26:
How do I use the Reset command? Just type RESET?

luc-github, 2018-03-26:
`[ESP444]RESEST`
then `[ESP444]RESTART`

luc-github, 2018-03-28:
@vader333 
Any update ? 
Is you issue fixed ?

luc-github, 2018-03-29:
upload is very slow and subject to error as it is serial upload
but even upload fail listing the files is working - right ?

vader333, 2018-03-29:
Yup. Listing file works

luc-github, 2018-03-29:
You upload to which directory ? /sd or /ext ?

vader333, 2018-03-29:
/sd

luc-github, 2018-03-29:
and can you share the name of the file you try to upload - I will try to reproduce

vader333, 2018-03-29:
MD1_xyzCalibration_cube.gcode

luc-github, 2018-03-29:
remove MD1_ from file name
Smoothie does not like name starting like GCODE command (MXXX)

luc-github, 2018-03-29:
I meet this issue before and I have just checked with small file - upload failed until file is renamed

luc-github, 2018-03-29:
but my comment about upload stay same :  Serial upload take ages ...

vader333, 2018-03-29:
Ah, okies, I'll try that! ^_^

vader333, 2018-03-29:
Yes! It was able to upload! Thank you so much for your help!

closed 2018-03-29 by vader333

luc-github, 2018-03-29:
great enjoy 😸