luc-github/ESP3D #214: Support for GRBL firmware

Georges760, 2018-04-01:
Hi Luc,

amazing jog you did on this project, I love it.

I plan to add GRBL fimware support on ESP3D. Basically GRBL is a very simple Marlin, that does not handle Temperature and Extruder (in fact there is a spindle instead of Extruder).

I will let you know my progress on it.

luc-github, 2018-04-01:
Hi @Georges760 thank you actually grbl FW is already done : 
https://github.com/luc-github/ESP3D/issues/179#issuecomment-373820850

I did not pushed it to repository yet but should be done soon

luc-github, 2018-04-01:
@Georges760 you use GRBL for with which machine ? looks like milling machine for what I see - right ?

Georges760, 2018-04-01:
Great news... less job to do ;) I see it is very freah work, so I will wait for your push ;)

I am customizing GRBL for my own personnal project : a polar coordinate with TMC2130 driver for silent and sensorless homing... currently customizing GRBL and doing a PCB for it.

I would like to tell you more about it but not here... maybe you can send me an email in french , you can check it on my website in my github profile ;)

luc-github, 2018-04-01:
checked already ^_^, toulouse and wanchai HK, 2 nice places 
I will do for sure

Georges760, 2018-04-01:
i saw also your photo of HK taked from Kowloon on your youtube page ;) haha

Look like we have many interests in common...

luc-github, 2018-04-01:
Yes I love Wanchai area and part of my familly are in Toulouse ^_^

I do not see your email in your website - and I am not sure you use gitter chat - so I have dropped you a linkedin invitation and here my email luc[at]tech-hunters[dot]com

hope to chat later

luc-github, 2018-04-01:
Also the Youtube page is not mine - same name but definitly not me ^_^

Maglin, 2018-04-14:
This is exactly what I was looking for.  When might this be available?  I have an MPCNC that I'm looking to control on the cheap with a Mega and RAMPs 1.4 running 5 drivers.  Two of the drivers are just dueled X and Y so easy enough to just jumper those if I can't config through GRBL.  The interface was my limiting factor.  I was hoping an ESP32 with SD card could be used and just connected to wirelessly to run the machine was around.  If you would like another tester I'd be willing to give it a go right now.  I'm not overly happy running Marlin on it as performance isn't what I want.  I have an SD module inbound currently for my ESP32.

luc-github, 2018-04-14:
Currently  the UI to configure and monitor GRBL FW is working and can be pushed but the host part : reading SD card and sending GCODE command by command is not yet ready

Also as you mention ESP32 - there is currently an issue with async webserver I need to did in, not related to my code but need to fix to use it on ESP32 - an I am currently traveling  so need time

luc-github, 2018-04-21:
I have just pushed update for Async and so grbl support in ESP3D and ESP3D-WEBUI
I was not able to find solution for ESP32 in async server part so I have rewriten the command processor in ESP-WEBUI to avoid any command overlap - now all commands go to FIFO and give better control to requests

closed 2018-04-21 by luc-github