luc-github/ESP3D #215: SD Card - full names

sulicz, 2018-04-09:
Why are file names on SD card in 8+3 format? Is it limit of printers firmware (Marlin) or is possible to make it full?

luc-github, 2018-04-09:
Yes it is marlin limitation, others fw do not have such limitation

sulicz, 2018-04-09:
Is problem on marlin site? May be can be sloved..

luc-github, 2018-04-09:
I do not think so, it whas already raised on issues and on also on Marlin  : github.com/MarlinFirmware/Marlin/issues/6165
But you can try Marlin Kimbra which support long files

M20 command send 8.3 filenames only, and using M33 to get long filenames for display as only 8.3 filenames can be used for print/del commands is too performance/serial bandwidth consuming

luc-github, 2018-04-09:
Another solution - easy this one : use wifi SD card like toshiba flashair and this :  https://github.com/luc-github/Flashair-FM or embeded file uploader

sulicz, 2018-04-11:
Thanks for help.. I changed to repetier... funny.. there si limitation for 26 letters :smile: 
Is possible to implement filter for uploading filenames to force 8+3 or max 26 letter for each SW? I think this is think what I can try to implement my self... it will be funny.. :rofl:

sulicz, 2018-04-11:
In my fork I modified uploder to create short file names for Marlin. (I revert my printer to Marlin)

sulicz, 2018-04-12:
I only cut length of name and extension, so there not comes ~1. I will create config parameter to enable for this feature. If user wants he can enable it. For me is very uncomfortable check length of filenames and rename it. I print something and then I gcode not use again.

luc-github, 2018-04-21:
I think issue can be closed -

closed 2018-04-21 by luc-github