luc-github/ESP3D #216: Upload using Slic3r - octoprint

sulicz, 2018-04-09:
Is there any possibility to use slic3rs function for octoprint for uploading files directly to sd card?

luc-github, 2018-04-09:
No, it use M28/M29 commands to upload to printer sdcard , upload using octoprint is uploading on octopi  computer. not sd  card 
Currently  only  only azteeg  x5  wifi allows  to  share we card and do fast  upload

sulicz, 2018-04-09:
I did not studied sources of ESP3D, Slic3r or octoprint. But can by used same method for uploading thought web for uploading from octoprint? Some small hack? I ask before I spend some hours of studiing something useless. Thansk

luc-github, 2018-04-09:
as said the only way without hardware modification of your printer board is to upload using M28/M29

this is not related to ESP3D = so again No

luc-github, 2018-04-10:
Another solution - easy this one : use wifi SD card like toshiba flashair and this : https://github.com/luc-github/Flashair-FM or embeded file uploader

sulicz, 2018-04-11:
I now checked protocol of slic3r -> octoprint upload. It uses HTTP POST request. I think can be possible to accept this request and store data to file. May you can use it. To me is too hard to implement it, now.

POST /api/files/local HTTP/1.1
TE: deflate,gzip;q=0.3
Connection: TE, close
Host: 127.0.0.1
User-Agent: libwww-perl/6.15
Content-Length: 1250967
Content-Type: multipart/form-data; boundary=xYzZY
X-Api-Key: C04E1BFB15A143B7B833D45E1418E998

--xYzZY
Content-Disposition: form-data; name="file"; filename="CalibrationZ.gcode"
Content-Type: text/plain

; generated by Slic3r 1.33.8 on 2018-04-11 at 21:46:18

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.67mm
; infill extrusion width = 0.67mm
; solid infill extrusion width = 0.67mm
; top infill extrusion width = 0.67mm

M190 S90 ; set bed temperature and wait for it to be reached
M80 ; start ATX

luc-github, 2018-04-11:
this is what is used today already - the post is not the bottleneck it is the M28/M29 protocol

sulicz, 2018-04-12:
I am not sure i I understand good. Why is problem in M28/M29? In my opinion is possible to handle this post request and store this gcode using M28/M29 in some folder, for example in folder with name of X-Api-Key - user can specific. It will be not so secure, if there will not be checked key, but will be nice hack of this feature.

luc-github, 2018-04-12:
Procees is following:
* Post send data upload to ESP3D 
* ESP3D send M23 command to Marlin to inform that all following information need to be write to SD card
* ESP3D send all commands received by POST to Marlin thrue Serial which is very slow compare to TCP
* Once Post data are all received and Serial data are all sent, ESP3D inform Marlin  using M29 that no more data will be sent and SD file can be closed and normal communication can be resumed.

I do not understand what you mean by storing code in some folder ( SD ? ESP?) to do what ?

sulicz, 2018-04-12:
Yes this I get - I red you code. But I am not sure if I can write it so good. This folder I mean on SD card. To separate files uploaded by slic3r for other.

sulicz, 2018-04-12:
Sorry, I will try it again. I do not know if you use Slic3r. There is a function to direct upload gcode file to octoprint. It have security key which is generated on octoprint and request for upload must have the same (shown as X-Api-Key:). Slic3r uploads send post request on http://_host_/api/files/local and i get idea that will be nice, if we can use this featuer with ESP3D. In slic3r make gcode and directly send it to SD card.

luc-github, 2018-04-12:
No I do not use Slic3r  - I understand now

sulicz, 2018-04-12:
What you think, can you do it? I do not think is complicate. But when I read you code, there is many waiting, repeats and error checks and I am not shure I can write it so good and so fast as you.

luc-github, 2018-04-12:
I do not use Slic3r neither octoprint, so I need to see
I think this will impact the authentication on ESP3D if enabled

luc-github, 2018-04-12:
May slicer change url upload path ?

sulicz, 2018-04-12:
What I found it is hardcoded in code. And I found one small complication. To test connection slic3r opens url /api/version and need response.
Here is some documentation 
http://docs.octoprint.org/en/master/api/version.html
http://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder
It seems that octoprint can upload directly to SD, bud slic3r do not support it and use api/file/local

luc-github, 2018-04-12:
so you should write a pluggin for slic3r, this would work with ethernet board like smoothie and duet

sulicz, 2018-04-12:
Slic3r is not modular, so it will need modification of all program and I am not sure if this change will be accepted to master

luc-github, 2018-04-12:
are you sure, looks like  latest version support pluggin/extension 
How_to_create_a_custom_Slicer_version_with_selected_extensions_pre

luc-github, 2018-04-12:
https://www.slicer.org/wiki/Documentation/Nightly/SlicerApplication/ExtensionsManager#How_to_create_a_custom_Slicer_version_with_selected_extensions_pre-installed.3F

sulicz, 2018-04-12:
Thanks for imformation. I will check it. What API (post request) I should use to upload to SD on ESP3D?

probonopd, 2018-09-08:
I have implemented __partial OctoPrint API for the ESP8266__ in https://github.com/probonopd/WirelessPrinting/. So you can print from slicers directly to the 3D printer via WLAN using the ESP8266.